espressif/esp-idf #11408: Not able to stay in ad2p connected

espradio, 2023-05-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am trying the standard AD2P SOURCE program on ESP32. However, When trying to connect to a specific BT speaker it is NOT working.(It is not able staying A2DP connected state for more than 3 seconds)
But it works on other BT speakers. Here is the full log.
Branch: master

```

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7104
load:0x40078000,len:15660
load:0x40080400,len:4
load:0x40080404,len:3872
entry 0x40080648
I (29) boot: ESP-IDF v5.2-dev-544-g54576b7528-dirty 2nd stage bootloader
I (29) boot: compile time May 16 2023 18:40:08
I (32) boot: Multicore bootloader
I (36) boot: chip revision: v3.0
I (39) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (49) boot.esp32: SPI Flash Size : 2MB
I (53) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot: End of partition table
I (96) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=2a9c4h (174532)                                                                                                                                                              map
I (168) esp_image: segment 1: paddr=0003a9ec vaddr=3ffbdb60 size=04570h ( 17776)                                                                                                                                                              load
I (175) esp_image: segment 2: paddr=0003ef64 vaddr=40080000 size=010b4h (  4276)                                                                                                                                                              load
I (177) esp_image: segment 3: paddr=00040020 vaddr=400d0020 size=a7450h (685136)                                                                                                                                                              map
I (429) esp_image: segment 4: paddr=000e7478 vaddr=400810b4 size=17454h ( 95316)                                                                                                                                                              load
I (481) boot: Loaded app from partition at offset 0x10000
I (481) boot: Disabling RNG early entropy source...
I (493) cpu_start: Multicore app
I (493) cpu_start: Pro cpu up.
I (493) cpu_start: Starting app cpu, entry point is 0x400812e0
I (0) cpu_start: App cpu up.
I (513) cpu_start: Pro cpu start user code
I (513) cpu_start: cpu freq: 160000000 Hz
I (513) cpu_start: Application information:
I (518) cpu_start: Project name:     a2dp_source
I (523) cpu_start: App version:      v5.2-dev-544-g54576b7528-dirty
I (530) cpu_start: Compile time:     May 16 2023 18:47:01
I (536) cpu_start: ELF file SHA256:  e39c137f72a7e95b...
I (542) cpu_start: ESP-IDF:          v5.2-dev-544-g54576b7528-dirty
I (549) cpu_start: Min chip rev:     v0.0
I (554) cpu_start: Max chip rev:     v3.99
I (559) cpu_start: Chip rev:         v3.0
I (564) heap_init: Initializing. RAM available for dynamic allocation:
I (571) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (577) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (583) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (589) heap_init: At 3FFCBC20 len 000143E0 (80 KiB): DRAM
I (595) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (602) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (608) heap_init: At 40098508 len 00007AF8 (30 KiB): IRAM
I (616) spi_flash: detected chip: gd
I (619) spi_flash: flash io: dio
W (622) spi_flash: Detected size(16384k) larger than the size in the binary imag                                                                                                                                                             e header(2048k). Using the size in the binary image header.
[... 225 lines cut ...]
W (201386) BT_APPL: reset flags
I (201396) BT_AV: bt_app_av_sm_hdlr state: 4, event: 0x0
W (201466) BT_HCI: hcif conn complete: hdl 0x80, st 0x0
I (201476) BT_AV: event: 16
E (201776) BT_APPL: bta_av_rc_create ACP handle exist for shdl:0
W (201886) BT_APPL: new conn_srvc id:18, app_id:0
W (201886) BT_APPL: bta_dm_pm_ssr conn_srvc id:18, app_id:0
W (201886) BT_APPL: bta_dm_pm_ssr:2, lat:1200
I (201896) BT_AV: bt_app_av_sm_hdlr state: 4, event: 0x0
I (201896) BT_AV: a2dp connected
E (202096) BT_BTM: tBTM_SEC_DEV:0x3ffd35ac rs_disc_pending=0

W (202126) BT_HCI: hcif link supv_to changed: hdl 0x80, supv_to 8000
W (202146) BT_HCI: hcif link supv_to changed: hdl 0x80, supv_to 8000
W (208066) BT_HCI: hcif disc complete: hdl 0x80, rsn 0x8
W (208066) BT_APPL: bta_dm_act no entry for connected service cbs
W (208066) BT_AVCT: ccb 0 not allocated
E (208066) BT_APPL: bta_av_rc_create found duplicated handle:0
I (208076) BT_AV: bt_app_av_sm_hdlr state: 5, event: 0x0
I (208086) BT_AV: a2dp disconnected
I (211386) BT_AV: bt_app_av_sm_hdlr state: 3, event: 0xff00
I (211386) BT_AV: a2dp connecting to peer: f8:23:45:44:23:43
W (211386) BT_APPL: reset flags
I (211396) BT_AV: bt_app_av_sm_hdlr state: 4, event: 0x0
```

espradio, 2023-05-20:
@wmy-espressif Any thoughts?

espradio, 2023-05-22:
It is an BT android device found in car

shangke1112, 2023-05-24:
[libbtdm.tar.gz](https://github.com/espressif/esp-idf/files/11552158/libbtdm.tar.gz)
Please use this lib for testing ,and send me the log.

espradio, 2023-05-28:
Hi @shangke1112 
Please find the log with debug image

```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7132
ho 0 tail 12 room 4
load:0x40078000,len:15672
ho 0 tail 12 room 4
load:0x40080400,len:4
load:0x40080404,len:3884
entry 0x40080650
I (33) boot: ESP-IDF v5.2-dev-823-g903af13e84-dirty 2nd stage bootloader
I (33) boot: compile time May 28 2023 18:38:20
I (35) boot: Multicore bootloader
I (39) boot: chip revision: v3.0
I (43) boot.esp32: SPI Speed      : 40MHz
I (48) boot.esp32: SPI Mode       : DIO
I (52) boot.esp32: SPI Flash Size : 2MB
I (57) boot: Enabling RNG early entropy source...
I (62) boot: Partition Table:
I (66) boot: ## Label            Usage          Type ST Offset   Length
I (73) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (81) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (88) boot:  2 factory          factory app      00 00 00010000 00100000
I (96) boot: End of partition table
I (100) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=2a9d4h (174548)                                                                                                                                                              map
I (171) esp_image: segment 1: paddr=0003a9fc vaddr=3ffbdb60 size=04574h ( 17780)                                                                                                                                                              load
I (179) esp_image: segment 2: paddr=0003ef78 vaddr=40080000 size=010a0h (  4256)                                                                                                                                                              load
I (181) esp_image: segment 3: paddr=00040020 vaddr=400d0020 size=a754ch (685388)                                                                                                                                                              map
I (433) esp_image: segment 4: paddr=000e7574 vaddr=400810a0 size=173f8h ( 95224)                                                                                                                                                              load
I (485) boot: Loaded app from partition at offset 0x10000
I (485) boot: Disabling RNG early entropy source...
I (497) cpu_start: Multicore app
I (497) cpu_start: Pro cpu up.
I (497) cpu_start: Starting app cpu, entry point is 0x40081220
I (0) cpu_start: App cpu up.
I (517) cpu_start: Pro cpu start user code
I (517) cpu_start: cpu freq: 160000000 Hz
I (517) cpu_start: Application information:
I (522) cpu_start: Project name:     a2dp_source
I (527) cpu_start: App version:      v5.2-dev-823-g903af13e84-dirty
I (534) cpu_start: Compile time:     May 28 2023 18:38:15
I (540) cpu_start: ELF file SHA256:  16e39f079...
I (546) cpu_start: ESP-IDF:          v5.2-dev-823-g903af13e84-dirty
I (553) cpu_start: Min chip rev:     v0.0
I (557) cpu_start: Max chip rev:     v3.99
I (562) cpu_start: Chip rev:         v3.0
I (567) heap_init: Initializing. RAM available for dynamic allocation:
I (574) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (580) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (586) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (592) heap_init: At 3FFCBC30 len 000143D0 (80 KiB): DRAM
I (598) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (605) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (611) heap_init: At 40098498 len 00007B68 (30 KiB): IRAM
I (619) spi_flash: detected chip: gd
I (622) spi_flash: flash io: dio
W (626) spi_flash: Detected size(16384k) larger than the size in the binary imag                                                                                                                                                             e header(2048k). Using the size in the binary image header.
I (640) coexist: coex firmware version: ebddf30
I (645) app_start: Starting scheduler on CPU0
I (649) app_start: Starting scheduler on CPU1
I (649) main_task: Started on CPU0
I (659) main_task: Calling app_main()
I (679) BTDM_INIT: BT controller compile version [ad2b7cd]
[... 163 lines cut ...]
set state id 1,state 24
set state id 1,state 24
set state id 1,state 2
set state id 1,state 22
set state id 1,state 2
set state id 1,state 9
set state id 1,state 2
set state id 1,state 20
set state id 1,state 2
set state id 1,state 1
set state id 1,state 60
set state id 1,state 14
W (93749) BT_HCI: hcif conn complete: hdl 0x80, st 0x0
I (93749) BT_AV: event: 16
set state id 1,state 1
set state id 1,state 8
set state id 1,state 1
set state id 1,state 1
set state id 1,state 36
set state id 1,state 31
set state id 1,state 1
set state id 1,state 41
set state id 1,state 42
set state id 1,state 44
```

shangke1112, 2023-05-31:
According to the last log ,there may be an issue with encryption ,but I am not sure.Please use this lib and try again.
[libbtdm3.tar.gz](https://github.com/espressif/esp-idf/files/11611313/libbtdm3.tar.gz)

espradio, 2023-06-04:
Hi @shangke1112 Please find the log

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7132
ho 0 tail 12 room 4
load:0x40078000,len:15672
ho 0 tail 12 room 4
load:0x40080400,len:4
load:0x40080404,len:3884
entry 0x40080650
I (33) boot: ESP-IDF v5.2-dev-917-g37d11ea04d-dirty 2nd stage bootloader
I (33) boot: compile time Jun  4 2023 23:05:49
I (35) boot: Multicore bootloader
I (39) boot: chip revision: v3.0
I (43) boot.esp32: SPI Speed      : 40MHz
I (48) boot.esp32: SPI Mode       : DIO
I (52) boot.esp32: SPI Flash Size : 2MB
I (57) boot: Enabling RNG early entropy source...
I (62) boot: Partition Table:
I (66) boot: ## Label            Usage          Type ST Offset   Length
I (73) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (81) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (88) boot:  2 factory          factory app      00 00 00010000 00100000
I (96) boot: End of partition table
I (100) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=2aa3ch (174652)                                                                                                                                                              map
I (171) esp_image: segment 1: paddr=0003aa64 vaddr=3ffbdb60 size=04574h ( 17780)                                                                                                                                                              load
I (179) esp_image: segment 2: paddr=0003efe0 vaddr=40080000 size=01038h (  4152)                                                                                                                                                              load
I (181) esp_image: segment 3: paddr=00040020 vaddr=400d0020 size=a7634h (685620)                                                                                                                                                              map
I (433) esp_image: segment 4: paddr=000e765c vaddr=40081038 size=17460h ( 95328)                                                                                                                                                              load
I (485) boot: Loaded app from partition at offset 0x10000
I (485) boot: Disabling RNG early entropy source...
I (497) cpu_start: Multicore app
I (497) cpu_start: Pro cpu up.
I (497) cpu_start: Starting app cpu, entry point is 0x40081220
I (0) cpu_start: App cpu up.
I (517) cpu_start: Pro cpu start user code
I (517) cpu_start: cpu freq: 160000000 Hz
I (517) cpu_start: Application information:
I (522) cpu_start: Project name:     a2dp_source
I (527) cpu_start: App version:      v5.2-dev-917-g37d11ea04d-dirty
I (534) cpu_start: Compile time:     Jun  4 2023 23:05:44
I (540) cpu_start: ELF file SHA256:  0b70bf53c...
I (546) cpu_start: ESP-IDF:          v5.2-dev-917-g37d11ea04d-dirty
I (553) cpu_start: Min chip rev:     v0.0
I (557) cpu_start: Max chip rev:     v3.99
I (562) cpu_start: Chip rev:         v3.0
I (567) heap_init: Initializing. RAM available for dynamic allocation:
I (574) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (580) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (586) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (592) heap_init: At 3FFCBC30 len 000143D0 (80 KiB): DRAM
I (599) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (605) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (611) heap_init: At 40098498 len 00007B68 (30 KiB): IRAM
I (619) spi_flash: detected chip: gd
I (622) spi_flash: flash io: dio
W (626) spi_flash: Detected size(16384k) larger than the size in the binary imag                                                                                                                                                             e header(2048k). Using the size in the binary image header.
I (640) coexist: coex firmware version: ebddf30
I (645) app_start: Starting scheduler on CPU0
I (649) app_start: Starting scheduler on CPU1
I (649) main_task: Started on CPU0
I (659) main_task: Calling app_main()
I (679) BTDM_INIT: BT controller compile version [4f4ce33]
set state id 2,state 0 A0: 0x8001ad29
set state id 0,state 0 A0: 0x8004ef19
set state id 6,state 127 A0: 0x800447a8
[... 278 lines cut ...]
set state id 1,state 8 A0: 0x800291f8
set state id 1,state 1 A0: 0x80025b4c
set state id 1,state 1 A0: 0x8001d0a9
set state id 1,state 36 A0: 0x8002101e
set state id 1,state 31 A0: 0x8002101e
set state id 1,state 1 A0: 0x8001ce74
set state id 1,state 41 A0: 0x80020290
set state id 1,state 42 A0: 0x8001fe0c
set state id 1,state 44 A0: 0x8001fd98
I (151419) BT_AV: bt_app_av_sm_hdlr state: 4, event: 0xff00
I (161419) BT_AV: bt_app_av_sm_hdlr state: 4, event: 0xff00
I (171419) BT_AV: bt_app_av_sm_hdlr state: 3, event: 0xff00
I (171419) BT_AV: a2dp connecting to peer: f8:23:45:44:23:43
detach pid 1,reason 34 A0: 0x8002aaf6
set state id 1,state 25 A0: 0x80020424
set state id 1,state 27 A0: 0x8001f8d0
set state id 1,state 0 A0: 0x8002f471
W (172269) BT_L2CAP: L2CAP got sec_comp for unknown BD_ADDR

W (172269) BT_HCI: hcif disc complete: hdl 0x80, rsn 0x8
W (172269) BT_APPL: bta_dm_act no entry for connected service cbs
W (172279) BT_BTC: BTA_AV_OPEN_EVT::FAILED status: 3

I (172279) BT_AV: bt_app_av_sm_hdlr state: 4, event: 0x0
```

shangke1112, 2023-06-05:
According to this log ,our device send LMP_START_ENCRYPTION_REQ ,but the other device do not send LMP_ACCEPT or LMP_NOT_ACCEPT, so our devce timed out .
This lib is used to confirm whether the other device supports encryption, please try again.
[libbtdm4.tar.gz](https://github.com/espressif/esp-idf/files/11651110/libbtdm4.tar.gz)

espradio, 2023-06-12:
Hi
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7132
ho 0 tail 12 room 4
load:0x40078000,len:15672
ho 0 tail 12 room 4
load:0x40080400,len:4
load:0x40080404,len:3884
entry 0x40080650
I (33) boot: ESP-IDF v5.2-dev-917-g37d11ea04d-dirty 2nd stage bootloader
I (33) boot: compile time Jun 12 2023 15:16:38
I (35) boot: Multicore bootloader
I (39) boot: chip revision: v3.0
I (43) boot.esp32: SPI Speed      : 40MHz
I (48) boot.esp32: SPI Mode       : DIO
I (52) boot.esp32: SPI Flash Size : 2MB
I (57) boot: Enabling RNG early entropy source...
I (62) boot: Partition Table:
I (66) boot: ## Label            Usage          Type ST Offset   Length
I (73) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (81) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (88) boot:  2 factory          factory app      00 00 00010000 00100000
I (96) boot: End of partition table
I (100) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=2aa6ch (174700)                                                                                                                                                              map
I (171) esp_image: segment 1: paddr=0003aa94 vaddr=3ffbdb60 size=04574h ( 17780)                                                                                                                                                              load
I (179) esp_image: segment 2: paddr=0003f010 vaddr=40080000 size=01008h (  4104)                                                                                                                                                              load
I (181) esp_image: segment 3: paddr=00040020 vaddr=400d0020 size=a7680h (685696)                                                                                                                                                              map
I (433) esp_image: segment 4: paddr=000e76a8 vaddr=40081008 size=17490h ( 95376)                                                                                                                                                              load
I (485) boot: Loaded app from partition at offset 0x10000
I (485) boot: Disabling RNG early entropy source...
I (497) cpu_start: Multicore app
I (497) cpu_start: Pro cpu up.
I (497) cpu_start: Starting app cpu, entry point is 0x40081220
I (0) cpu_start: App cpu up.
I (517) cpu_start: Pro cpu start user code
I (517) cpu_start: cpu freq: 160000000 Hz
I (517) cpu_start: Application information:
I (522) cpu_start: Project name:     a2dp_source
I (527) cpu_start: App version:      v5.2-dev-917-g37d11ea04d-dirty
I (534) cpu_start: Compile time:     Jun 12 2023 15:16:34
I (540) cpu_start: ELF file SHA256:  10d5a4caa...
I (546) cpu_start: ESP-IDF:          v5.2-dev-917-g37d11ea04d-dirty
I (553) cpu_start: Min chip rev:     v0.0
I (557) cpu_start: Max chip rev:     v3.99
I (562) cpu_start: Chip rev:         v3.0
I (567) heap_init: Initializing. RAM available for dynamic allocation:
I (574) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (580) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (586) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (593) heap_init: At 3FFCBC30 len 000143D0 (80 KiB): DRAM
I (599) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (605) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (612) heap_init: At 40098498 len 00007B68 (30 KiB): IRAM
I (619) spi_flash: detected chip: gd
I (622) spi_flash: flash io: dio
W (626) spi_flash: Detected size(16384k) larger than the size in the binary imag                                                                                                                                                             e header(2048k). Using the size in the binary image header.
I (640) coexist: coex firmware version: ebddf30
I (645) app_start: Starting scheduler on CPU0
I (649) app_start: Starting scheduler on CPU1
I (649) main_task: Started on CPU0
I (659) main_task: Calling app_main()
I (679) BTDM_INIT: BT controller compile version [5caffe9]
set state id 2,state 0 A0: 0x8001ad29
set state id 0,state 0 A0: 0x8004ef19
set state id 6,state 127 A0: 0x800447a8
set state id 262,state 127 A0: 0x800447a8
[... 649 lines cut ...]
slmp0 10
set state id 1,state 8 A0: 0x800291f8
I (122689) BT_AV: event: 16
set state id 1,state 1 A0: 0x80025b4c
set state id 1,state 1 A0: 0x8001d0a9
slmp0 77
slmp0 5
slmp0 81
slmp0 120
slmp0 255
slmp0 255
set state id 1,state 36 A0: 0x8002101e
slmp0 22
set state id 1,state 31 A0: 0x8002101e
set state id 1,state 1 A0: 0x8001ce74
slmp0 30
set state id 1,state 41 A0: 0x80020290
slmp0 32
set state id 1,state 42 A0: 0x8001fe0c
slmp0 34
set state id 1,state 44 A0: 0x8001fd98
I (131429) BT_AV: bt_app_av_sm_hdlr state: 4, event: 0xff00
I (141429) BT_AV: bt_app_av_sm_hdlr state: 4, event: 0xff00

```

shangke1112, 2023-06-14:
this log:
detach pid 1,reason 19 A0: 0x8001f310 .
Reason 19 means the remote device send a disconnect to our device ,the previous log has a value of 34.
It seems that these two test result are different. May I ask if there is any  issues with other devices ?

espradio, 2023-06-15:
Any other logs will help?

shangke1112, 2023-06-19:
What I mean is there  any issues using the "BT_Car" wiht other devices ?
The simplest solution to this problem is to caption packets to analysis, do you have a device for captuing packets? It is difficult to determine the behavior of the other device through our decice's log.

espradio, 2023-06-24:
>What I mean is there any issues using the "BT_Car" wiht other devices ?
No.
>do you have a device for captuing packets?
Unfortunately  I dont have

shangke1112, 2023-06-26:
Modify the macro "SC_MODE_INCLUDED" to  "TRUE"  in the file bt_target.h , this will change the encryption mode .
#define SC_MODE_INCLUDED TRUE

Also ,please try a few more times and send the log to me.

xiongweichao, 2024-07-23:
Hi @espradio ,

May I ask if this issue still exists?

Thanks

closed 2024-08-08 by Sherry616