espressif/esp-idf #11409: correct placement of the _ATTR

franz-ms-muc, 2023-05-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi, 

here is a question about the memory defines:

The GCC documentation says the following: ( https://gcc.gnu.org/onlinedocs/gcc/Common-Variable-Attributes.html )
```
struct duart a __attribute__ ((section ("DUART_A"))) = { 0 };
struct duart b __attribute__ ((section ("DUART_B"))) = { 0 };
char stack[10000] __attribute__ ((section ("STACK"))) = { 0 };
int init_data __attribute__ ((section ("INITDATA")));
```
--> So after the variable.
In your code, the define is before the variable...

see here: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/memory-types.html#rtc-slow-memory

The question is, what is better/right, or what should I take?

idea--list, 2023-05-18:
One more question from my side: Should we place those attributes in the declarations or in the definitions?
I am asking that because on IDF 4.x i got used to put such declarations both in the declaration and also in the definition of  a function that needed some extra care. However IDF 5.0.1 is always showing up a warning because of that and now i am not confident where are we intended to place those attibutes?

KaeLL, 2023-05-18:
@franz-ms-muc https://gcc.gnu.org/onlinedocs/gcc/Attribute-Syntax.html#Type-Attributes-1

@idea--list https://gcc.gnu.org/onlinedocs/gcc/Function-Attributes.html
> Function attributes are introduced by the __attribute__ keyword in the declaration of a function

diplfranzhoepfinger, 2023-07-12:
@igrr your thougts ?