espressif/esp-idf #11410: i2c falling edge read

CC-MASTER05, 2023-05-16:
### Is your feature request related to a problem?

Hello,

I have implement I2C protocol on LTC2497 chip but i require to read in falling edge SCLK but i am not able to found any configuration for that can you suggest or give me idea how can i do?

I am using I2C protocol with SCL and SDA.

Also in attachment i am attaching my I2C signal as well as datasheet instruction for reading also i am attaching my code also.

Idf version: 5.0.1
Communication: I2C
Refrence Voltage : 3.3 V
Pullup : SCL SDA both line in 

```
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "driver/i2c.h"

#define I2C_PORT I2C_NUM_0
#define I2C_ADDR 0x14

void i2c_master_init()
{
    i2c_config_t i2c_config = {
        .mode = I2C_MODE_MASTER,
        .sda_io_num = GPIO_NUM_1,
        .scl_io_num = GPIO_NUM_2,
        .sda_pullup_en = GPIO_PULLUP_ENABLE,
        .scl_pullup_en = GPIO_PULLUP_ENABLE,
        .master.clk_speed = 100000,
    };
    i2c_param_config(I2C_PORT, &i2c_config);
    i2c_driver_install(I2C_PORT, I2C_MODE_MASTER, 0, 0, 0);
}

void app_main()
{
    i2c_master_init();

    uint8_t buffer[3];
    while (1)
    {
        // Send the start condition
        i2c_cmd_handle_t cmd = i2c_cmd_link_create();
        i2c_master_start(cmd);

        // Send the I2C address of the LTC2497 for write operation
        i2c_master_write_byte(cmd, (I2C_ADDR << 1) | I2C_MASTER_WRITE, true);

        // Send the command byte to configure the LTC2497 for single-ended measurement on channel
        i2c_master_write_byte(cmd, 0xB0, true);

        // Send the repeated start condition to switch to read mode
        i2c_master_start(cmd);

        // Send the I2C address of the LTC2497 for read operation
        i2c_master_write_byte(cmd, (I2C_ADDR << 1) | I2C_MASTER_READ, true);
        vTaskDelay(1 / portTICK_PERIOD_MS);
        // Read the three data bytes from the LTC2497
        i2c_master_read(cmd, buffer, 3, I2C_MASTER_LAST_NACK);

        // Send the stop condition
        i2c_master_stop(cmd);

        // Execute the I2C transaction
        i2c_master_cmd_begin(I2C_PORT, cmd, pdMS_TO_TICKS(1000));
        i2c_cmd_link_delete(cmd);

        // Extract the voltage data from the buffer
        int32_t adc_value = ((buffer[0] << 24) | (buffer[1] << 16) | (buffer[2] << 8) | buffer[3]) >> 8;

        adc_value >>= 2;
        adc_value &= 0x3FFFFF;
        // Calculate the voltage in volts (Vref = 3.3V)
        float voltage = (float)(adc_value * 3.3) / 8388608;

        // Print the voltage
        printf("Voltage: %.2f V\n", voltage);
        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }
}
```
#### Datasheet instruction
[image]

#### Output Signal

[image]

#### Datasheet

[Datasheet](https://www.analog.com/media/en/technical-documentation/data-sheets/2497fb.pdf)

mythbuster5, 2023-05-17:
I don't think read data in falling edge in I2C obeys to official protocol. So this I2C driver can't configure that stuff...perhaps.

o-marshmallow, 2023-05-17:
Hi @CC-MASTER05 ,

As @mythbuster5 , reading during a low-phase of SCL clock does **not** comply with the I2C specification.
Out of curiosity, why do you need to read SDA line when SCL is low? The datasheet of the LTC2497 you linked clearly states that the device is I2C-compliant and, as for the official specification, even put the emphasis on SCL being low to change SDA:
[image]

In short, you can simplify your code to the following:
```
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "driver/i2c.h"

#define I2C_PORT I2C_NUM_0
#define I2C_ADDR 0x14

esp_err_t i2c_master_init()
{
    i2c_config_t i2c_config = {
        .mode = I2C_MODE_MASTER,
        .sda_io_num = GPIO_NUM_1,
        .scl_io_num = GPIO_NUM_2,
        .sda_pullup_en = GPIO_PULLUP_ENABLE,
        .scl_pullup_en = GPIO_PULLUP_ENABLE,
        .master.clk_speed = 100000,
    };
    i2c_param_config(I2C_PORT, &i2c_config);
    return i2c_driver_install(I2C_PORT, I2C_MODE_MASTER, 0, 0, 0);
}

void app_main()
{
    esp_err_t err = i2c_master_init();
    if (err != ESP_OK) {
        printf("Error init I2C\n");
        return;
    } 
    uint8_t reg = 0xB0;
    uint8_t buffer[3];

    while (1)
    {
        err = i2c_master_write_read_device(I2C_PORT, I2C_ADDR, &reg, 1, buffer, sizeof(buffer));
        if (err != ESP_OK) {
            printf("Error %x occurred during the I2C transfer\n", err);
            continue;
        }

        // Extract the voltage data from the buffer
        int32_t adc_value = ((buffer[0] << 24) | (buffer[1] << 16) | (buffer[2] << 8) | buffer[3]) >> 8;

        adc_value >>= 2;
        adc_value &= 0x3FFFFF;
        // Calculate the voltage in volts (Vref = 3.3V)
        float voltage = (float)(adc_value * 3.3) / 8388608;

        // Print the voltage
        printf("Voltage: %.2f V\n", voltage);
        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }
}

```

CC-MASTER05, 2023-05-17:
Hello @o-marshmallow ,

Actually i want to read my LTC2497 chip and i am not able to read those chip so i have raise a query on Analog Digital Inc and as per those suggestion i am reading wrong edge on so they comment me this 
```
I see some mismatch in your timing diagram. According to datasheet timing diagram in the page 15 data will be shifted out on the falling edge of the SCLK. But in your timing diagram data shifted out on the rising edge.
```
After this comment i have read all the I2C document but i am not able to read this way if you able to help it will be good.

Thanks,
Fenil

o-marshmallow, 2023-05-17:
@CC-MASTER05 I am still not sure to understand the problem honestly. The logic analyzer results you gave doesn't show any error, on the contrary, we can see that the slave device shifts the bits at the right moment: when SCL is low.
[image]

closed 2023-06-26 by fenilGhoghari