espressif/esp-idf #11412: Linker does not link constructors of instances located in files without linked function

timoxd7, 2023-05-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.3.4

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

Creating an Instance of a Class in global scope should call its constructor prior `app_main()` execution.
This should always be true, even if inside the file containing the Instance is no other function.

### What is the actual behavior?

If there is a Function inside the .cpp file that is somehow linked to the app_main() (throu direct call or a call to a function nested in other functions) this behaviour is correct. Otherwise, if there is only the global Instance inside the .cpp file, the constructor will never be called and the instance will never be created.

### Steps to reproduce.

Please have a look at this repo: [https://github.com/timoxd7/ConstructorLinkageBug](https://github.com/timoxd7/ConstructorLinkageBug)

There is a config inside the main component called `MAIN_CPP_FORCE_COMPILATION`. This will only add a call to an `empty_function()` defined inside the .cpp file. It does not add anything else.

If you then use the size command, you can see that only the main.c.obj is produced, while with this config enabled, there are main.c.obj, main.cpp.obj and custom_class.cpp.obj produced.

igrr, 2023-05-16:
Hi @timoxd7, please check the explanation of the issue here: https://github.com/espressif/esp-idf/tree/master/examples/build_system/cmake/plugins#ensuring-that-the-plugin-code-is-included-into-the-executable.

The reason why you observe this difference between direct `gcc` invocation and IDF build system is that your gcc invocation links object files directly into the executable. IDF build system first combines the object files into static libraries and then links them. The logic of GNU linker when dealing with objects in static libraries is different from that when object files are linked directly. The example README linked above has some suggestions on how to avoid this issue.

(Do note that the `WHOLE_ARCHIVE` option mentioned in that example has been added in IDF v5.0, and you won't be able to use it in v4.3.x.)

timoxd7, 2023-05-30:
Thank you, did not know this is expected behaviour. I will close the issue then.

closed 2023-05-30 by timoxd7