espressif/esp-idf #11413: ESP-C2/ESP8684 vs ESP32 I2C timeout error on writing more than 16bytes to a slave device

AndreasKohn, 2023-05-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

IDF 5.0.2

### Operating System used.

Windows

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP8684 DevKitM1 v1.1

### Power Supply used.

USB

### What is the expected behavior?

I have implemented code on a ESP32 platform which uses the I2C bus to communicate with a slave device. This works perfectly stable and reliable. Now I moved the code to a ESP8694 platform and encounter I2C bus timeouts. I would expect to run the same code on the ESP8684 without any issues as well

### What is the actual behavior?

Instead of the expected behavior, the function i2c_master_write_to_device() returns with a timeout failure if I send more than 16bytes in one function call (register address + 16bytes payload = 17bytes).
The function works as expected if I reduce the payload by one byte (register address + 15bytes payload = 16bytes).
Reading bytes from the slave with more than 16 bytes seems to work as expected.
It seems that in the faulty case the SCL line stops clocking, so that the slave device does not remove the ACK signal.

I actually isolated the problem from my code and used a modified Espressif example (i2c_simple).
The example now reads 16 bytes from a slave register, then writes 15 bytes to a slave register and then writes 16 bytes to a slave register. The last operation fails. Except the order of the function calls which I changed for the example, I just changed the main clock to 24Mhz for the ESP8684 devKit.
 
Attached are some LA screenshots that show the working and the not working transmission as well as details on the ACK phase of the last written byte.  

Here on the left the successful transfer of 15 bytes payload, on the right the failing transfer of 16 bytes:

[image]

Detail of the ACK phase for the successful case, with the clock signal to remove the ACK by the slave:

[image]

Detail of the ACK phase for the unsuccessful case, where the clock transition to remove the ACK by the slave is missing. The marker show the expected clock signal change:

[image]

Here the code that can be used to reproduce the problem:

```
#include <stdio.h>
#include "esp_log.h"
#include "driver/i2c.h"

static const char *TAG = "i2c-simple-example";

#define I2C_MASTER_SCL_IO           CONFIG_I2C_MASTER_SCL      /*!< GPIO number used for I2C master clock */
#define I2C_MASTER_SDA_IO           CONFIG_I2C_MASTER_SDA      /*!< GPIO number used for I2C master data  */
#define I2C_MASTER_NUM              0                          /*!< I2C master i2c port number, the number of i2c peripheral interfaces available will depend on the chip */
#define I2C_MASTER_FREQ_HZ          400000                     /*!< I2C master clock frequency */
#define I2C_MASTER_TX_BUF_DISABLE   0                          /*!< I2C master doesn't need buffer */
#define I2C_MASTER_RX_BUF_DISABLE   0                          /*!< I2C master doesn't need buffer */
#define I2C_MASTER_TIMEOUT_MS       1000

#define R3916_SENSOR_ADDR           0x50        			   /*!< Slave address of the ST25R3916 */
#define R3916_WHO_AM_I_REG_ADDR     0x7F        			   /*!< Register addresses of the "who am I" register */

#define R3916_1ST_REG_ADDR          0x00        		  	   /*!< Register addresses of 1st register */

/**
 * @brief Read a sequence of bytes from a MPU9250 sensor registers
 */
static esp_err_t r3916_register_read(uint8_t reg_addr, uint8_t *data, size_t len)
{
    return i2c_master_write_read_device(I2C_MASTER_NUM, R3916_SENSOR_ADDR, &reg_addr, 1, data, len, I2C_MASTER_TIMEOUT_MS / portTICK_PERIOD_MS);
}

/**
 * @brief Write a byte to a MPU9250 sensor register
 */
static esp_err_t r3916_register_write(uint8_t* dataBuf, uint8_t len)
{
    return i2c_master_write_to_device(I2C_MASTER_NUM, R3916_SENSOR_ADDR, dataBuf, len, I2C_MASTER_TIMEOUT_MS / portTICK_PERIOD_MS);
}

/**
 * @brief i2c master initialization
 */
static esp_err_t i2c_master_init(void)
{
    int i2c_master_port = I2C_MASTER_NUM;

    i2c_config_t conf = {
        .mode = I2C_MODE_MASTER,
        .sda_io_num = I2C_MASTER_SDA_IO,
        .scl_io_num = I2C_MASTER_SCL_IO,
        .sda_pullup_en = GPIO_PULLUP_ENABLE,
        .scl_pullup_en = GPIO_PULLUP_ENABLE,
        .master.clk_speed = I2C_MASTER_FREQ_HZ,
    };

    i2c_param_config(i2c_master_port, &conf);

    return i2c_driver_install(i2c_master_port, conf.mode, I2C_MASTER_RX_BUF_DISABLE, I2C_MASTER_TX_BUF_DISABLE, 0);
}

void app_main(void)
{
    ESP_ERROR_CHECK(i2c_master_init());
    ESP_LOGI(TAG, "I2C initialized successfully");

    /* Demonstrate reading chunk of data with success, register address plus 16 bytes of payload */
    uint8_t dataBufR[16];
    ESP_ERROR_CHECK(r3916_register_read(R3916_1ST_REG_ADDR | 0x40, dataBufR, sizeof(dataBufR)));
    ESP_LOGI(TAG, "16 byte successfully read ");

    /* Demonstrate writing a chunk of data with success, register address plus 15 bytes of payload */
    uint8_t dataBufW1[] = {R3916_1ST_REG_ADDR, 0xA5, 0xA5, 0xA5, 0xA5, 0xA5, 0xA5, 0xA5, 0xA5, 0xA5, 0xA5, 0xA5, 0xA5, 0xA5, 0xA5, 0xA5 };
    ESP_ERROR_CHECK(r3916_register_write(dataBufW1, sizeof(dataBufW1)));
    ESP_LOGI(TAG, "15 byte successfully written ");

    /* Demonstrate writing a chunk of data with failure (timout), register address plus 16 bytes of payload */
    uint8_t dataBufW2[] = {R3916_1ST_REG_ADDR, 0xA5, 0xA5, 0xA5, 0xA5, 0xA5, 0xA5, 0xA5, 0xA5, 0xA5, 0xA5, 0xA5, 0xA5, 0xA5, 0xA5, 0xA5, 0xA5 };
    ESP_ERROR_CHECK(r3916_register_write(dataBufW2, sizeof(dataBufW2)));
    ESP_LOGI(TAG, "16 byte successfully written ");

    ESP_ERROR_CHECK(i2c_driver_delete(I2C_MASTER_NUM));
    ESP_LOGI(TAG, "I2C de-initialized successfully");
}

```

### Steps to reproduce.

Simply execute the example code on a ESP8694 devKit with an appropriate I2C slave attached.
You should see  ESP_ERROR_CHECK macro raising an exception.

### Debug Logs.

```plain
see LA screenshots in the description above
```

### More Information.

Tried it with two different I2C slave devices always with the same faulty outcome. Applied other fixes that are recommended in other tickets with respect to I2C problems with no improvement. (the forced stop on timeout, timing adjustments)

AndreasKohn, 2023-05-16:
Here in addition the terminal output of the example program:
[image]

o-marshmallow, 2023-05-17:
Hello @AndreasKohn ,

The version of IDF you are using had a bug in the SDA sample timing. It may not be linked but to make sure this isn't your problem can you please try using a more recent version, like latest master (https://github.com/espressif/esp-idf/commit/54576b7528b182256c027de86eb605a172bc2821) ? Or cherry-pick this commit a0f8434f9382016db6f3b102051569b5cbaea2c0.

Another potential problem could be the slope of your signals, I feel you aren't using external pull-ups. When clocking the bus at 400KHz, you must be using external pull-up resistors (ideally 2.2Kohm or less, but no more than 4.7Kohm), as the internal pull-ups are too weak.

I just tested on my side, with a C2 and 4.7Kohm extern pull-ups, and commit (https://github.com/espressif/esp-idf/commit/54576b7528b182256c027de86eb605a172bc2821), I could write 14, 15, 16 and even 32 bytes in a row, so at the moment I would exclude the software bug in the driver state machine as the cause:
[image]

AndreasKohn, 2023-05-17:
Hi @o-marshmallow,
thanks for your quick response. Here is what I did in the meantime based on your suggestions:

- pull-ups on SDA and SCL are already in place (2.4k), I removed the internal pull-ups for the GPIO configuration, but this did not change the behavior.
- I patched my IDF version with modification you have suggested (https://github.com/espressif/esp-idf/commit/a0f8434f9382016db6f3b102051569b5cbaea2c0), this also did not change the behavior.
- I reduced the bus speed to 100KHz, also without any positive impact.
- finally I switched to the master branch, latest version as stated above and this removed the issue.

My application that is running fine on the ESP32 is still not fully functional on the ESP-C2 but at the moment I see this more in my area of the SW.
Thanks for your help to solve the issue described above. In the end it seems to be an issue in the IDF and not the HW, but there might be also other changes in the baseline that fixed the problem.
In case I still find issues related  to the IDF, I will come back with a new ticket.

o-marshmallow, 2023-05-18:
@AndreasKohn Thanks for the feedback! Closing the issue then.

closed 2023-05-18 by o-marshmallow