espressif/esp-idf #11416: Issue with RGB Endian Setting in ST7789 Driver causing wrong order of colors

giddeonzeix, 2023-05-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello,

I've encountered an issue related to the ST7789 LCD driver.

According to the ST7789 datasheet (https://www.waveshare.com/w/upload/a/ae/ST7789_Datasheet.pdf), section 9.1.28 MADCTL (36h): Memory Data Access Control, bit D3 is used to declare the RGB/BGR order.

However, in the ESP-IDF LCD component (specifically in components/esp_lcd/include/esp_lcd_panel_vendor.h line 22), this bit is used to set the RGB data endian:
lcd_color_rgb_endian_t rgb_endian; /*!< Set RGB data endian: RGB or BGR */

This seems to be a misunderstanding. The actual endian setting for the ST7789 is specified in the datasheet, section 9.2.1 RAMCTRL (B0h): RAM Control. The second parameter's bit D3 (ENDIAN) defines the order of the colors. If this bit is not set, the color order will be red, blue, green instead of the expected red, green, blue.

I was able to fix this issue by adding an extra initialization step after calling esp_lcd_panel_reset(panel_handle); and esp_lcd_panel_init(panel_handle);

// Extra initialization
esp_lcd_panel_io_tx_param(io_handle, RAMCTRL, (uint8_t[]){0x00, 0xE8}, 2);

Could you please add an extra flag for the RGB endian setting in the esp_lcd_panel_vendor.h file to correctly configure the color order for the ST7789 LCD?

Thank you for your attention to this matter.

suda-morris, 2023-05-17:
Hi @giddeonzeix I see. The ST7789 and many other LCDs like arranging the ram data in "big-endian" by default. 

[image]

In our example, we enable [this Kconfig option](https://github.com/espressif/esp-idf/blob/master/examples/peripherals/lcd/spi_lcd_touch/sdkconfig.defaults#L2), so the LVGL library can change the RAM data endian on its side. In applications that are not using LVGL library, I'm not sure if there're also such a configuration. But if not, yeah, your solution of overriding the `RAMCTRL` is also a good idea!

Note, the `RAMCTRL` is a vendor-specific register (see it's on the command table **2**). So we can't guarantee there will be such a feature on other LCDs.

giddeonzeix, 2023-05-17:
I apologize for the confusion. Let me provide a revised version of the text you shared:

I would like to clarify that my previous statement was not specifically related to LVGL. Instead, I developed an LVGL driver for the LILYGO T-Display S3 using IDF 5.01 and the ESP_LCD component. In this implementation, I utilized the I80 parallel bus interface and the ST7789 panel. Before incorporating LVGL, I conducted a preliminary test to ensure the driver's functionality by using the `esp_lcd_panel_draw_bitmap` function (refer to the C function below). This test aimed to verify color accuracy and other aspects.

During this testing phase, I discovered that the default configuration of the ESP_LCD using the ST7789 panel on the ESP32 platform has the incorrect endianness. The color order is incorrect, particularly with regards to the placement of the green color. Regardless of whether the color order is RGB or BGR, green should appear in the middle. While it is possible to address this issue by adding another layer to correct the color order, it would be more appropriate to modify the default configuration to ensure correct operation.

Based on my research across various versions of the ST7789 panel (VW, V, VI, S, etc.), all of them have the RAMCTRL register at 0xB0, which includes an ENDIAN bit (D3). Moreover, the ESP_LCD component already includes a vendor-specific ST7789 panel. I have not encountered any datasheets for the ST7789 that do not include the RAMCTRL register.

I hope this revised explanation provides clearer information regarding your situation.

```c
#define RGB565_Blue 0x001F
#define RGB565_Green 0x07E0
#define RGB565_Red 0xF800

void fill_color_test()
{
    uint16_t *picture = (uint16_t *)malloc(DISP_BUF_SIZE_BYTES);
    memset(picture, 0, DISP_BUF_SIZE_BYTES);

    for (int y = 0; y < 50; ++y)
    {
        int ofs1 = y * EXAMPLE_LCD_H_RES;
        int ofs2 = (y + 50) * EXAMPLE_LCD_H_RES;
        int ofs3 = (y + 100) * EXAMPLE_LCD_H_RES;
        for (int x = 0; x < EXAMPLE_LCD_H_RES; ++x)
        {
            picture[x + ofs1] = RGB565_Red;
            picture[x + ofs2] = RGB565_Green;
            picture[x + ofs3] = RGB565_Blue;
        }
    }
    esp_lcd_panel_draw_bitmap(panel_handle, 0, 0, EXAMPLE_LCD_H_RES, EXAMPLE_LCD_V_RES, (uint16_t *)picture);
    vTaskDelay(pdMS_TO_TICKS(1000));
}
```

closed 2023-09-08 by espressif-bot

gongfan99, 2025-04-04:
I just found the ENDIAN bit (D3) of `RAMCTRL` register may not work as we expected. 
I have an esp23s3 development board (https://www.waveshare.com/wiki/ESP32-S3-Touch-LCD-1.69). It has the ST7789 driver. The IDF I use is ESP-IDF v5.4.0

**The only settings that work correctly about the color is the following:**
1. invert color: `esp_lcd_panel_invert_color(panel_handle, true)`
2. in `esp_lcd_panel_dev_config_t`, set `rgb_ele_order` to `LCD_RGB_ELEMENT_ORDER_RGB`
3. in `esp_lcd_panel_dev_config_t`, set `data_endian` to `LCD_RGB_DATA_ENDIAN_BIG` (this is the default so can be omitted)
4. in lvgl's flush callback, use the function `lv_draw_sw_rgb565_swap` to swap bytes

Because the function `lv_draw_sw_rgb565_swap` had some overhead, I wanted to elinimate it. Therefore, I changed setting#3 and #4:  
3. in `esp_lcd_panel_dev_config_t`, set `data_endian` to `LCD_RGB_DATA_ENDIAN_LITTLE`
4. in lvgl's flush callback, do not use the function `lv_draw_sw_rgb565_swap`

In theory this would work. But the color messed up after the change: red becomes blue, green becomes red, blue becomes green.

I looked at the esp_lcd source code (https://github.com/espressif/esp-idf/blob/release/v5.4/components/esp_lcd/src/esp_lcd_panel_st7789.c) It appears the code is correctly setting D3 of `RAMCTRL` if `LCD_RGB_DATA_ENDIAN_LITTLE` is chosen. So I really don't know where it went wrong.

@giddeonzeix were you able to use `LCD_RGB_DATA_ENDIAN_LITTLE` to avoid using `lv_draw_sw_rgb565_swap`?