espressif/esp-idf #11417: 在用IDF编译I2S时报错

murarduino, 2023-05-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

4.4.4

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

使用IDF编译I2S，I2S配置如下：
 const int I2S_PORT_NUM = I2S_NUM_0;

  i2s_config_t i2s_config_DAC = {
      .mode = (i2s_mode_t)(I2S_MODE_MASTER | I2S_MODE_TX),
      .sample_rate = 22050,
      .bits_per_sample = I2S_BITS_PER_SAMPLE_16BIT, /*为何不是8bit?*/
      .channel_format = I2S_CHANNEL_FMT_ONLY_RIGHT, /*为何选择只是右通道?*/
      .communication_format = (i2s_comm_format_t)(I2S_COMM_FORMAT_STAND_I2S),
      .intr_alloc_flags = ESP_INTR_FLAG_LEVEL1,
      .dma_buf_count = 16,
      .dma_buf_len = 256,
      .use_apll = true};

  i2s_pin_config_t pin_config = {
      .bck_io_num = I2S_BCLK,
      .ws_io_num = I2S_LRC,
      .data_out_num = I2S_DOUT,
      .data_in_num = I2S_PIN_NO_CHANGE};

编译提示错误：
missing initializer for member 'i2s_driver_config_t::tx_desc_auto_clear' [-Wmissing-field-initializers]
missing initializer for member 'i2s_pin_config_t::mck_io_num' [-Wmissing-field-initializers]

### What is the actual behavior?

原先用Arduino架构编译时，可以正常通过。

### Steps to reproduce.

1. 尝试添加两个成员
 const int I2S_PORT_NUM = I2S_NUM_0;

  i2s_config_t i2s_config_DAC = {
      .mode = (i2s_mode_t)(I2S_MODE_MASTER | I2S_MODE_TX),
      .sample_rate = 22050,
      .bits_per_sample = I2S_BITS_PER_SAMPLE_16BIT, /*为何不是8bit?*/
      .channel_format = I2S_CHANNEL_FMT_ONLY_RIGHT, /*为何选择只是右通道?*/
      .communication_format = (i2s_comm_format_t)(I2S_COMM_FORMAT_STAND_I2S),
      .intr_alloc_flags = ESP_INTR_FLAG_LEVEL1,
      .dma_buf_count = 16,
      .dma_buf_len = 256,
      .use_apll = true,
      .tx_desc_auto_clear = false};

  i2s_pin_config_t pin_config = {
      .bck_io_num = I2S_BCLK,
      .ws_io_num = I2S_LRC,
      .data_out_num = I2S_DOUT,
      .data_in_num = I2S_PIN_NO_CHANGE,
      .mck_io_num = I2S_PIN_NO_CHANGE};

报错：
D:/Esp-Doc/ESP-decoder/components/audio_mix/audio_mix.cpp:80:34: warning: missing initializer for member 'i2s_driver_config_t::fixed_mclk' [-Wmissing-field-initializers]
       .tx_desc_auto_clear = false};
                                  ^
D:/Esp-Doc/ESP-decoder/components/audio_mix/audio_mix.cpp:80:34: warning: missing initializer for member 'i2s_driver_config_t::mclk_multiple' [-Wmissing-field-initializers]
D:/Esp-Doc/ESP-decoder/components/audio_mix/audio_mix.cpp:80:34: warning: missing initializer for member 'i2s_driver_config_t::bits_per_chan' [-Wmissing-field-initializers]
D:/Esp-Doc/ESP-decoder/components/audio_mix/audio_mix.cpp:87:38: warning: missing initializer for member 'i2s_pin_config_t::mck_io_num' [-Wmissing-field-initializers]
       .mck_io_num = I2S_PIN_NO_CHANGE};
                                      ^
D:/Esp-Doc/ESP-decoder/components/audio_mix/audio_mix.cpp:87:38: error: designator order for field 'i2s_pin_config_t::mck_io_num' does not match declaration order in 'i2s_pin_config_t'
ninja: build stopped: subcommand failed.

L-KAYA, 2023-05-17:
这是因为C++编译器会要求初始化结构体中所有字段，并且还要保证字段顺序相同，而C编译器会将未赋值的变量自动设为默认值0。

因为IDF中不保证结构体在C++中bit的兼容性，需要手动解决这个问题。可根据`i2s.h`中`i2s_config_t`结构体的定义，按顺序依次给所有字段赋值即可，相对于当前已初始化的字段，其他新增字段赋值为0即可。

如：
```c
i2s_config_t i2s_config_DAC = {
    .mode = (i2s_mode_t)(I2S_MODE_MASTER | I2S_MODE_TX),
    .sample_rate = 22050,
    .bits_per_sample = I2S_BITS_PER_SAMPLE_16BIT, 
    .channel_format = I2S_CHANNEL_FMT_ONLY_RIGHT, 
    .communication_format = (i2s_comm_format_t)(I2S_COMM_FORMAT_STAND_I2S),
    .intr_alloc_flags = ESP_INTR_FLAG_LEVEL1,
    .dma_buf_count = 16,
    .dma_buf_len = 256,
    .use_apll = true,
    .tx_desc_auto_clear = false,
    .fixed_mclk = 0,
    .mclk_multiple = 0,
    .bits_per_chan = 0
};

i2s_pin_config_t pin_config = {
    .mck_io_num = I2S_PIN_NO_CHANGE, 
    .bck_io_num = I2S_BCLK,
    .ws_io_num = I2S_LRC,
    .data_out_num = I2S_DOUT,
    .data_in_num = I2S_PIN_NO_CHANGE
};
```

L-KAYA, 2023-05-17:
是的，要加在第一行，顺序必须相同，不然还是会报错
[image]

murarduino, 2023-05-18:
> 是的，要加在第一行，顺序必须相同，不然还是会报错 [image]

好的，再次感谢指导。

closed 2023-05-18 by murarduino