espressif/esp-idf #11419: ESP32-S2 Deep sleep power consumption question

DylanGWork, 2023-05-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi all,

I have a phantom power spike during the deep sleep mode which I cannot figure out, it appears to be happening every 30ms.
It's very small and doesn't impact my overall system much, but I'd like to get to the bottom of it.
This is what it looks like on the nrf power profiler:
[image]
The spikes either side of the grey area is the co-processor polling some sensors ever 100ms at about 3mA, the unknown spikes in the shaded grey area reach about 400uA and last for 10ms and occur about every 30ms.
I've run an oscilloscope over all GPIO's in case it's my hardware and I can't find anything there.

I have attempted to make a skeleton version of the code using the Risc-v GPIO example and removing the GPIO polling and the while loop in co-processor and added a short timer to prevent crashing but the example holds at around 1.5mA which is a little confusing, so I haven't been able to create a sample project to share.

Two questions:
Has anyone tested a stable low power current on the ESP32-S2?
Is there an example of a low power mode with NO functionality to test the lower most limits of the ESP32-S2 current draw (so I can search for power leakages).

Cheers,
Dylan

esp-wzh, 2023-05-17:
Hi~ @DylanGWork  I can't tell what module leakage you have encountered since I can't use a example to reproduce the problem,
you can check the typical value of the current in each sleep mode in section 4.6.2 of the [esp32-s2_datasheet_en](https://www.espressif.com/sites/default/files/documentation/esp32-s2_datasheet_en.pdf)

I think the [ulp_adc example](https://github.com/espressif/esp-idf/tree/release/v5.1/examples/system/ulp_riscv/adc) in IDF is more similar to your application, this is the current graph I measured on esp32s2 using IDF release/v5.1 branch (commit_id:[6cfc6f53b](https://github.com/espressif/esp-idf/commit/6cfc6f53beb9ef0800e8ec534babc603a8836177))

[image]
(* *I adjusted the measurement interval to 2s*)

It's base current is about 30uA when not measuring, I don't see power spikes during sleep. You can refer to this exapmle to rule out influencing factors.

DylanGWork, 2023-05-20:
Hi, 
Thank you for measuring the deepsleep current, very helpful. I'm getting about 31uA at the bottom limit which is very close to your measurement. 
I'm using the ESP32-S2-Mini-2U.

closed 2023-06-20 by Alvin1Zhang