espressif/esp-idf #11420: esp32s3 interface usb camera

Nadeemranm, 2023-05-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Espressif  ESP-IDF  supporting  the all types usb cameras or this is support only some particalur models usb camera ?

tore-espressif, 2023-06-05:
Hello @Nadeemranm ,
USB cameras implementations differ for various models and vendors. At this time, we do not claim support for 'all' USB cameras on the market. You can experiment with out [UVC example](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/host/uvc):, that uses [usb_host_uvc component](https://components.espressif.com/components/espressif/usb_host_uvc) or you can have a look at example of [usb_stream component](https://components.espressif.com/components/espressif/usb_stream) which also adds support for audio streaming.

If you could share more information about you camera and planned usage, we might be able to help more.
Thanks!

Nadeemranm, 2023-06-05:
Usb camera lcd display code not  are the campile
when i do the run usb camera lcd display
I have faced this type of error
error msg is no version of idf match >=4.4.1,<4.5.0
and project depends on idf (=4.4),project requires idf(>=4.4.0<4.4.1>
so because project depends on the idf (>=4.4.1) ,version solving failed
pls help me to resolve this error
Thank you!!

tore-espressif, 2023-06-05:
Could you please update your IDF to v5 and try again?

Nadeemranm, 2023-06-05:
Sir i am already using v5.0 version
In vs code i have downloaded the Espressif idf extension
but still same error!
I have already campile usb camera wifi transfer code campile successfully

roma-jam, 2023-06-06:
Hi @Nadeemranm, 

Sorry for the delay with the answers. 
Before start, let me clarify some moments. 

Here is the issue opened in esp-idf related project. 
The one example that you can try to launch with an external USB camera is here: https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/host/uvc. 

Have you tried to launch it and do you have any error regarding its compiling procedure?

Nadeemranm, 2023-06-07:
Sir i am working on usb_camera lcd_display
https://github.com/espressif/esp-iot-solution/tree/usb/add_usb_solutions/examples/usb/host/usb_camera_lcd_display
Whenever i compile this code me getting error msg is no version of idf
match >=4.4.1,<4.5.0
and project depends on idf (=4.4),project requires idf(>=4.4.0<4.4.1>
so because project depends on the idf (>=4.4.1) ,version solving failed
Whereas i am using the v5.0.0 version of esp-idf
What is this  problem ? and in code where i should have to be change ?

roma-jam, 2023-06-07:
Hi @Nadeemranm, 

Thank you for the explanation, I see and I am able to reproduce the error. Thanks a lot for your cooperation. 

Before I proceed, may I ask for some extra information. 
You pointed that you try to build esp-iot-solution example "usb_camera_lcd_display" from the branch usb/add_usb_solutions, not from the master branch of esp-iot-solutions. 
Are there any reason to use specifically this branch (and not the master) to build the example "usb_camera_lcd_display"?

Nadeemranm, 2023-06-07:
No sir!
I have just completed usb camera live streaming through on this
"examples/usb/host/usb_camera_wifi_transfer"
Now  i just want to be usb camera live streaming   displayed on my "ILI9486
DISPLAY  module  from this "examples/usb/host/usb_camera_lcd_display" but
this code not compiling
I can use any example code for usb camera to lcd display for display
interface controller like "i2c ,spi and i2s"
if you have another example code or option plz me suggestion me

roma-jam, 2023-06-07:
@Nadeemranm thanks for the additional information. 

Please, pay attention to the point that "examples/usb/host/usb_camera_wifi_transfer" is present only in branch usb/add_usb_solutions. 
But the example "examples/usb/host/usb_camera_lcd_display" can not be build from that branch due to error. 

If you want to build and use example "examples/usb/host/usb_camera_lcd_display" it is better to use master branch of esp-iot-solution. 

May I ask you to try compile this example from the master branch (https://github.com/espressif/esp-iot-solution/tree/master/examples/usb/host/usb_camera_lcd_display) ?

Nadeemranm, 2023-06-07:
Hello sir!
I am confuse for take to this topic master branch and branch
usb/add_usb_solutions
The branch  is related to the  version  esp-idf?
sir plz example something about  branch!!

roma-jam, 2023-06-07:
@Nadeemranm,

Not exactly. Here the branch is related not to the version esp-idf, but to the version esp-iot-solution (which is another repository). 
For better understanding of git branches you can easily find some information in Internet with specific request. 
For example, here is a good explanation of branches on git-scm: https://git-scm.com/book/en/v2/Git-Branching-Branches-in-a-Nutshell
But you may find any other, which would be more suitable for you. 

The main idea I am speaking about is that there two branches (we can say to simplification: two different versions) of esp-iot-solution: master branch and usb/add_usb_solutions branch. 
First one (master) has the examples/usb/host/usb_camera_lcd_display, which can be compiled, but doesn't has a usb_camera_wifi_transfer. 
Second one (usb/add_usb_solutions) has the usb_camera_wifi_transfer example, but the example usb_camera_lcd_display has an error during the compilation. 

I assume, that the correct question would be: how it is possible to use both examples 'usb_camera_lcd_display' and 'usb_camera_wifi_transfer' and which branch should be used for that.

Anyway, that is not related to esp-idf and its version so I suggest proceed this in esp-iot-solution related issue (which I guess is this one: https://github.com/espressif/esp-iot-solution/issues/266) and find a way to build both examples.

I will close this issue, as it is not related to esp-idf.
If there is anything else, feel free to reopen it.

closed 2023-06-07 by roma-jam

Nadeemranm, 2023-06-12:
Hello sir!!
Finally this code compile successfully and usb camera working properly
https://github.com/espressif/esp-iot-solution/tree/master/examples/usb/host/usb_camera_lcd_display
But i have problem ,i am confused , which lcd i should have to be use with
this code
https://github.com/espressif/esp-iot-solution/tree/master/examples/usb/host/usb_camera_lcd_display
i have "ILI9486 LCD " can use this with this code ?
pls suggestion me that lcd so compatible with code example
thank you

Nadeemranm, 2023-06-13:
???

lijunru-hub, 2023-06-14:
https://github.com/espressif/esp-iot-solution/tree/master/examples/usb/host/usb_camera_lcd_display
This routine only supports v5.0 and above, and must be burned with esp32-s3-lcd-ev-board development board. See readme.md for more details 

branch usb/add_usb_solutions The above example is very old and not recommended.

lijunru-hub, 2023-06-14:
If you want to use another screen driver, you need to rewrite the code by referring to the example code

Nadeemranm, 2023-06-14:
Can i use this "ILI9486" lcd driver with this example code?

Nadeemranm, 2023-06-16:
??

roma-jam, 2023-06-16:
Hi @Nadeemranm, 

Sorry for such a long response. This issue seems to be closed as it was related to the USB camera interface and building problem with the example usb_camera_lcd_display. 

Any questions about supported types of LCD, I would recommend to solve with opening a new issue, related to the LCD question in esp-iot-solution repo. 

Thanks.

Nadeemranm, 2023-06-16:
Hello sir!1
i am using this example code
https://github.com/0015/ESP32-S2-S3-Projects/tree/main/usb_camera_lcd_display
but code is not  compiling . i got this types of error "LoyanGFX could not
be found .this could be becouse : comonent name was misspelled , the
component was not added to the build the component has been moved to the
IDF component manger or has been removed and refactored into some other
compenent"
pls help
thank you !!

Nadeemranm, 2023-06-19:
??

roma-jam, 2023-06-19:
Hi @Nadeemranm, 

Seems that you try to add local component LoyanGFX.
For use components from the esp-idf as well as components from the local folder, please refer to the `Component Configuration` part of the esp-idf documentation. 
You can find all the information regarding the configuration here: https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/build-system.html#component-configuration

Thanks

Nadeemranm, 2023-06-20:
Hello sir!!
When i compile this code this types of error
  CMake Error at /home/nadeem/esp/esp-idf/tools/cmake/build.cmake:245
(message):
  Failed to resolve component 'LovyanGFX'.
Call Stack (most recent call first):
  /home/nadeem/esp/esp-idf/tools/cmake/build.cmake:281
(__build_resolve_and_add_req)
  /home/nadeem/esp/esp-idf/tools/cmake/build.cmake:574
(__build_expand_requirements)
  /home/nadeem/esp/esp-idf/tools/cmake/project.cmake:440 (idf_build_process)
  CMakeLists.txt:20 (project)
How to resolve this issue?
pls help

Thanks !
[image: WhatsApp Image 2023-06-20 at 11.07.26.jpeg]
I have attached the my project's component folder

Nadeemranm, 2023-06-22:
???

roma-jam, 2023-06-22:
Hi @Nadeemranm, 

If you want to add any other component and build it from the local folder it is better to read how to do that in documentation:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/build-system.html#component-configuration

If there is more questions about building process of the esp-iot-solution example, it is better to open a new ticket in esp-iot-solution regarding your questions. 

Thanks

Nadeemranm, 2023-08-10:
Hello sir !
good morning
I am using this example code "
https://github.com/0015/ESP32-S2-S3-Projects/tree/main/usb_camera_lcd_display
i am using ESP32-S3-DevkitC-1 with usb camera and ili9488 16 bit parallel
line interface usb camera working properly and lcd working properly , i
have problem usb camera data not printing on lcd why sir ?
usb camera working properly !
But live streaming not displaying the lcd ili9488
plz help me sir
I (0) cpu_start: App cpu up. I (210) cpu_start: Pro cpu start user code I
(210) cpu_start: cpu freq: 240000000 Hz I (210) cpu_start: Application
information: I (213) cpu_start: Project name: uvc_mjpeg_lcd_lovyangfx I
(220) cpu_start: App version: 1 I (224) cpu_start: Compile time: Aug 8 2023
15:34:59 <https://www.youtube.com/watch?v=4I8ohXMIhjs&t=43200s> I (230)
cpu_start: ELF file SHA256: bb092d8e3de5b093... I (236) cpu_start: ESP-IDF:
v5.0-541-g885e501d99-dirty I (243) cpu_start: Min chip rev: v0.0 I (248)
cpu_start: Max chip rev: v0.99 I (252) cpu_start: Chip rev: v0.1 I (257)
heap_init: Initializing. RAM available for dynamic allocation: I (264)
heap_init: At 3FC957D0 len 00053F40 (335 KiB): D/IRAM I (271) heap_init: At
3FCE9710 len 00005724 (21 KiB): STACK/DRAM I (277) heap_init: At 3FCF0000
len 00008000 (32 KiB): DRAM I (284) heap_init: At 600FE010 len 00001FF0 (7
KiB): RTCRAM I (290) spi_flash: detected chip: generic I (295) spi_flash:
flash io: qio W (299) spi_flash: Detected size(8192k) larger than the size
in the binary image header(4096k). Using the size in the binary image
header. I (312) sleep: Configure to isolate all GPIO pins in sleep state I
(318) sleep: Enable automatic switching of GPIO sleep configuration I (326)
cpu_start: Starting scheduler on PRO CPU. I (0) cpu_start: Starting
scheduler on APP CPU. I (341) gpio: GPIO[47]| InputEn: 0| OutputEn: 1|
OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 I (351) gpio: GPIO[48]|
InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 I
(727) UVC_STREAM: Isoc transfer mode I (727) UVC_STREAM: UVC Streaming
Config Succeed I (777) UVC_STREAM: UVC Streaming Starting I (807)
UVC_STREAM: Port=1 init succeed I (807) UVC_STREAM: Waiting USB Connection I
(9424) UVC_STREAM: line 367 HCD_PORT_EVENT_CONNECTION I (9474) UVC_STREAM:
Resetting Port I (9534) UVC_STREAM: Setting Port FIFO I (9534) UVC_STREAM:
Getting Port Speed I (9534) UVC_STREAM: Port speed = 1 I (9536) UVC_STREAM:
USB Speed: full-speed I (9541) UVC_STREAM: Set Device Addr = 1 I (9546)
UVC_STREAM: Set Device Addr Done I (9550) UVC_STREAM: get device desc I
(9557) UVC_STREAM: get device desc, actual_num_bytes:26 * Device descriptor
* bLength 18 bDescriptorType 1 bcdUSB 2.00 bDeviceClass 0xef bDeviceSubClass
0x2 bDeviceProtocol 0x1 bMaxPacketSize0 64 idVendor 0xab20 idProduct
0x1276 bcdDevice
0.00 iManufacturer 1 iProduct 2 iSerialNumber 3 bNumConfigurations 1 I
(9584) UVC_STREAM: get short config desc I (9591) UVC_STREAM: get config
desc, actual_num_bytes:16 W (9595) UVC_STREAM: Configuration descriptor
larger than control transfer max length I (9603) UVC_STREAM: Set Device
Configuration = 1 I (9609) UVC_STREAM: Set Device Configuration Done I
(9614) UVC_STREAM: SET_CUR Probe I (9621) UVC_STREAM: SET_CUR Probe Done I
(9623) UVC_STREAM: GET_CUR Probe bmHint: 0001 bFormatIndex: 1
bFrameIndex: 1 dwFrameInterval:
666666 wKeyFrameRate: 0 wPFrameRate: 0 wCompQuality: 0
wCompWindowSize: 0 wDelay:
0 dwMaxVideoFrameSize: 38400 dwMaxPayloadTransferSize: 944 dwClockFrequency:
0 bmFramingInfo: 0 bPreferredVersion: 0 bMinVersion: 0 bMaxVersion: 0
bInterfaceNumber:
0 I (9655) UVC_STREAM: GET_CUR Probe Done, actual_num_bytes:34 I (9662)
UVC_STREAM: SET_CUR COMMIT I (9668) UVC_STREAM: SET_CUR COMMIT Done I
(9670) UVC_STREAM: Set Device Interface = 1, Alt = 1 I (9676) UVC_STREAM:
Set Device Interface Done I (9681) UVC_STREAM: Sample processing task
started I (9687) UVC_STREAM: Camera Start Streaming E (9876) uvc_demo:
callback! frame_format = 7, seq = 1, width = 320, height = 240, length =
2825, ptr = 1070161944 E (9940) uvc_demo: callback! frame_format = 7, seq =
2, width = 320, height = 240, length = 2722, ptr = 1070161944 E (10008)
uvc_demo: callback! frame_format = 7, seq = 3, width = 320, height = 240,
length = 2720, ptr = 1070161944 E (10076) uvc_demo: callback! frame_format
= 7, seq = 4, width = 320, height = 240, length = 3148, ptr = 1070161944 E
(10140) uvc_demo: callback! frame_format = 7, seq = 5, width = 320, height
= 240, length = 3148, ptr = 1070161944 E (10208) uvc_demo: callback!
frame_format = 7, seq = 6, width = 320, height = 240, length = 3098, ptr =
1070161944 E (10276) uvc_demo: callback! frame_format = 7, seq = 7, width =
320, height = 240, length = 3234, ptr = 1070161944 E (10340) uvc_demo:
callback! frame_format = 7, seq = 8, width = 320, height = 240, length =
3220, ptr = 1070161944 E (10408) uvc_demo: callback! frame_format = 7, seq
= 9, width = 320, height = 240, length = 3178, ptr = 1070161944 E (10476)
uvc_demo: callback! frame_format = 7, seq = 10, width = 320, height = 240,
length = 3343, ptr = 1070161944 E (10540) uvc_demo: callback! frame_format
= 7, seq = 11, width = 320, height = 240, length = 3288, ptr = 1070161944 E
(10608) uvc_demo: callback! frame_format = 7, seq = 12, width = 320, height
= 240, length = 3265, ptr
Show less
Reply
<https://www.youtube.com/channel/UCp3Wjcz3F9thEm9TFfapgvA>

Nadeemranm, 2023-08-10:
Hello sir good morning
i am using this"
https://github.com/0015/ESP32-S2-S3-Projects/tree/main/usb_camera_lcd_display
"
i  am using ESP32-S3-DevkitC-1 and usb camera and ili9488 16 bit parallel
line interface lcd
my usb camera working properly  and lcd little bit working not properly i
means  usb camera live streaming not displaying on lcd
I (0) cpu_start: App cpu up.
I (210) cpu_start: Pro cpu start user code
I (210) cpu_start: cpu freq: 240000000 Hz
I (210) cpu_start: Application information:
I (213) cpu_start: Project name:     uvc_mjpeg_lcd_lovyangfx
I (220) cpu_start: App version:      1
I (224) cpu_start: Compile time:     Aug  9 2023 18:17:54
I (230) cpu_start: ELF file SHA256:  0926a7a732204901...
I (236) cpu_start: ESP-IDF:          v5.0-541-g885e501d99-dirty
I (243) cpu_start: Min chip rev:     v0.0
I (248) cpu_start: Max chip rev:     v0.99
I (252) cpu_start: Chip rev:         v0.1
I (257) heap_init: Initializing. RAM available for dynamic allocation:
I (264) heap_init: At 3FC958D0 len 00053E40 (335 KiB): D/IRAM
I (271) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (277) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (284) heap_init: At 600FE010 len 00001FF0 (7 KiB): RTCRAM
I (290) spi_flash: detected chip: generic
I (295) spi_flash: flash io: qio
W (299) spi_flash: Detected size(8192k) larger than the size in the binary
image header(4096k). Using the size in the binary image header.
I (312) sleep: Configure to isolate all GPIO pins in sleep state
I (318) sleep: Enable automatic switching of GPIO sleep configuration
I (326) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (341) gpio: GPIO[47]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0|
Pulldown: 0| Intr:0
I (590) UVC_STREAM: Isoc transfer mode
I (590) UVC_STREAM: UVC Streaming Config Succeed
I (640) UVC_STREAM: UVC Streaming Starting
I (670) UVC_STREAM: Port=1 init succeed
I (670) UVC_STREAM: Waiting USB Connection
I (920) UVC_STREAM: line 367 HCD_PORT_EVENT_CONNECTION
I (970) UVC_STREAM: Resetting Port
I (1030) UVC_STREAM: Setting Port FIFO
I (1030) UVC_STREAM: Getting Port Speed
I (1030) UVC_STREAM: Port speed = 1
I (1032) UVC_STREAM: USB Speed: full-speed
I (1037) UVC_STREAM: Set Device Addr = 1
I (1042) UVC_STREAM: Set Device Addr Done
I (1046) UVC_STREAM: get device desc
I (1053) UVC_STREAM: get device desc, actual_num_bytes:26
*** Device descriptor ***
bLength 18
bDescriptorType 1
bcdUSB 2.00
bDeviceClass 0xef
bDeviceSubClass 0x2
bDeviceProtocol 0x1
bMaxPacketSize0 64
idVendor 0xab20
idProduct 0x1276
bcdDevice 0.00
iManufacturer 1
iProduct 2
iSerialNumber 3
bNumConfigurations 1
I (1080) UVC_STREAM: get short config desc
I (1086) UVC_STREAM: get config desc, actual_num_bytes:16
W (1091) UVC_STREAM: Configuration descriptor larger than control transfer
max length
I (1099) UVC_STREAM: Set Device Configuration = 1
[... 79 lines cut ...]
height = 240, length = 6340, ptr = 1070162196
E (3104) uvc_demo: callback! frame_format = 7, seq = 27, width = 320,
height = 240, length = 6324, ptr = 1070162196
E (3172) uvc_demo: callback! frame_format = 7, seq = 28, width = 320,
height = 240, length = 6329, ptr = 1070162196
E (3236) uvc_demo: callback! frame_format = 7, seq = 29, width = 320,
height = 240, length = 6320, ptr = 1070162196
E (3304) uvc_demo: callback! frame_format = 7, seq = 30, width = 320,
height = 240, length = 6322, ptr = 1070162196
E (3372) uvc_demo: callback! frame_format = 7, seq = 31, width = 320,
height = 240, length = 6344, ptr = 1070162196
E (3436) uvc_demo: callback! frame_format = 7, seq = 32, width = 320,
height = 240, length = 6355, ptr = 1070162196
E (3504) uvc_demo: callback! frame_format = 7, seq = 33, width = 320,
height = 240, length = 6369, ptr = 1070162196
E (3572) uvc_demo: callback! frame_format = 7, seq = 34, width = 320,
height = 240, length = 6365, ptr = 1070162196
E (3636) uvc_demo: callback! frame_format = 7, seq = 35, width = 320,
height = 240, length = 6359, ptr = 1070162196
E (3704) uvc_demo: callback! frame_format = 7, seq = 36, width = 320,
height = 240, length = 6366, ptr = 1070162196
E (3772) uvc_demo: callback! frame_format = 7, seq = 37, width = 320,
height = 240, length = 6350, ptr = 1070162196
E (3836) uvc_demo: callback! frame_format = 7, seq = 38, width = 320,
height = 240, length = 6368, ptr = 1070162196