espressif/esp-idf #11421: Crash when calling esp_ota_end() with "Cache disabled but cached memory region accessed"

xBiohazardx, 2023-05-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.2-153-gbfe07cbef9

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board with esp32s3

### Power Supply used.

USB

### What is the expected behavior?

I expect the ota update to verify and throw an error if the received binary is invalid.

### What is the actual behavior?

It crashes on esp_ota_end call with "Cache disabled but cached memory region accessed".

### Steps to reproduce.

1. Use modbus library to receive program binary data.
2. Use esp_ota_write to write data to ota partition
3. Call esp_ota_end after all the data has been received

### Debug Logs.

```plain
D (309059) esp_image: reading image header @ 0x310000
D (309059) esp_image: image header: 0xe9 0x05 0x02 0x03 40375448
I (309069) esp_image: segment 0: paddr=00310020 vaddr=3c030020 size=1036ch ( 66412) map
D (309079) esp_image: free data page_count 0x000000fb
Guru Meditation Error: Core  / panic'ed (Cache disabled but cached memory region accessed).
Write back error occurred while dcache tries to write back to flash

Core  0 register dump:
PC      : 0x40056f6f  PS      : 0x00060434  A0      : 0x82019552  A1      : 0x3fca6750
A2      : 0x3fc9b3c4  A3      : 0x3c056040  A4      : 0x00000020  A5      : 0x3fc9b3d4
A6      : 0x203c2064  A7      : 0x00000033  A8      : 0x00000000  A9      : 0x3fca6720
A10     : 0x00000020  A11     : 0x000001ca  A12     : 0x600c0024  A13     : 0x0000000f
A14     : 0x00000010  A15     : 0x00000040  SAR     : 0x00000009  EXCCAUSE: 0x00000007
EXCVADDR: 0x00000000  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0x00000000

Backtrace: 0x40056f6c:0x3fca6750 |<-CORRUPTED

ELF file SHA256: 2565e40135f768ab

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0x29 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375539
0x40375539: esp_restart_noos_dig at /home/philipp/esp/esp-idf/components/esp_system/esp_system.c:64 (discriminator 1)

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x1694
load:0x403c9700,len:0xba0
load:0x403cc700,len:0x2d2c
entry 0x403c98e8
```

### More Information.

I think my issue is similar to this one: https://esp32.com/viewtopic.php?t=19141
I could not make the solutions provided in the post above work for me.

alisitsyn, 2023-05-19:
Hello @xBiohazardx,

The information in this issue is not enough to find the reason for your crash. Please you answer to below questions.
1. Please provide the sdkconfig file. Could you use core dump to save some critical information about crash. 
2. Please provide some example app to reproduce the issue and clarify your modbus library initialization.
3. Which esp-modbus component version is used in your application.
4. Attach the log file of your application with the debug verbosity enabled in the kconfig menu.

Thank you for information.

xBiohazardx, 2023-05-22:
Thank you for your fast reply.

I figured the issue myself out.
For anyone else having a similar issue:
I was calling _strtok_ with the version string from _esp_app_desc_t_ which changed the version string. So when esp_ota_end was using it, the string was NULL.
Copying the version string in a buffer and then calling _strtok_ on the buffer resolved the issue for me.

closed 2023-05-22 by xBiohazardx