espressif/esp-idf #11423: ESP-IDF not highlighting the right code

roblatour, 2023-05-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

For the release 5.1 esp-idf example: wifi - iwit found here:
https://github.com/espressif/esp-idf/blob/release/v5.1/examples/wifi/itwt/main/itwt.c

I have the trigger-enabled option set:

[image]

and it looks fine in the config file:

[image]

However, in the IDF in the code, it appears the wrong code is being highlighted based on its value:

[image]

Regardless, it seems to be processing correctly (I added the following as a validation):

[image]

So the problem seems to be with what the IDF is highlighting in terms of the code to be executed (as shown in the third screenshot above).

suda-morris, 2023-05-18:
I think it's about if you have set the **compileCommands** correctly in the [c_cpp_properties.json](https://code.visualstudio.com/docs/cpp/c-cpp-properties-schema-reference) file

roblatour, 2023-05-18:
I did not have an entry for "compileCommands" in the c_cpp_properties.json file.  

Regardless, I added one, but have the same results:
`
{
    "configurations": [
        {
            "name": "ESP-IDF",
            "compilerPath": "c:\\esp-r-5-1\\.espressif\\tools\\riscv32-esp-elf\\esp-12.2.0_20230208\\riscv32-esp-elf\\bin\\riscv32-esp-elf-gcc.exe",
			"compileCommands": "E:/Documents/esp32/WeatherStation/build/compile_commands.json",
            "includePath": [
                "${config:idf.espIdfPath}/components/**",
                "${config:idf.espIdfPathWin}/components/**",
                "${config:idf.espAdfPath}/components/**",
                "${config:idf.espAdfPathWin}/components/**",
                "${workspaceFolder}/**"
            ],
            "browse": {
                "path": [
                    "${config:idf.espIdfPath}/components",
                    "${config:idf.espIdfPathWin}/components",
                    "${config:idf.espAdfPath}/components/**",
                    "${config:idf.espAdfPathWin}/components/**",
                    "${workspaceFolder}"
                ],
                "limitSymbolsToIncludedHeaders": false
            },
            "configurationProvider": "ms-vscode.cmake-tools"
        }
    ],
    "version": 4
}
`

roblatour, 2023-05-18:
I think you are right!

closed 2023-05-18 by roblatour