espressif/esp-idf #11424: Flash encryption on ESP32-C3 with OTA as Access Point

prazsmarid, 2023-05-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello,
I want to configure the flash encryption on ESP32-C3 with OTA as Access Point, but I have problems with the encryption part.
Tehnically, I want to generate a pre-encrypted file of the firmware.
I have wrote the encryption key in the BLOCK_KEY0 (BLOCK4) with this command: espefuse.py -p COM35 burn_key BLOCK_KEY0 `cryptkey.bin XTS_AES_128_KEY --no-write-protect --no-read-protect`
I have tried to encrypt the flash file with this command: espsecure.py encrypt_flash_data --aes_xts --keyfile .\cryptkey.bin --`output cryptedflash.bin .\build\throughput_client_demo.bin --address 0x20000 `
Now if I try to upload the flash with the 
`idf.py build | idf.py -p COM35 flash`
or 
`idf.py build | idf.py -p COM35 encrypted-app-flash `
or
`idf.py build | idf.py -p COM35 encrypted-flash `
commands, the device is not booting anymore, it is bootloping everytime with these errors:
```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0x9 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403cfa8a
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5988,len:0x2ae4
load:0x403cc710,len:0x948
load:0x403ce710,len:0x487c
entry 0x403cc710
I (25) boot: ESP-IDF v5.0.1-dirty 2nd stage bootloader
I (25) boot: compile time 15:05:24
I (25) boot: chip revision: v0.4
I (27) boot.esp32c3: SPI Speed      : 80MHz
I (32) boot.esp32c3: SPI Mode       : DIO
I (37) boot.esp32c3: SPI Flash Size : 4MB
I (42) boot: Enabling RNG early entropy source...
I (47) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (65) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (73) boot:  2 phy_init         RF data          01 01 00011000 00001000
I (80) boot:  3 ota_0            OTA app          00 10 00020000 0015e000
I (88) boot:  4 ota_1            OTA app          00 11 00180000 0015e000
I (95) boot: End of partition table
I (99) esp_image: segment 0: paddr=00020020 vaddr=3c100020 size=3a578h (238968) map
I (144) esp_image: segment 1: paddr=0005a5a0 vaddr=3fc92e00 size=02c9ch ( 11420) load
I (146) esp_image: segment 2: paddr=0005d244 vaddr=40380000 size=02dd4h ( 11732) load
I (152) esp_image: segment 3: paddr=00060020 vaddr=42000020 size=fad18h (1027352) map
I (312) esp_image: segment 4: paddr=0015ad40 vaddr=40382dd4 size=0ffc8h ( 65480) load
I (330) boot: Loaded app from partition at offset 0x20000
I (330) boot: Checking flash encryption...
E (330) flash_encrypt: Flash Encryption cannot be enabled (CRYPT_CNT (0) is write protected)
E (339) boot: OTA app partition slot 0 is not bootable
E (345) esp_image: image at 0x180000 has invalid magic byte (nothing flashed here?)
E (353) boot: OTA app partition slot 1 is not bootable
E (359) boot: No bootable app partitions in the partition table
```
Here is the summary of the fuse bits.

```
PS E:\Software\Espressif\frameworks\esp-idf-v5.0.1\examples\_projects\RGB_BLE_ESP32> espefuse.py -p COM35 summary
espefuse.py v4.5
Connecting...
Detecting chip type... ESP32-C3

=== Run "summary" command ===
EFUSE_NAME (Block) Description  = [Meaningful Value] [Readable/Writeable] (Hex Value)
----------------------------------------------------------------------------------------
[... 121 lines cut ...]
SPI_PAD_CONFIG_Q (BLOCK1)                          SPI Q (D1) pad                                     = 0 R/W (0b000000)
SPI_PAD_CONFIG_D (BLOCK1)                          SPI D (D0) pad                                     = 0 R/W (0b000000)
SPI_PAD_CONFIG_CS (BLOCK1)                         SPI CS pad                                         = 0 R/W (0b000000)
SPI_PAD_CONFIG_HD (BLOCK1)                         SPI HD (D3) pad                                    = 0 R/W (0b000000)
SPI_PAD_CONFIG_WP (BLOCK1)                         SPI WP (D2) pad                                    = 0 R/W (0b000000)
SPI_PAD_CONFIG_DQS (BLOCK1)                        SPI DQS pad                                        = 0 R/W (0b000000)
SPI_PAD_CONFIG_D4 (BLOCK1)                         SPI D4 pad                                         = 0 R/W (0b000000)
SPI_PAD_CONFIG_D5 (BLOCK1)                         SPI D5 pad                                         = 0 R/W (0b000000)
SPI_PAD_CONFIG_D6 (BLOCK1)                         SPI D6 pad                                         = 0 R/W (0b000000)
SPI_PAD_CONFIG_D7 (BLOCK1)                         SPI D7 pad                                         = 0 R/W (0b000000)

Usb Config fuses:
DIS_USB_JTAG (BLOCK0)                              Disables USB JTAG. JTAG access via pads is control = False R/W (0b0)
                                                   led separately
DIS_USB_DEVICE (BLOCK0)                            Disables USB DEVICE                                = False R/W (0b0)
USB_EXCHG_PINS (BLOCK0)                            Exchanges USB D+ and D- pins                       = False R/W (0b0)
DIS_USB_SERIAL_JTAG_DOWNLOAD_MODE (BLOCK0)         Disables USB-Serial-JTAG download feature in UART  = False R/W (0b0)
                                                   download boot mode

Wdt Config fuses:
WDT_DELAY_SEL (BLOCK0)                             Selects RTC WDT timeout threshold at startup       = False R/W (0b0)
```

How can I fix the pre-encrypted flash uploading?
Thanks in advance.

AdityaHPatwardhan, 2023-05-23:
Hi @prazsmarid 
I think you need to perform two additional steps as below.
1) Write protect the flash encryption key in key block 0 ( BLOCK4)
It can be done with following command
```
espefuse.py -p $ESPPORT --chip esp32c3 write_protect_efuse BLOCK4
```
2) Enable flash encryption for the device
```
espefuse.py -p $ESPPORT burn_efuse SPI_BOOT_CRYPT_CNT 0x7
espefuse.py -p $ESPPORT write_protect_efuse SPI_BOOT_CRYPT_CNT
```

Please note you shall also need to burn flash encryption related security eFuses which are burned by the bootloader when flash encryption is enabled via the botloader

prazsmarid, 2023-05-23:
Dear @AdityaHPatwardhan,
Thank you for your reply.
It seems on the ESP32C3 which I have tried earlier I have write protected the SPI_BOOT_CRYPT_CNT byte.
I have got an another ESP32C3, I have tried these commands on it:
```
espefuse.py -p COM38 burn_key BLOCK_KEY0 newcryptkey.bin XTS_AES_128_KEY --no-write-protect --no-read-protect
espefuse.py -p COM38 burn_efuse SPI_BOOT_CRYPT_CNT
idf.py encrypted-flash monitor
```
Now when it boots it seems the flash is encrypted, but when I type the 
```
esptool.py -p COM38 -b 460800 read_flash 0 0x200000 flash_contents.bin
```
it reads the flash content without any kind of encryption, I can clearly see the Strings in the dump.
This is normal?

AdityaHPatwardhan, 2023-05-23:
Please note that flash won't be encrypted unless it is explicitly encrypted by someone.
When Flash encryption is enabled via the bootloader then bootloader encrypts all the partitions ( which are marked as encrypted) one by one when it enables the flash encryption.
In your case bootloader is not enabling flash encryption hence we need to write the encrypted partition.
You can encrypt the partition externally as you have the key available outside 
When you use the command `idf.py encrypted-flash` it would only apply `--encrypt` option while flashing the `bootloader ` and `application` image.
Other partitions would be flashed in plaintext.
Do you know the address that you are reading from belongs to which partition table entry ?

AdityaHPatwardhan, 2023-05-30:
Hi @prazsmarid 
Can you do the following 
1) ``` hexdump build/bootloader/bootloader.bin```
2) ``` esptool.py -p PORT read_flash 0x0 0xB000 bootloader_read.bin```
3) ``` hexdump bootloader_read.bin```

and share the output of command 1 and 3.

>
```
espefuse.py -p COM38 burn_key BLOCK_KEY0 newcryptkey.bin XTS_AES_128_KEY --no-write-protect --no-read-protect
espefuse.py -p COM38 burn_efuse SPI_BOOT_CRYPT_CNT
idf.py encrypted-flash monitor
```

In this case as you have enabled flash encryption. write-encrypted must write the flash in encrypted format and 
flash read from the esp32c3 device must be in encrypted format.

Please give complete logs of all the steps that you are following.

Thanks,
Aditya

prazsmarid, 2023-06-11:
Hello,
Here are the outputs:
1.  hexdump build/bootloader/bootloader.bin
`E:\Software\Espressif\frameworks\esp-idf-v5.0.1\examples\_projects\RGB_BLE_ESP32> hexdump build/bootloader/bootloader.bin
`000000  e9 03 02 2f 10 c7 3c 40 ee 00 00 00 05 00 03 03
000010  00 63 00 00 00 00 00 01 20 58 cd 3f a8 16 00 00
000020  ff ff ff ff 41 73 73 65 72 74 20 66 61 69 6c 65
000030  64 20 69 6e 20 25 73 2c 20 25 73 3a 25 64 20 28
000040  25 73 29 0d 0a 00 00 00 61 62 6f 72 74 28 29 20
000050  77 61 73 20 63 61 6c 6c 65 64 20 61 74 20 50 43
000060  20 30 78 25 30 38 78 0d 0a 00 00 00 62 6f 6f 74
000070  00 00 00 00 1b 5b 30 3b 33 31 6d 45 20 28 25 6c
000080  75 29 20 25 73 3a 20 6c 6f 61 64 20 70 61 72 74
000090  69 74 69 6f 6e 20 74 61 62 6c 65 20 65 72 72 6f
0000a0  72 21 1b 5b 30 6d 0a 00 20 69 73 20 6e 6f 74 20
0000b0  62 6f 6f 74 61 62 6c 65 00 00 00 00 1b 5b 30 3b
0000c0  33 31 6d 45 20 28 25 6c 75 29 20 25 73 3a 20 46
0000d0  61 63 74 6f 72 79 20 61 70 70 20 70 61 72 74 69
0000e0  74 69 6f 6e 25 73 1b 5b 30 6d 0a 00 1b 5b 30 3b
0000f0  33 31 6d 45 20 28 25 6c 75 29 20 25 73 3a 20 46
000100  61 63 74 6f 72 79 20 74 65 73 74 20 61 70 70 20
000110  70 61 72 74 69 74 69 6f 6e 25 73 1b 5b 30 6d 0a
000120  00 00 00 00 1b 5b 30 3b 33 31 6d 45 20 28 25 6c
000130  75 29 20 25 73 3a 20 4f 54 41 20 61 70 70 20 70
000140  61 72 74 69 74 69 6f 6e 20 73 6c 6f 74 20 25 64
000150  25 73 1b 5b 30 6d 0a 00 1b 5b 30 3b 33 32 6d 49
000160  20 28 25 6c 75 29 20 25 73 3a 20 4c 6f 61 64 65
000170  64 20 61 70 70 20 66 72 6f 6d 20 70 61 72 74 69
000180  74 69 6f 6e 20 61 74 20 6f 66 66 73 65 74 20 30
000190  78 25 78 1b 5b 30 6d 0a 00 00 00 00 1b 5b 30 3b
0001a0  33 31 6d 45 20 28 25 6c 75 29 20 25 73 3a 20 45
0001b0  72 72 6f 72 20 69 6e 20 77 72 69 74 65 5f 6f 74
0001c0  61 64 61 74 61 20 6f 70 65 72 61 74 69 6f 6e 2e
0001d0  20 65 72 72 20 3d 20 30 78 25 78 1b 5b 30 6d 0a
0001e0  00 00 00 00 1b 5b 30 3b 33 32 6d 49 20 28 25 6c
0001f0  75 29 20 25 73 3a 20 53 65 74 20 61 63 74 75 61
000200  6c 20 6f 74 61 5f 73 65 71 3d 25 64 20 69 6e 20
000210  6f 74 61 64 61 74 61 5b 30 5d 1b 5b 30 6d 0a 00
000220  1b 5b 30 3b 33 32 6d 49 20 28 25 6c 75 29 20 25
000230  73 3a 20 44 69 73 61 62 6c 69 6e 67 20 52 4e 47
000240  20 65 61 72 6c 79 20 65 6e 74 72 6f 70 79 20 73
000250  6f 75 72 63 65 2e 2e 2e 1b 5b 30 6d 0a 00 00 00
000260  44 52 4f 4d 00 00 00 00 1b 5b 30 3b 33 31 6d 45
000270  20 28 25 6c 75 29 20 25 73 3a 20 49 6d 61 67 65
000280  20 63 6f 6e 74 61 69 6e 73 20 6d 75 6c 74 69 70
000290  6c 65 20 25 73 20 73 65 67 6d 65 6e 74 73 2e 20
0002a0  4f 6e 6c 79 20 74 68 65 20 6c 61 73 74 20 6f 6e
0002b0  65 20 77 69 6c 6c 20 62 65 20 6d 61 70 70 65 64
0002c0  2e 1b 5b 30 6d 0a 00 00 49 52 4f 4d 00 00 00 00
0002d0  1b 5b 30 3b 33 31 6d 45 20 28 25 6c 75 29 20 25
0002e0  73 3a 20 62 6f 6f 74 6c 6f 61 64 65 72 5f 6d 6d
0002f0  61 70 28 30 78 25 78 2c 20 30 78 25 78 29 20 66
000300  61 69 6c 65 64 1b 5b 30 6d 0a 00 00 52 46 20 64
000310  61 74 61 00 74 65 73 74 20 61 70 70 00 00 00 00
000320  57 69 46 69 20 64 61 74 61 00 00 00 4e 56 53 20
000330  6b 65 79 73 00 00 00 00 65 66 75 73 65 00 00 00
000340  55 6e 6b 6e 6f 77 6e 20 64 61 74 61 00 00 00 00
000350  4f 54 41 20 64 61 74 61 00 00 00 00 55 6e 6b 6e
000360  6f 77 6e 20 61 70 70 00 4f 54 41 20 61 70 70 00
000370  66 61 63 74 6f 72 79 20 61 70 70 00 75 6e 6b 6e
000380  6f 77 6e 00 1b 5b 30 3b 33 31 6d 45 20 28 25 6c
000390  75 29 20 25 73 3a 20 46 61 69 6c 65 64 20 74 6f
0003a0  20 76 65 72 69 66 79 20 70 61 72 74 69 74 69 6f
0003b0  6e 20 74 61 62 6c 65 1b 5b 30 6d 0a 00 00 00 00
0003c0  1b 5b 30 3b 33 32 6d 49 20 28 25 6c 75 29 20 25
0003d0  73 3a 20 50 61 72 74 69 74 69 6f 6e 20 54 61 62
0003e0  6c 65 3a 1b 5b 30 6d 0a 00 00 00 00 1b 5b 30 3b
0003f0  33 32 6d 49 20 28 25 6c 75 29 20 25 73 3a 20 23
000400  23 20 4c 61 62 65 6c 20 20 20 20 20 20 20 20 20
000410  20 20 20 55 73 61 67 65 20 20 20 20 20 20 20 20
[... 3420 lines cut ...]
008a70  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
008a80  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
008a90  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
008aa0  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
008ab0  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
008ac0  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
008ad0  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
008ae0  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
008af0  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
008b00  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
008b10  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
008b20  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
008b30  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
008b40  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
008b50  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
008b60  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
008b70  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
008b80  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
008b90  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
008ba0  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
008bb0  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
008bc0  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
008bd0  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
008be0  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
008bf0  ff ff ff ff ff ff ff

AdityaHPatwardhan, 2023-06-12:
Hi @prazsmarid I saw both binaries for the bootloader. They are as expected. As you can see the data read from flash i.e. ( bootloader_read.bin) is encrypted and does not contain useful information if you dont have the key. This is expected behaviour of the flash encryption.

> Now when it boots it seems the flash is encrypted, but when I type the
> ```
>    esptool.py -p COM38 -b 460800 read_flash 0 0x200000 flash_contents.bin
>```
> it reads the flash content without any kind of encryption, I can clearly see the Strings in the dump.
This is normal?

Does this resolve your query?
As we have witnessed, Flash contents are encrypted and cannot be read in plaintext once they are written in encrypted format.

prazsmarid, 2023-06-12:
Dear @AdityaHPatwardhan ,
Thank you for the help, since this time the flash encryption is working, I not know why it not worked at the first try, but now everything is good.
Now I have an another question:
Until the flash encryption was not enabled, the **esp_ota_write** function worked, but since the firmware file is encrypted, the magic byte is changed. Do you know how to disable magic byte check in the esp_ota_write function?
This is the OTA code that I use:

> esp_err_t update_post_handler(httpd_req_t *req)
> {
> 	char buf[1000];
> 	esp_ota_handle_t ota_handle;
> 	int remaining = req->content_len;
> 	const esp_partition_t *ota_partition = esp_ota_get_next_update_partition(NULL);
> 	ESP_ERROR_CHECK(esp_ota_begin(ota_partition, OTA_SIZE_UNKNOWN, &ota_handle));
> 
> 	while (remaining > 0) {
> 		int recv_len = httpd_req_recv(req, buf, MIN(remaining, sizeof(buf)));
> 
> 		// Timeout Error: Just retry
> 		if (recv_len == HTTPD_SOCK_ERR_TIMEOUT) {
> 			continue;
> 
> 		// Serious Error: Abort OTA
> 		} else if (recv_len <= 0) {
> 			httpd_resp_send_err(req, HTTPD_500_INTERNAL_SERVER_ERROR, "Protocol Error");
> 			return ESP_FAIL;
> 		}
> 
> 		// Successful Upload: Flash firmware chunk
> 		if (esp_ota_write(ota_handle, (const void *)buf, recv_len) != ESP_OK) {
> 			httpd_resp_send_err(req, HTTPD_500_INTERNAL_SERVER_ERROR, "Flash Error");
> 			return ESP_FAIL;
> 		}
> 
> 		remaining -= recv_len;
> 	}
> 
> 	// Validate and switch to new OTA image and reboot
> 	if (esp_ota_end(ota_handle) != ESP_OK || esp_ota_set_boot_partition(ota_partition) != ESP_OK) {
> 			httpd_resp_send_err(req, HTTPD_500_INTERNAL_SERVER_ERROR, "Validation / Activation Error");
> 			return ESP_FAIL;
> 	}
> 
> 	httpd_resp_sendstr(req, "Firmware update complete, rebooting now!\n");
> 
> 	vTaskDelay(500 / portTICK_PERIOD_MS);
> 	esp_restart();
> 
> 	return ESP_OK;
> }

and this is the error that I get when I try to upload the encrypted firmware file which I made using the 
`PS` E:\Software\Espressif\frameworks\esp-idf-v5.0.1\examples\_projects\RGB_BLE_ESP32> espsecure.py encrypt_flash_data --aes_xts --keyfile .\newcryptkey.bin --address 0x20000 --output my-app-ciphertext.bin .\build\throughput_client_demo.bin

> E (57940) esp_ota_ops: OTA image has invalid magic byte (expected 0xE9, saw 0x99)
> W (57940) httpd_txrx: httpd_resp_send_err: 500 Internal Server Error - Flash Error
> W (57940) httpd_uri: httpd_uri: uri handler execution failed

AdityaHPatwardhan, 2023-06-13:
Hi @prazsmarid 
The OTA partition should not be encrypted.
When flash encryption is enabled, the OTA still should be sent to the device in plaintext format.
The device would then apply flash encryption while writing it to the flash.
Please refer to the [simple OTA example](https://github.com/espressif/esp-idf/tree/master/examples/system/ota/simple_ota_example) which works fine when flash encryption has been enabled for the device.

If you want to send the OTA image on cloud in encrypted format then there is a different example called [pre_encrypted_ota](https://github.com/espressif/esp-idf/tree/master/examples/system/ota/pre_encrypted_ota). Please note that pre-encrypted OTA is not related to flash encryption in any way. It uses different encryption algorithm (RSA) for encrypting the OTA image.

AdityaHPatwardhan, 2023-06-23:
Hi @prazsmarid Any update?

AdityaHPatwardhan, 2023-06-27:
Hi @prazsmarid I will go forward and close the issue, If you still face any problem please feel free to re-open the issue.
Thanks,
Aditya

closed 2023-06-27 by AdityaHPatwardhan