espressif/esp-idf #11425: [ESP32-S2] DAC channels sometimes swap in DAC_CHANNEL_MODE_ALTER mode

railbox, 2023-05-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-4784-g6cfc6f53be

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32-S2

### Power Supply used.

USB

### What is the expected behavior?

- The DAC channels should not be swapped in DAC_CHANNEL_MODE_ALTER mode when using dac_continuous_write. 

### What is the actual behavior?

Sometimes if i reset CPU during DAC DMA processing the DAC channels are swapped. Rebooting the CPU one or several times resolve the problem.

### Steps to reproduce.

1. Configure 2 DAC channels in DAC_CHANNEL_MODE_ALTER mode
2. Start DAC output by dac_continuous_write -> channels are not swapped
3. Reboot CPU several times till swapping occur.

### Debug Logs.

```plain
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
```

anadezhd, 2023-05-22:
I will try to do it, but not it the low priority as I`m switching to the I2S DAC any way.

Qumbetlian, 2023-06-30:
Hi, here is a code that reproduce this bug not only at restarting, but even while running.
[esp32s2-channel-swapping.zip](https://github.com/espressif/esp-idf/files/11923144/esp32s2-channel-swapping.zip)
The swapping is occurring when the data buffer are rewritten by the function dac_continuous_write_cyclically(). If you connect the pins GPIO17 and GPIO18  to an oscilloscope, you can see that:

[image]

Qumbetlian, 2023-07-02:
There is another way to produce two different signal on esp32-s2 DAC's pins? Can someone point me to a simple working example?

Qumbetlian, 2023-07-02:
It seems that if you use the source clock DAC_DIGI_CLK_SRC_ALL, channels swapping happens less often, but it happens.

L-KAYA, 2023-07-05:
Have looked into this issue, it seems due to the ADC/DAC module is not reset on existing, you can try to append `DPORT_APB_SARADC_RST` to [system_internal.c:50](https://github.com/espressif/esp-idf/blob/release/v5.1/components/esp_system/port/soc/esp32s2/system_internal.c#L50C85-L50C85).

The bug will be fixed in release/v5.1 branch soon

Qumbetlian, 2023-07-10:
Hi L-KAYA. I have a suggestion for a change to be made in dac_continuous_write_cyclically function. These lines to be removed:
_~~uint32_t split = 1;~~_
_~~/* To spread data more averagely, average the last two descriptors */~~_
_~~split = (buf_size * DAC_16BIT_ALIGN_COEFF < handle->cfg.buf_size * 2) ? 3 - split : 1;~~_
This line:
_size_t load_bytes = s_dac_load_data_into_buf(handle, handle->bufs[i], handle->cfg.buf_size, buf, ~~buf_size / split~~);_
to be changed in:
_size_t load_bytes = s_dac_load_data_into_buf(handle, handle->bufs[i], handle->cfg.buf_size, buf, handle->cfg.buf_size);_
This may be helpfull when you want to use an on_convert_done callback function that do something usualy at regular intervals within signal cycle.

L-KAYA, 2023-07-11:
Thanks for your kind suggestion! But it might not the best solution to generate the callback per cycle.

The main concern of spreading data more averagely is based on following:
1. There must be at least two descriptors to link as a ring, so that the DMA controller will loop this ring infinitely. This is the basic logic behind the function `dac_continuous_write_cyclically`. So we can't mount the buffer to one descriptor even it is short enough to be held by one descriptor.
2. As you know, the interrupt only triggers when a descriptor is finished sending by DMA, and as we have multiple descriptors in `cyclical` mode, the interrupt will be triggered multiple times as well. If we do want a callback when finished one cycle, the best way might be counting the interrupt times in the ISR and callback when descriptors finished one round.
3. To make sure the last block of buffer is not too short. If the buffer that mounted is too short, the interrupt will be triggered very soon, which is possible to miss this interrupt when haven't finished dealing with the last interrupt.

If you think the callback for one cycle done is necessary, maybe we can create a new feature request issue.

L-KAYA, 2023-07-11:
Update the state about the `channel swap` bug
1. Have found the solution to fix the issue for the case `restart the chip`, we need to reset `SARADC` module together with `SPI3_DMA` module.
2. Have not found the solution to fix the issue for that case `DMA stopped and restart`. The channels are possible to swap after the DMA stopped, have tried all kinds of reset, but did not work.

L-KAYA, 2023-07-14:
Not yet, the bug has not totally been fixed, still need further test to find out a workaround or something

Qumbetlian, 2023-07-14:
Regarding my proposal of change in the function dac_continuous_write_cyclically, what actually want to say, is not to use just one descriptor, more descriptors are good, but I want to say that would be better that all the descriptors to have in their buffers the same number of bytes. So that, on_convert_done callback function to be called at regular interval of time.

Qumbetlian, 2023-07-14:
Now, using this code: 
[main.zip](https://github.com/espressif/esp-idf/files/12051713/main.zip)

I get on oscilloscope this pattern: 
[image]

Qumbetlian, 2023-07-14:
I modified the I did modifed the function dac_continuous_write_cyclically as in my proposal previously posted and now I have the correct pattern:
[image]

abouatef, 2024-12-26:
is this still open ? And can it happen when using a single DAC ?

dhalbert, 2025-03-25:
I have also encountered this bug, which is still present in v5.3.2. I tried resetting various things such as the DMA using the private .h files, but found that only a full reset by calling  `dac_continuous_disable(handle);  dac_continuous_del_channels(handle);` and then calling `dac_continuous_new_channels()` allow over again fixed the problem.

frak0d, 2025-04-14:
i am facing the same issue but with non-cyclic `dac_continuous_write` and workaround suggested by @dhalbert didn't work for me.

Narukara, 2026-11-13:
Hi everyone,

I've recently been working on a DAC stereo example and encountered a similar issue. I believe I've identified the root cause of the channel swapping issue on ESP32-S2 when using synchronous DAC output with alternating channels (`dac_continuous_write` & `DAC_CHANNEL_MODE_ALTER`).

The issue stems from the DAC's internal channel alternation mechanism, which continues switching between channels even when no data is available. Since the DMA data stream can arrive at any time, it may not be synchronized with the current channel alternation state.

I'm planning to address this issue in the near future.