espressif/esp-idf #11429: [TWAI] Filter mask problem

mickeyl, 2023-05-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.2-153-gbfe07cbef9

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32C3-DEVKIT-C1

### Power Supply used.

USB

### What is the expected behavior?

With the following code:

```c
  twai_general_config_t g_config = TWAI_GENERAL_CONFIG_DEFAULT((gpio_num_t)CONFIG_TWAI_TX, (gpio_num_t)CONFIG_TWAI_RX, TWAI_MODE_NORMAL);
  twai_timing_config_t t_config = TWAI_TIMING_CONFIG_500KBITS();
  twai_filter_config_t f_config = {
      .acceptance_code = (uint32_t)0x18DAF900 << 3,
      .acceptance_mask = (uint32_t(TWAI_EXTD_ID_MASK) << 3),
      .single_filter = true,
  };

```

I would expect only packets with arbitration of 0x18DAF900 to pass the controller filters.

### What is the actual behavior?

Packets with arbitration 0x640 are coming through.

### Steps to reproduce.

```c
    twai_general_config_t g_config = TWAI_GENERAL_CONFIG_DEFAULT((gpio_num_t)CONFIG_TWAI_TX, (gpio_num_t)CONFIG_TWAI_RX, TWAI_MODE_NORMAL);
    twai_timing_config_t t_config = TWAI_TIMING_CONFIG_500KBITS();
    twai_filter_config_t f_config = {
        .acceptance_code = (uint32_t)0x18DAF900 << 3,
        .acceptance_mask = (uint32_t(TWAI_EXTD_ID_MASK) << 3),
        .single_filter = true,
    };

    //Install TWAI driver
    if (twai_driver_install(&g_config, &t_config, &f_config) == ESP_OK) {
        printf("Driver installed\n");
    } else {
        printf("Failed to install driver\n");
        return;
    }

    //Start TWAI driver
    if (twai_start() == ESP_OK) {
        printf("Driver started\n");
    } else {
        printf("Failed to start driver\n");
        return;
    }

    while (true) {
        twai_message_t message;
        if (twai_receive(&message, pdMS_TO_TICKS(10000)) == ESP_OK) {
            printf("Message received\n");
        } else {
            printf("Failed to receive message\n");
            return;
        }

        //Process received message
        if (message.extd) {
            printf("Message is in Extended Format\n");
        } else {
            printf("Message is in Standard Format\n");
        }
        printf("ID is %08X\n", (unsigned int)message.identifier);
        if (!(message.rtr)) {
            for (int i = 0; i < message.data_length_code; i++) {
                printf("Data byte %d = %2X\n", i, message.data[i]);
            }
        }
    }
```

### Debug Logs.

```plain
Message received
Message is in Extended Format
ID is 18DAF900
Data byte 0 = 12
Data byte 1 = 34
Message received
Message is in Standard Format
ID is 00000640
```

Dazza0, 2023-06-13:
@mickeyl 

`acceptance_mask` bits sets which bits of the code are don't care. If we are matching exactly `0x18DAF900`, then all those bits should be `0` in the mask. Does the following fix the issue:

```c
.acceptance_mask = (uint32_t) ~((TWAI_EXTD_ID_MASK) << 3),
```

wanckl, 2025-07-07:
@mickeyl  Hi, filter config also have different format in new driver, and I think have more clear guide here

https://docs.espressif.com/projects/esp-idf/en/latest/esp32p4/api-reference/peripherals/twai.html#filter-configuration