espressif/esp-idf #11431: Cant upload bootloader after secure boot and flash encrypt

hdpklm, 2023-05-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.2-dev-24-gf404fe96b1-dirty

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

Hello.

I encrypted the flash on the esp32 and enabled secure boot, in development mode

everything works fine except the spiffs partition

The esp32 can't find the spiffs partition, I try to change the flag on the partition table to encrypt and without the encrypt flag, but nothing changes.

so i modify the spiffs spec in the bootloader and flash the new bootloader. but in this case the esp32 cannot read anything.
```sh
esptool.py --chip esp32 write_flash --encrypt 0x00000 ./build/bootloader/bootloader.bin --force
```

I think the bootloader is not encrypted while flashing, or some other problem, I don't know what's going on.
I get this message in the terminal
```sh
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57
```
I search for this error and its means, the bootloader its not encrypt, but I use this command `--encrypt`  to encrypt then its write in flash
With the esp32-s3, after enabling secure boot mode, and encrypting the flash, I can flash many bootloaders without any problem.
 

### What is the actual behavior?

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

### Steps to reproduce.

...

AdityaHPatwardhan, 2023-05-19:
The default offset for the bootloader in case of esp32 is 0x1000.
`esptool.py --chip esp32 write_flash --encrypt 0x1000 ./build/bootloader/bootloader.bin --force`

KaeLL, 2023-05-19:
@AdityaHPatwardhan 
> v1 is 0x000

?

hdpklm, 2023-05-20:
Hello

after read a lot of issues and documentation, the conclusion its to enable the Reflashable and save the sign key to reflash the bootloader, and if you dont rebuild the progect or the bootloader, so you can reflash the bootloader without any problem.

I make this changes with a new ESP32: 
```sh
idf.py menuconfig
1. security features -> Enable flash encryption boot -> Checked
2. security features -> Enable flash encryption boot -> Enable usage mode -> developer
3. security features -> Potentially insecure options -> Require flash encryption to be already enabled -> unChecked
4. security features -> Enable hardware Secure Boot in bootloader -> Checked
5. security features -> Secure bootloader mode -> Reflashable
6. partition table -> partition table -> Custom partition table CSV
7. partition table -> offset -> 0xE000

# cerate a sign key
espsecure.py generate_signing_key secure_boot_signing_key.pem

# recompile progect
idf.py fullclean
idf.py bootloader
idf.py build

espsecure generate_flash_encryption_key my_flash_encryption_key.bin
espefuse --port PORT burn_key flash_encryption my_flash_encryption_key.bin

idf.py bootloader-flash
idf.py flash

idf.py monitor
```

to this point everithink its ok

then I try to flash the same bootloader without make any change, and without rebuild, just reflash the bootloader with this command
```sh
esptool.py --chip esp32 write_flash --encrypt 0x01000 ./build/bootloader/bootloader.bin --force
esptool.py --chip esp32 write_flash --encrypt 0x00000 ./build/bootloader/bootloader-reflash-digest.bin --force
```

and no one of this 2 command get good rerults
its olny show this error

```sh
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff00b8,len:11648
load:0x40078000,len:25440
load:0x40080400,len:4
load:0x40080404,len:3876
secure boot check fail
ets_main.c 371 
ets Jun  8 2016 00:22:57
```

note that I have the "build" folder added in the git so that nothing changes, and I can recover the bootloader if something happens

I think it's much easier for you (as developers) and for us (as consumers) to publish the docs, in a step-by-step list. and any step that has some risk, put a question mark next to it and when I hover over it, it explains the consequences of applying that line and a link to the full document in case anyone wants to consume it
like this first block I put in this post

hdpklm, 2023-05-21:
the best way is to create manualy the "secure-bootloader-digest.bin" by 32 random bytes and then burn in efuse block2, and use this key to digest bootloader before flash it

closed 2023-05-21 by hdpklm

AdityaHPatwardhan, 2023-05-23:
@hdpklm Have you taken a look at the following link [reflashable bootloader](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/secure-boot-v1.html#re-flashable-software-bootloader)

It gives the refreshing steps in details along with explanations.
Please let me know what problem did you face in these steps, I can update the document if it seems unclear in any way.

hdpklm, 2023-05-23:
> @hdpklm Have you taken a look at the following link [reflashable [...]

the esp-documents its so complicated, and the information its dispersed in different locations of the page.

I think its easy for everyone to understand the documents, but for me, I preferred obtaining all information about one thing in the same section, with the warranty, the errors, efuse related, and some examples, I lose about 20 pcs of esp32 this month, because I try to do wath I find in the document.

normally I read the section that I need, for example: I need to encrypt the flash, so I need to find everything related to flash encryption in this section, with notes of common error.

thank you so much
have a nice day

hdpklm, 2023-05-23:
ok thank you, but I think its better for reduce issues, because I find a lot of issues ask for this problem, so I think is beter to mentioned this emulate in document

thank you

hdpklm, 2023-05-23:
I don't know if you like how I explain it in this document, but it's easier for me to see all the consecutive commands, and apply the one I want.

Many comments are missing from this document, and notices of irreversible steps.

But when I want to learn how to encrypt the code in the esp32. First everything has to work very well. and then I start to investigate each instruction how it works and what it is for

but if the code doesn't work, I have to find out why it doesn't work, and how to make it work, and then start researching

https://github.com/hdpklm/Frameworks-Setup/blob/master/ESP32/ESP32-Encrypt.md