espressif/esp-idf #11433: Race condition in esp_ipc_isr_stall_other_cpu/esp_ipc_isr_release_other_cpu

pguyot, 2023-05-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.4 and master

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

qemu

### Power Supply used.

USB

### What is the expected behavior?

DPORT_STALL_OTHER_CPU_START() and DPORT_STALL_OTHER_CPU_END() should work as documented.

### What is the actual behavior?

There is a race condition that can be exhibited when calling nvs_flash_init() in qemu under a significant load, including in GitHub actions.
As an example of cases, please see this report which links to a dedicated repository:
https://github.com/espressif/qemu/issues/74

The problem is solved by the following patch against v4.4.4:
https://github.com/pguyot/esp32_nvs_flash_init_wdt/blob/patch_race_condition/esp_ipc_isr.patch

The associate run is here:
https://github.com/pguyot/esp32_nvs_flash_init_wdt/actions/runs/5014583807

I believe what happens is the following race condition:

|CPU0|CPU1|
|-|-|
| enters esp_ipc_isr_stall_other_cpu||
| - sets setting esp_ipc_isr_finish_cmd to 0||
| - enters esp_ipc_isr_call_and_wait||
||enters esp_ipc_isr_handler|
||- sets esp_ipc_isr_start_fl to 1|
|- exits esp_ipc_isr_call_and_wait||
|exits esp_ipc_isr_stall_other_cpu||
|enters esp_ipc_isr_release_other_cpu||
|- sets setting esp_ipc_isr_finish_cmd to 1||
|exits esp_ipc_isr_release_other_cpu||
|enters esp_ipc_isr_stall_other_cpu||
|- sets setting esp_ipc_isr_finish_cmd to 0||
|- - enters esp_ipc_isr_call_and_wait||
|| - enters esp_ipc_isr_waiting_for_finish_cmd|
|| - - stuck waiting for esp_ipc_isr_finish_cmd to be 1|

The patch consists in ensuring esp_ipc_isr_finish_cmd is done before exiting, so CPU0 does not reenter `esp_ipc_isr_stall_other_cpu` before CPU1 read `esp_ipc_isr_finish_cmd`.

I am not sure this could happen on real hardware.

### Steps to reproduce.

Please see https://github.com/espressif/qemu/issues/74

KonstantinKondrashov, 2023-05-22:
Thanks for the confirmation.
I am closing this issue now because there is https://github.com/espressif/esp-idf/pull/11447 (I am going to apply it).

see https://github.com/espressif/esp-idf/pull/11447 for further updates,

closed 2023-05-22 by KonstantinKondrashov

AxelLin, 2023-07-14:
Just remind this issue was reported against v4.4.4 but the fix is not yet available in v4.4 branch. @KonstantinKondrashov