espressif/esp-idf #11435: RFCOMM disconnects after 20-30 seconds if no data sent after connection established

fizzyade, 2023-05-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.1

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

Custom Board

### Power Supply used.

External 5V

### What is the expected behavior?

I have opened a connection to the ESP32 which is running code based on the classic spp_acceptor demo, I expect to be able to establish a connection and it to remain open until either side has some data to send.

### What is the actual behavior?

If I open the connect but do not send any data, then a number of seconds after the connection was opened the ESP32 appears to close the connection.  I see the following message in the ESP32 log:

port_rfc_closed RFCOMM connection in server:1 state 2 closed: Peer connection failed (res: 16)

### Steps to reproduce.

1. Create a RFcomm based application on the ESP32.
2. Open the port but do not send any data.
3. Observe the connection closing after a short period, 30-120 seconds.

### Debug Logs.

```plain
W (64210783) BT_APPL: new conn_srvc id:26, app_id:255
W (64217983) BT_HCI: hci cmd send: sniff: hdl 0x81, intv(400 800)
W (64218003) BT_HCI: hcif mode change: hdl 0x81, mode 2, intv 800, status 0x0
I (64218003) static void BluetoothSerialPort::gapCallback(esp_bt_gap_cb_event_t, esp_bt_gap_cb_param_t*): ESP_BT_GAP_MODE_CHG_EVT mode:2 bda:[0A:62:30:30:28:15]
W (64238313) BT_RFCOMM: port_rfc_closed RFCOMM connection in server:1 state 2 closed: Peer connection failed (res: 16)
W (64238313) BT_HCI: hcif disc complete: hdl 0x81, rsn 0x8
I (64238323) static void BluetoothSerialPort::gapCallback(esp_bt_gap_cb_event_t, esp_bt_gap_cb_param_t*): event: 17
```

xiongweichao, 2023-05-23:
Hi @fizzyade ,

Maybe this problem has been solved. Please use the latest [release/v5.0](https://github.com/espressif/esp-idf/tree/release/v5.0) to verify if this problem still exists.

Thanks

xiongweichao, 2023-05-23:
@fizzyade Maybe you misunderstand what I mean. I said [release/v5.0 ](https://github.com/espressif/esp-idf/tree/release/v5.0)(branch) not [v5.0](https://github.com/espressif/esp-idf/tree/v5.0) (tag).

xiongweichao, 2023-05-24:
Hi @fizzyade ,

May I ask whether you have tested it with release/v5.0?  Please let me know if the problem still exist.

Thanks

fizzyade, 2023-05-24:
> Hi @fizzyade ,
> 
> May I ask whether you have tested it with release/v5.0? Please let me know if the problem still exist.
> 
> Thanks

Yes, and it does seem to have resolved the issue.

closed 2023-05-24 by fizzyade