espressif/esp-idf #11436: Changing default configuration options

ChrisJones79, 2023-05-18:
I'm trying to understand the EPS-IDF build system better and make it work for me.

Normally, I use an ESP32-S3 N32R8V, which is not the default CONFIG_IDF_TARGET.

I have been reading through the project configuration sections and I just do not seem to be getting the right combination of settings.

What I am trying to do:
1. Default target : esp32s3
2. Default flash: 32 MB
3. etc...

I have set up Kconfig.projbuild. It shows up fine in menuconfig.

```
# put here your custom config value
menu "Board Selection"
    choice BOARD_NAME
        prompt "Pick a board"
        default ESP32S3
        help
            Select the board you are using. This will configure the right defaults for the rest of the menus.
        
        config ESP32S3
            bool "ESP32-S3"
        config ESP32S2
            bool "ESP32-S2"
        config ESP32
            bool "ESP32"
        config ESP32C3
            bool "ESP32-C3"
        config LILYGO_HMI
            bool "LILYGO T-HMI"
        config ESPEYE
            bool "ESP-EYE"
        config ESP32CAM
            bool "ESP32-CAM"
        config ESPADF
            bool "ESP-ADF"
    endchoice

    config BOARD_NAME
        string
        default "ESP32-S3" if ESP32S3
        default "ESP32-S2" if ESP32S2
        default "ESP32" if ESP32
        default "ESP32-C3" if ESP32C3
        default "LILYGO T-HMI" if LILYGO_HMI
        default "ESP-EYE" if ESPEYE
        default "ESP32-CAM" if ESP32CAM
        default "ESP-ADF" if ESPADF

    menu "ESP32-S3 Flash"
        default S3_FLASH_32
        depends on ESP32S3

        choice 
            bool "Flash Size [MB]"
            default S3_FLASH_32
            help
                Flash size of the ESP32-S3 module.

            config S3_FLASH_32
                bool "32 MB"
            config S3_FLASH_16
                bool "16 MB"
            config S3_FLASH_8
                bool "8 MB"            
        endchoice

        config S3_FLASH_SIZE
            bool 
            default 32 if S3_FLASH_32
            default 16 if S3_FLASH_16
            default 8 if S3_FLASH_8
    endmenu

    menu "ESP32-S3 PSRAM"
        default S3_PSRAM_8
        depends on ESP32S3

        choice 
            bool "PSRAM Size [MB]"
            default S3_PSRAM_8
            help
                PSRAM size of the ESP32-S3 module.

            config S3_PSRAM_8
                bool "8 MB"
            config S3_PSRAM_4
                bool "4 MB"
            config S3_PSRAM_0
                bool "0 MB"
        endchoice

        config S3_PSRAM_SIZE
            bool 
            default 8 if S3_PSRAM_8
            default 4 if S3_PSRAM_4
            default 0 if S3_PSRAM_0
    endmenu

endmenu

```

I want to be able to use those values I've set as default (so they work without calling menuconfig) as the project configuration.

I tried using conditional configuration in my project CMakeLists.txt which I will show here:
```
# The following lines of boilerplate have to be in your project's
# CMakeLists in this exact order for cmake to work correctly
cmake_minimum_required(VERSION 3.5)

set (SDKCONFIG_DEFAULTS sdkconfig.defaults)
set (CONFIG_IDF_TARGET )

if (BOARD_NAME)
    if (ESP32S3)
        UNSET (CONFIG_IDF_TARGET_ESP32)
        SET (CONFIG_IDF_TARGET_ESP32S3)
        SET (CONFIG_IDF_TARGET "esp32s3")
    endif()
endif()    
 
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(app-template)
```

I have also tried creating a sdkconfig.defaults file where I tried:

```
CONFIG_IDF_TARGET="esp32s3"
CONFIG_IDF_TARGET_ESP32S3=y
```

My procedure for checking any of these changes was to 
1. make sure vs-code target was set to ESP32 in the status bar
2. Revert sdkconfig to be set to esp32 settings
3. Clean all
4. Open, save, close menuconfig (relying on defaults defined in projbuild)
5. Build

Any help would be greatly appreciated!

Also, I would like to be doing other defaults like 240 MHz, minimal components added, and some others.
7.

ChrisJones79, 2023-05-18:
I think I should have changed the BOARD_NAME to bool for that work with the conditional in CMakeLists.

Would that look like:

```
config BOARD_NAME
        bool
        default y if ESP32S3
```
or
```
        default true if ESP32S2
```
or
```
        default if ESP32
        default "ESP32-C3" if ESP32C3
        default "LILYGO T-HMI" if LILYGO_HMI
        default "ESP-EYE" if ESPEYE
        default "ESP32-CAM" if ESP32CAM
        default "ESP-ADF" if ESPADF

ESP-Marius, 2024-08-19:
Seems like the initial question here was already resolved so i'll mark it as closed. 

Feel free to reopen if you think there is still something we need to fix here.

closed 2024-08-19 by ESP-Marius