espressif/esp-idf #11437: esp_wifi_scan_start failure if called after WiFi.begin()

mahboud, 2023-05-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

What is included with Arduino 2.0.9

### Operating System used.

macOS

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32C3M1 and a custom board I've made, both show similar behavior

### Power Supply used.

USB

### What is the expected behavior?

I should be able to run `esp_wifi_scan_start()` after `WiFi.begin()` has been called and then disconnected and cleaned up, and get valid results.

### What is the actual behavior?

Scenario 1:

Call `esp_wifi_scan_start()`, and then 
   `esp_wifi_scan_get_ap_records(&number, ap_info);` // Number is 2
   `esp_wifi_scan_get_ap_num(&ap_count);`  // count is 2
   
Scenario 2:
Call `WiFi.begin();`, then later `WiFi.disconnect(true);` Now call `esp_wifi_scan_start()`, and then 
   `esp_wifi_scan_get_ap_records(&number, ap_info);` // ** Number is 0**
   `esp_wifi_scan_get_ap_num(&ap_count);`  // count is 2
   

### Steps to reproduce.

```
vod startNet() {
...
wifi_scan2();

  WiFi.disconnect(true);
  ESP_LOGE("start", "disconnect again");

  wl_status_t status = WiFi.begin();
  while (status != WL_CONNECT_FAILED) {
    delay(100);
    status = WiFi.status();
    if (status == WL_CONNECTED) {
      ESP_LOGI("start ---", "%s connected! status %d and IP %s", WiFi.SSID(),
               status, WiFi.localIP().toString().c_str());
      break;
    }
  }
  WiFi.disconnect(true);
  do {
    delay(100);
    status = WiFi.status();
    ESP_LOGI("start", "Waiting Disconnect status %d", status);
  } while (status != WL_NO_SHIELD && status != WL_IDLE_STATUS);

  wifi_scan2();
....
}

void wifi_scan2(void) {
// Location 1

  esp_wifi_stop();
  esp_wifi_deinit();
  ESP_LOGE("try", "deinit");
  ESP_LOGE("scan", "1");

  esp_netif_t* sta_netif = esp_netif_get_handle_from_ifkey("WIFI_STA_DEF");
  if (!sta_netif) {
    sta_netif = esp_netif_create_default_wifi_sta();
  }

// I've already tried doing this
  //   esp_netif_t* netif = esp_netif_get_handle_from_ifkey("WIFI_STA_DEF");
  //   if (netif) {
  //       esp_netif_destroy(netif);
// in Location 1 above, and instead doing:
  //  esp_netif_t *sta_netif = esp_netif_create_default_wifi_sta();

  assert(sta_netif);
  ESP_LOGE("scan", "2");

  wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
  ESP_ERROR_CHECK(esp_wifi_init(&cfg));
  ESP_LOGE("scan", "3");

  wifi_ap_record_t ap_info[DEFAULT_SCAN_LIST_SIZE];
  memset(ap_info, 0, sizeof(ap_info));

  ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA));
  ESP_ERROR_CHECK(esp_wifi_start());
  ESP_LOGE("scan", "4");

  wifi_scan_config_t scanConf = {.ssid = NULL,
                                 .bssid = NULL,
                                 .channel = 0,
                                 .show_hidden = false,
                                 .scan_type = WIFI_SCAN_TYPE_ACTIVE};
  scanConf.scan_time.active.max = 350;
  scanConf.scan_time.active.min = 150;

    uint16_t number = DEFAULT_SCAN_LIST_SIZE;
    uint16_t ap_count = 0;

    esp_wifi_scan_start(&scanConf, true);

    ESP_ERROR_CHECK(esp_wifi_scan_get_ap_records(&number, ap_info));
    ESP_ERROR_CHECK(esp_wifi_scan_get_ap_num(&ap_count));
    ESP_LOGI(TAG, "Total BSSIDs Seen = %u (%u)", ap_count, number);
  esp_wifi_stop();
  esp_wifi_deinit();
  esp_netif_destroy(sta_netif);

}

```

### Debug Logs.

```plain
Running the above:

[ 3190][E][KiniNetworking.cpp:856] wifi_scan2(): [try] deinit
[ 3195][E][KiniNetworking.cpp:868] wifi_scan2(): [scan] 1
[ 3201][E][KiniNetworking.cpp:876] wifi_scan2(): [scan] 2
[ 3214][E][KiniNetworking.cpp:880] wifi_scan2(): [scan] 3
[ 3265][E][KiniNetworking.cpp:887] wifi_scan2(): [scan] 4
[ 6465][KiniNetworking.cpp:904] wifi_scan2(): [scan] Total BSSIDs Seen = 2 (2)
[ 6467][E][KiniNetworking.cpp:150] startNetwork(): [start] disconnect again
[ 6476][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 0 - WIFI_READY
[ 6480][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 2 - STA_START
[ 6535][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[ 8505][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[ 8505][D][WiFiGeneric.cpp:996] _eventCallback(): STA IP: 10.1.1.181, MASK: 255.255.255.0, GW: 10.1.1.1
[ 8587][KiniNetworking.cpp:158] startNetwork(): [start ---] McCormick connected! status 3 and IP 10.1.1.181
[ 8589][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[ 8594][W][WiFiGeneric.cpp:955] _eventCallback(): Reason: 8 - ASSOC_LEAVE
[ 8602][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 3 - STA_STOP
[ 8709][KiniNetworking.cpp:166] startNetwork(): [start] Waiting Disconnect status 255
E (8749) wifi_init: Failed to deinit Wi-Fi driver (0x3001)
[ 8712][E][KiniNetworking.cpp:856] wifi_scan2(): [try] deinit
[ 8717][E][KiniNetworking.cpp:868] wifi_scan2(): [scan] 1
[ 8722][E][KiniNetworking.cpp:876] wifi_scan2(): [scan] 2
[ 8734][E][KiniNetworking.cpp:880] wifi_scan2(): [scan] 3
[ 8737][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 2 - STA_START
[ 8740][E][KiniNetworking.cpp:887] wifi_scan2(): [scan] 4
[ 11879][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 1 - SCAN_DONE
[ 11944][KiniNetworking.cpp:904] wifi_scan2(): [scan] Total BSSIDs Seen = 2 (0)
[ 11945][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 3 - STA_STOP

If I comment out the code around WiFi.begin(), then the logs are:

[ 3190][E][KiniNetworking.cpp:857] wifi_scan2(): [try] deinit
[ 3196][E][KiniNetworking.cpp:869] wifi_scan2(): [scan] 1
[ 3201][E][KiniNetworking.cpp:877] wifi_scan2(): [scan] 2
[ 3214][E][KiniNetworking.cpp:881] wifi_scan2(): [scan] 3
[ 3265][E][KiniNetworking.cpp:888] wifi_scan2(): [scan] 4
[ 6466][KiniNetworking.cpp:905] wifi_scan2(): [scan] Total BSSIDs Seen = 2 (2)
[ 6567][KiniNetworking.cpp:167] startNetwork(): [start] Waiting Disconnect status 255
E (6607) wifi_init: Failed to deinit Wi-Fi driver (0x3001)
[ 6570][E][KiniNetworking.cpp:857] wifi_scan2(): [try] deinit
[ 6575][E][KiniNetworking.cpp:869] wifi_scan2(): [scan] 1
[ 6580][E][KiniNetworking.cpp:877] wifi_scan2(): [scan] 2
[ 6593][E][KiniNetworking.cpp:881] wifi_scan2(): [scan] 3
[ 6595][E][KiniNetworking.cpp:888] wifi_scan2(): [scan] 4
[ 9796][KiniNetworking.cpp:905] wifi_scan2(): [scan] Total BSSIDs Seen = 2 (2)
```

### More Information.

Clearly there are some leftovers from "WiFi.begin()" that aren't cleaned out with a "WiFi.disconnect(true);"

I've tried to handle that with:

```
  esp_wifi_stop();
  esp_wifi_deinit();
  ESP_LOGE("try", "deinit");
  ESP_LOGE("scan", "1");

   esp_netif_t* netif = esp_netif_get_handle_from_ifkey("WIFI_STA_DEF");
   if (netif) {
   esp_netif_destroy(netif);
```

There's still something amiss.

mahboud, 2023-05-19:
Having two `WiFi.enableSTA()` like this, doesn't to correct whatever happens in the first one, and the scan fails:

```
  WiFi.enableSTA(true);
  WiFi.enableSTA(false);
  wifi_scan2();

```

mhdong, 2023-05-19:
Hi @mahboud Could you provide a complete reproduce demo project?

mhdong, 2023-05-22:
Hi @mahboud 
WiFi.begin() will register callback function:
`/**
 * private
 * scan callback
 * @param result  void *arg
 * @param status STATUS
 */
void WiFiScanClass::_scanDone()
{
    esp_wifi_scan_get_ap_num(&(WiFiScanClass::_scanCount));
    if(WiFiScanClass::_scanCount) {
        WiFiScanClass::_scanResult = new wifi_ap_record_t[WiFiScanClass::_scanCount];
        if(!WiFiScanClass::_scanResult || esp_wifi_scan_get_ap_records(&(WiFiScanClass::_scanCount), (wifi_ap_record_t*)_scanResult) != ESP_OK) {
            WiFiScanClass::_scanCount = 0;
        }
    }
    WiFiScanClass::_scanStarted=0; //Reset after a scan is completed for normal behavior
    WiFiGenericClass::setStatusBits(WIFI_SCAN_DONE_BIT);
    WiFiGenericClass::clearStatusBits(WIFI_SCANNING_BIT);
}`
function esp_wifi_scan_get_ap_records will be called after scan done.

`/**
  * @brief     Scan all available APs.
  *
  * @attention If this API is called, the found APs are stored in WiFi driver dynamic allocated memory and the
  *            will be freed in esp_wifi_scan_get_ap_records, so generally, call esp_wifi_scan_get_ap_records to cause
  *            the memory to be freed once the scan is done
  * @attention The values of maximum active scan time and passive scan time per channel are limited to 1500 milliseconds.
  *            Values above 1500ms may cause station to disconnect from AP and are not recommended.
  *
  * @param     config  configuration of scanning
  * @param     block if block is true, this API will block the caller until the scan is done, otherwise
  *                         it will return immediately
  *
  * @return
  *    - ESP_OK: succeed
  *    - ESP_ERR_WIFI_NOT_INIT: WiFi is not initialized by esp_wifi_init
  *    - ESP_ERR_WIFI_NOT_STARTED: WiFi was not started by esp_wifi_start
  *    - ESP_ERR_WIFI_TIMEOUT: blocking scan is timeout
  *    - ESP_ERR_WIFI_STATE: wifi still connecting when invoke esp_wifi_scan_start
  *    - others: refer to error code in esp_err.h
  */
esp_err_t esp_wifi_scan_start(const wifi_scan_config_t *config, bool block);`
If esp_wifi_scan_get_ap_records is called, the found APs are stored in WiFi driver dynamic allocated memory and the will be freed.
So when you call esp_wifi_scan_get_ap_records again, you get nothing.

closed 2023-06-12 by Alvin1Zhang