espressif/esp-idf #11438: sdmmc_init_ocr: send_op_cond (1) returned 0x107

LazzyMozzie, 2023-05-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

V4.4.4

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

CustomBoard - ESP32S3

### Power Supply used.

External 5V

### What is the expected behavior?

when system is repoered or softrest, the sdcard running normally.

### What is the actual behavior?

when system is repoered or softrest, there is a probability thad sdcard does not respond, and return the error :

I (702) SD_FLASH: [../components/sd_flash/sd_flash.c:75] Initializing SD card
I (702) SD_FLASH: [../components/sd_flash/sd_flash.c:82] Using SDMMC peripheral
I (722) RIL: [../main/cloud_data_sync/ril_task.c:74] RIL_PIN_RESET:1
I (802) SD_FLASH: [../components/sd_flash/sd_flash.c:127] Mounting filesystem
D (802) sdmmc_periph: peripheral version 5342270a, hardware config 03c44c83
D (802) intr_alloc: Connected src 30 to int 18 (cpu 0)
E (802) sdmmc_periph: -----------use  SOC_SDMMC_USE_GPIO_MATRIX
E (802) sdmmc_periph: -----------pull up en 0
E (802) sdmmc_periph: --------------- pin 1
D (802) sdmmc_periph: using GPIO36 as clk pin
I (802) gpio: GPIO[36]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
E (802) sdmmc_periph: --------------- pin 2
D (802) sdmmc_periph: using GPIO35 as cmd pin
I (802) gpio: GPIO[35]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
E (802) sdmmc_periph: --------------- pin 3
D (802) sdmmc_periph: using GPIO37 as d0 pin
I (802) gpio: GPIO[37]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
D (802) sdmmc_periph: slot=1 host_div=10 card_div=20 freq=400kHz
D (802) sdmmc_periph: slot=1 width=1
W (802) vfs_fat_sdmmc: --------------SDIO MODE
D (802) sdmmc_req: process_command_response: error 0x107  (status=00000100)
D (822) sdmmc_req: process_command_response: error 0x107  (status=00000100)
D (822) sdmmc_sd: CMD8 timeout; not an SD v2.00 card
D (822) sdmmc_req: process_command_response: error 0x107  (status=00000100)
D (822) sdmmc_io: sdmmc_init_io: io_send_op_cond (1) returned 0x107; not IO card
D (822) sdmmc_req: process_command_response: error 0x107  (status=00000100)
D (822) sdmmc_req: process_command_response: error 0x107  (status=00000100)
D (822) sdmmc_req: process_command_response: error 0x107  (status=00000100)
D (822) sdmmc_cmd: sdmmc_send_cmd_send_op_cond: sdmmc_send_app_cmd err=0x107
D (822) sdmmc_common: send_op_cond timeout, trying MMC
D (822) sdmmc_req: process_command_response: error 0x107  (status=00000100)
D (832) sdmmc_req: process_command_response: error 0x107  (status=00000100)
[... 567 lines cut ...]
E (811) sdmmc_periph: --------------- pin 2
D (811) sdmmc_periph: using GPIO35 as cmd pin
I (811) gpio: GPIO[35]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
E (811) sdmmc_periph: --------------- pin 3
D (811) sdmmc_periph: using GPIO37 as d0 pin
I (811) gpio: GPIO[37]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
D (811) sdmmc_periph: slot=1 host_div=10 card_div=20 freq=400kHz
D (811) sdmmc_periph: slot=1 width=1
W (811) vfs_fat_sdmmc: --------------SDIO MODE
D (811) sdmmc_req: process_command_response: error 0x107  (status=00000100)
D (831) sdmmc_sd: SDHC/SDXC card
D (831) sdmmc_req: process_command_response: error 0x107  (status=00000100)
D (831) sdmmc_io: sdmmc_init_io: io_send_op_cond (1) returned 0x107; not IO card
I (921) RIL: [../main/cloud_data_sync/ril_task.c:74] RIL_PIN_RESET:0
D (10831) sdmmc_common: send_op_cond timeout, trying MMC
D (10831) sdmmc_req: process_command_response: error 0x107  (status=00000100)
D (10831) sdmmc_req: process_command_response: error 0x107  (status=00000100)
D (10831) sdmmc_req: process_command_response: error 0x107  (status=00000100)
D (10831) sdmmc_cmd: sdmmc_send_cmd_send_op_cond: sdmmc_send_app_cmd err=0x107
E (10831) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
D (10831) sdmmc_init: sdmmc_card_init: sdmmc_init_ocr returned 0x107
E (10831) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).
E (10831) SD_FLASH: [../components/sd_flash/sd_flash.c:141] Failed to initialize the card (ESP_ERR_TIMEOUT). Make sure SD card lines have pull-up resistors in place.
E (10831) DataReupload: [../main/data_reupload/data_reupload.c:464] SD card init failed ....
```

igrr, 2023-05-19:
This is probably the same issue as https://github.com/espressif/esp-idf/issues/8521. The fix has been merged into release/v4.4 branch in https://github.com/espressif/esp-idf/commit/fb6d9cd7417c90932529c2d83cdc42d5fbd6ff50 and will appear in the next bugfix release, v4.4.5.

LazzyMozzie, 2023-05-19:
@igrr hi, i just cherry-pick the commit ``` fb6dc9 ``` to the v4.4.4, it has no effect; And i alos find , the lower the sdio rate, the more likely this problem is to occur

aircable, 2023-11-06:
With the latest IDF: git describe --abbrev=0 --tags: v5.2-dev
the example storage/sd-card/sdmmc has the same problem, error 0x107, ESP_ERR_TIMEOUT on the ESP32S3_WROOM module. Please note that the Freenove board that uses this module has no connection on the SD card pin 2 (CD).
On an older IDF it worked as it should.

JGrossholtz, 2024-09-09:
I can confirm the problem with esp-idf 5.2.2 and a custom board using an esp32-s3. After a soft reset the issue goes away.

MrMoDDoM, 2024-09-11:
I'm currently facing the same issue with `v5.3-369-g466a392a76-dirty`, on a custom board with the below schematic, using the Arduino IDE to program the board.

[image]

An interesting fact is that when I use this function:

```
esp_err_t mount_sd(gpio_num_t pin_cmd, gpio_num_t pin_clk, gpio_num_t pin_d0, const char* mountpoint, bool format_if_failed, int max_files) {
    esp_err_t res;
    
    sdmmc_host_t host = SDMMC_HOST_DEFAULT();
    sdmmc_slot_config_t slot_config = SDMMC_SLOT_CONFIG_DEFAULT();

    host.flags = SDMMC_HOST_FLAG_1BIT;
    host.max_freq_khz = 40000;
    slot_config.width = 1;
    
    gpio_pad_select_gpio(pin_cmd);
    gpio_set_direction(pin_cmd, GPIO_MODE_INPUT_OUTPUT_OD);
    gpio_set_pull_mode(pin_cmd, GPIO_PULLUP_ONLY);
    gpio_set_level(pin_cmd, 1);
    
    gpio_pad_select_gpio(pin_clk);
    gpio_set_direction(pin_clk, GPIO_MODE_INPUT_OUTPUT_OD);
    gpio_set_pull_mode(pin_clk, GPIO_PULLUP_ONLY);
    gpio_set_level(pin_clk, 1);
    
    gpio_pad_select_gpio(pin_d0);
    gpio_set_direction(pin_d0,  GPIO_MODE_INPUT_OUTPUT_OD);
    gpio_set_pull_mode(pin_d0,  GPIO_PULLUP_ONLY);
    gpio_set_level(pin_d0,  1);
    
    esp_vfs_fat_sdmmc_mount_config_t mount_config = {
        .format_if_mount_failed = format_if_failed,
        .max_files              = max_files,
        .allocation_unit_size   = 1024
    };
    
    res = esp_vfs_fat_sdmmc_mount(mountpoint, &host, &slot_config, &mount_config, &sdcardCard);
    
    if (res != ESP_OK) {
        if (res == ESP_FAIL) {
            ESP_LOGE(TAG, "Failed to mount the SD card filesystem.");
        } else if (res == ESP_ERR_NO_MEM) {
            ESP_LOGE(TAG, "Failed to initialize the SD card: not enough memory).");
        } else if (res == ESP_ERR_INVALID_RESPONSE) {
            ESP_LOGE(TAG, "Failed to initialize the SD card: invalid response).");
        } else if (res == ESP_ERR_INVALID_STATE) {
            ESP_LOGE(TAG, "Failed to initialize the SD card: invalid state).");
        } else {
            ESP_LOGE(TAG, "Failed to initialize the SD card (%s). ", esp_err_to_name(res));
        }
        return res;
    }
    return res;
}
```
...the error is `ESP_ERR_TIMEOUT`

```
E (1061) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107    
E (1061) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).    
[   563][E][SDMMC_Test.ino:116] mount_sd(): [sdcard] Failed to initialize the SD card (ESP_ERR_TIMEOUT).    
```

But when I use the example shipped with the ArduinoIDE, the error message does not mention a timeout error:

```
E (1130) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
E (1130) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).
[   597][E][SD_MMC.cpp:194] begin(): Failed to initialize the card (0x107). Make sure SD card lines have pull-up resistors in place.
```

Last thing, when using another SD the error code changes from 0x107 to 0x108.. I'm not completely sure why.

Sadly the soft reset method mentioned by @JGrossholtz has no effects.

igrr, 2024-09-11:
@MrMoDDoM In that case your issue is probably unrelated to the one discussed in this ticked. The issue here is specifically for the case (quoting the original report):

> when system is re-powered or soft reset, there is a probability thad sdcard does not respond, and return the error

I suggest opening a separate issue, instead.

MrMoDDoM, 2024-09-12:
Sorry to bother you further, I just discovered that there was a connection problem with the SD holder, and everything works perfectly on ESP32-S3 with ESP-IDF version v5.3-369-g466a392a76-dirty.

My bad, sorry again.