espressif/esp-idf #11439: CMake Error at CMakeLists.txt:5 (message): Current directory is not   buildable.

icepick3000, 2023-05-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

What am I doing wrong here? Trying to instal idf 4.1 (the higher versions have a camera bug that in my case is problematic)

I issue `idf.py menuconfig`;

`Executing action: menuconfig
Running cmake in directory /Users/alex/Downloads/esp32firmware/esp-idf/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/Users/alex/.espressif/python_env/idf5.2_py3.9_env/bin/python -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /Users/alex/Downloads/esp32firmware/esp-idf"...
-- The C compiler identification is AppleClang 11.0.3.11030032
-- The CXX compiler identification is AppleClang 11.0.3.11030032
-- The ASM compiler identification is Clang with GNU-like command-line
-- Found assembler: /Library/Developer/CommandLineTools/usr/bin/cc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at CMakeLists.txt:5 (message):
  Current directory '/Users/alex/Downloads/esp32firmware/esp-idf' is not
  buildable.  Change directories to one of the example projects in
  '/Users/alex/Downloads/esp32firmware/esp-idf/examples' and try again.`

Running it from the examples directory doesnt work as well..

ESP-Marius, 2023-05-19:
You need to be in a folder with a project `CMakeLists.txt`. Try running it from e.g. `examples/get-started/hello_world`

icepick3000, 2023-05-19:
This works, thanks!

closed 2023-05-19 by icepick3000