espressif/esp-idf #11441: uart_echo sample (and my app) fails with "invalid header: 0xffffffff"

MartyMcFlyInTheSky, 2023-05-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-4665-gef64e4e5b3

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32, ESP32-mini

### Power Supply used.

USB

### What is the expected behavior?

Running the uart_echo sample works as expected, doing nothing as nothing is hooked up to the pins (should repeat while loop indefinitely).

### What is the actual behavior?

Building the uart_echo sample from examples\peripherals\uart builds and flashes, but when I run it the eventually prints "invalid header: 0xffffffff"

I tried it with the sample app as I was experiencing this problem with my production app when I merged new code into the repo that contained UART headers. Indeed, running anything that does not include UART works just fine (I tried with the blink example and my app BEFORE including the uart lib).

This is the whole output:

```
C:/dev/uart_echo/build -P C:/dev/esp/esp-idf/components/esptool_py/run_serial_tool.cmake"
esptool esp32 -p COM5 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 2MB 0x1000 bootloader/bootloader.bin 0x10000 uart_echo.bin 0x8000 partition_table/partition-table.bin
esptool.py v4.5.1
Serial port COM5
Connecting....
Chip is ESP32-U4WDH (revision v3.0)
Features: WiFi, BT, Dual Core, 240MHz, Embedded Flash, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: c8:f0:9e:47:ef:ec
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00007fff...
Flash will be erased from 0x00010000 to 0x00042fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Compressed 26704 bytes to 16722...
Writing at 0x00001000... (50 %)
Writing at 0x0000769c... (100 %)
Wrote 26704 bytes (16722 compressed) at 0x00001000 in 1.0 seconds (effective 224.7 kbit/s)...
Hash of data verified.
Compressed 206704 bytes to 115417...
Writing at 0x00010000... (12 %)
Writing at 0x0001d154... (25 %)
Writing at 0x00022d2c... (37 %)
Writing at 0x00028a45... (50 %)
Writing at 0x0002df73... (62 %)
Writing at 0x0003632f... (75 %)
Writing at 0x0003c51c... (87 %)
Writing at 0x00042252... (100 %)
Wrote 206704 bytes (115417 compressed) at 0x00010000 in 3.1 seconds (effective 534.2 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 103...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (103 compressed) at 0x00008000 in 0.1 seconds (effective 239.9 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
Executing action: monitor
Running idf_monitor in directory C:\dev\uart_echo
Executing "C:\dev\esp\env\Scripts\python.exe C:\dev\esp\esp-idf\tools/idf_monitor.py -p COM5 -b 115200 --toolchain-prefix xtensa-esp32-elf- --target esp32 --revision 0 C:\dev\uart_echo\build\uart_echo.elf --force-color -m 'C:\dev\esp\env\Scripts\python.exe' 'C:\dev\esp\esp-idf\tools\idf.py' '-p' 'COM5'"...
←[0;33m--- WARNING: GDB cannot open serial ports accessed as COMx←[0m
←[0;33m--- Using \\.\COM5 instead...←[0m
←[0;33m--- idf_monitor on \\.\COM5 115200 ---←[0m
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ting scheduler on CPU0
I (341) app_�ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7104
load:0x40078000,len:15624
load:0x40080400,len:4
0x40080400: _init at ??:?

ho 8 tail 4 room 4
load:0x40080404,len:3872
entry 0x40080648
I (32) boot: ESP-IDF v5.1-dev-4665-gef64e4e5b3-dirty 2nd stage bootloader
I (32) boot: compile time May 19 2023 14:28:32
I (34) boot: Multicore bootloader
I (38) boot: chip revision: v3.0
I (42) boot.esp32: SPI Speed      : 40MHz
I (47) boot.esp32: SPI Mode       : DIO
I (51) boot.esp32: SPI Flash Size : 2MB
I (56) boot: Enabling RNG early entropy source...
I (61) boot: Partition Table:
I (65) boot: ## Label            Usage          Type ST Offset   Length
I (72) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (79) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (87) boot:  2 factory          factory app      00 00 00010000 00100000
I (94) boot: End of partition table
I (99) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=0b168h ( 45416) map
I (123) esp_image: segment 1: paddr=0001b190 vaddr=3ffb0000 size=02168h (  8552) load
I (127) esp_image: segment 2: paddr=0001d300 vaddr=40080000 size=02d18h ( 11544) load
I (134) esp_image: segment 3: paddr=00020020 vaddr=400d0020 size=185bch ( 99772) map
I (174) esp_image: segment 4: paddr=000385e4 vaddr=40082d18 size=0a160h ( 41312) load
I (198) boot: Loaded app from partition at offset 0x10000
I (198) boot: Disabling RNG early entropy source...
I (209) cpu_start: Multicore app
I (210) cpu_start: Pro cpu up.
I (210) cpu_start: Starting app cpu, entry point is 0x400812b0
0x400812b0: call_start_cpu1 at C:/dev/esp/esp-idf/components/esp_system/port/cpu_start.c:159

I (0) cpu_start: App cpu up.
I (228) cpu_start: Pro cpu start user code
I (228) cpu_start: cpu freq: 160000000 Hz
I (228) cpu_start: Application information:
I (232) cpu_start: Project name:     uart_echo
I (237) cpu_start: App version:      1
I (242) cpu_start: Compile time:     May 19 2023 14:28:15
I (248) cpu_start: ELF file SHA256:  160667b233b38599...
I (254) cpu_start: ESP-IDF:          v5.1-dev-4665-gef64e4e5b3-dirty
I (261) cpu_start: Min chip rev:     v0.0
I (266) cpu_start: Max chip rev:     v3.99 
I (271) cpu_start: Chip rev:         v3.0
I (275) heap_init: Initializing. RAM available for dynamic allocation:
I (283) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (289) heap_init: At 3FFB2A20 len 0002D5E0 (181 KiB): DRAM
I (295) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (301) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (308) heap_init: At 4008CE78 len 00013188 (76 KiB): IRAM
I (315) spi_flash: detected chip: generic
I (318) spi_flash: flash io: dio
W (322) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (336) app_start: Starting scheduler on CPU0
I (341) app_start: Starting scheduler on CPU1
I (341) main_task: Started on CPU0
I (351) main_task: Calling app_main()
ets Jul 29 2019 12:21:46

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
ets Jul 29 2019 12:21:46

rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
ets Jul 29 2019 12:21:46

rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
ets Jul 29 2019 12:21:46

...
```

This last section with TG0WDT_SYS_RESET is repeated indefinitely.

Please note  that I used a barebone ESP32 dev-kit as well as an ESP32-mini dev-kit from espressif without hooking up any pin. But I did not find any hint that this is disallowed.

### Steps to reproduce.

1. Step: Download esp-idf (v5.1-dev-4665-gef64e4e5b3) and install toolchain
2. Step: Build "esp_echo" from examples and flash/run on ESP32 dev-kit
...

MartyMcFlyInTheSky, 2023-05-24:
Can anyone confirm? @songruo

dobairoland, 2023-05-31:
Please share your `sdkconfig` file from the project. Have you tried to remove it and rebuild the project?

Alvin1Zhang, 2025-01-08:
Thanks for reporting, will close due to can't reproduce, feel free to reopen with more updates/details.

closed 2025-01-08 by Alvin1Zhang