espressif/esp-idf #11442: Is infinite OTA possible with on chip encryption key generation?

Devyami-Apcon, 2023-05-19:
I have a firmware which I want to protect from intruders who can access my binary and make their separate first copy devices. 

Thus, I am planning to use flash encryption. 

Now to really flash encrypt, I feel eFuse will be used so that no one can software read the key. 
Also because there will be many devices, it's better to use separate encryption keys for all devices, meaning the encryption keys will be generated on run time on the chip. 

Here is the problem statement: 

I also have OTA capabilities. I read somewhere that only 4 times OTA is possible because it will burn the eFuse everytime I OTA. 
So because of limited eFuses I will be able to OTA my firmware with encryption just 4 times. 

Please give me a clarification on this

chipweinberger, 2023-05-19:
you can ota as much as you want.

the firmware will be written to flash, and re-encrypted with the key in the efuse during write.

mahavirj, 2023-05-22:
@Devyami-Apcon 

As @chipweinberger mentioned there is no limit on the number of OTA updates on flash encryption enabled device. EFuse configuration happens during the first boot or during manufacturing stage itself, no additional EFuses are burnt during OTA updates. There is exception to secure version field, but that is applicable only if anti-rollback scheme is enabled.

Note: typically SPI flash part supports ~100K program/erase cycles and hence that will only be a limiting factor here. 

Hope this helps!

closed 2023-05-22 by mahavirj