espressif/esp-idf #11445: mbedtls Linking Error - Could not create Symbolic Link

Josh-TapNoa, 2023-05-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.0.2

### Operating System used.

macOS

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Build - Particularly build esp-idf and esp-rainmaker (inside a docker) 

I do however get the same error when trying to start either idf.py build or idf.py menuconfig

I know the underline firmware builds on ubuntu 20.04 LTS

### What is the actual behavior?

CMake Error at /esp-idf/components/mbedtls/mbedtls/CMakeLists.txt:138 (message):
  Could not create symbolic link for:
  /esp-idf/components/mbedtls/mbedtls/library/error.c -->
Call Stack (most recent call first):
  /esp-idf/components/mbedtls/mbedtls/library/CMakeLists.txt:176 (link_to_source)

### Steps to reproduce.

Dockerfile (im new to docker so apologies if its a little ugly!) :
FROM python:3.9

ARG DEBIAN_FRONTEND=nointeractive
ARG CONTAINER_USER=esp
ARG USER_UID=1000
ARG USER_GID=$USER_UID

USER root

RUN apt-get clean

RUN apt-get update && apt-get install -y --no-install-recommends \
    bison \
    ca-certificates \
    ccache \
    check \
    coreutils \
    curl \
    flex \
    git \
    gperf \
    lcov \
    libffi-dev \
    libncurses-dev \
    libusb-1.0-0-dev \
    make \
    cmake \
    ninja-build \
    perl \
    unzip \
    wget \
    xz-utils \
    zip \
    grep \
  && apt-get clean \
  && rm -rf /var/lib/apt/lists/*

RUN update-alternatives --install /usr/bin/python python /usr/bin/python3 10 \
  && python -m pip install --upgrade \
    pip \
    virtualenv

RUN python3 --version

ENV PATH="/usr/bin/cmake:${PATH}"

RUN groupadd --gid $USER_GID $CONTAINER_USER \
   && adduser --uid $USER_UID --gid $USER_GID --disabled-password --gecos "" ${CONTAINER_USER} \
   && usermod -a -G dialout $CONTAINER_USER

RUN mkdir -p ~/.ssh 
RUN chmod 0700 ~/.ssh 
RUN ssh-keyscan github.com >> ~/.ssh/known_hosts

RUN echo "IdentityFile ~/.ssh/id_rsa" >> /etc/ssh/ssh_config
RUN echo "StrictHostKeyChecking no" >> /etc/ssh/ssh_config
RUN echo "Host github.com\n\tStrictHostKeyChecking no\n" >> /root/.ssh/config

COPY id_rsa /root/.ssh/id_rsa

RUN git clone --recursive --depth=1 https://github.com/espressif/esp-idf.git \
    && chown -R $CONTAINER_USER:$CONTAINER_USER /esp-idf \
    && chmod -R 777 /esp-idf

RUN git clone --recursive https://github.com/espressif/esp-rainmaker.git \ 
    && chown -R $CONTAINER_USER:$CONTAINER_USER /esp-rainmaker \
    && chmod -R 777 /esp-rainmaker

RUN git clone --recursive git@github.com:Josh-TapNoa/Wireless-Controller-New.git \ 
    && chown -R $CONTAINER_USER:$CONTAINER_USER /Wireless-Controller-New \
    && chmod -R 777 /Wireless-Controller-New

ENV LC_ALL=C.UTF-8
ENV LANG=C.UTF-8

ENV USER=${CONTAINER_USER}

RUN echo "source esp-idf/export.sh > /dev/null 2>&1" >> /root/.profile

ENV IDF_PATH=/esp-idf
ENV PATH="$PATH:$IDF_PATH"

ENV IDF_TOOLS_PATH=/esp-idf/tools
ENV PATH="$PATH:$IDF_TOOLS_PATH"

ENV IDF_CMAKE_PATH=/esp-idf/tools/cmake
ENV PATH="$PATH:$IDF_CMAKE_PATH"

ENV RMAKER_PATH=/esp-rainmaker/cli
ENV PATH="$PATH:$RMAKER_PATH"

ENV PYTHON_PATH=/home/esp/.local/bin
ENV PATH="$PATH:$PYTHON_PATH"

RUN python -m pip install -r $IDF_PATH/tools/requirements/requirements.core.txt

RUN python -m pip install -r $RMAKER_PATH/requirements.txt

RUN python $IDF_PATH/tools/idf_tools.py install cmake

RUN python $IDF_PATH/tools/idf_tools.py install --targets esp32s2

RUN python $IDF_PATH/tools/idf_tools.py install-python-env

ENV PATH="/usr/bin"

WORKDIR /Wireless-Controller-New/test

CMD . $IDF_PATH/export.sh && python $IDF_PATH/tools/idf.py set-target esp32s2 \ 
    && python $IDF_PATH/tools/idf.py menuconfig

### Build or installation Logs.

```plain
2023-05-19 17:36:22 Detecting the Python interpreter
2023-05-19 17:36:22 Checking "python3" ...
2023-05-19 17:36:22 Python 3.8.10
2023-05-19 17:36:22 "python3" has been detected
2023-05-19 17:36:22 Checking Python compatibility
2023-05-19 17:36:22 Checking other ESP-IDF version.
2023-05-19 17:36:23 Adding ESP-IDF tools to PATH...
2023-05-19 17:36:23 Checking if Python packages are up to date...
2023-05-19 17:36:23 Python requirements are satisfied.
2023-05-19 17:36:23 Requirement files:
2023-05-19 17:36:23  - /opt/esp/idf/tools/requirements/requirements.core.txt
2023-05-19 17:36:23 Python being checked: /opt/esp/python_env/idf5.0_py3.8_env/bin/python
2023-05-19 17:36:23 Added the following directories to PATH:
2023-05-19 17:36:23   
2023-05-19 17:36:23   /opt/esp/idf/components/espcoredump
2023-05-19 17:36:23   /opt/esp/idf/components/partition_table
2023-05-19 17:36:23   /opt/esp/idf/components/app_update
2023-05-19 17:36:23   /opt/esp/tools/xtensa-esp-elf-gdb/11.2_20220823/xtensa-esp-elf-gdb/bin
2023-05-19 17:36:23   /opt/esp/tools/riscv32-esp-elf-gdb/11.2_20220823/riscv32-esp-elf-gdb/bin
2023-05-19 17:36:23   /opt/esp/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin
2023-05-19 17:36:23   /opt/esp/tools/xtensa-esp32s2-elf/esp-2022r1-11.2.0/xtensa-esp32s2-elf/bin
2023-05-19 17:36:23   /opt/esp/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin
2023-05-19 17:36:23   /opt/esp/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin
2023-05-19 17:36:23   /opt/esp/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin
2023-05-19 17:36:23   /opt/esp/tools/cmake/3.24.0/bin
2023-05-19 17:36:23   /opt/esp/tools/openocd-esp32/v0.11.0-esp32-20221026/openocd-esp32/bin
2023-05-19 17:36:23   /opt/esp/python_env/idf5.0_py3.8_env/bin
2023-05-19 17:36:23   /opt/esp/idf/tools
2023-05-19 17:36:23 Done! You can now compile ESP-IDF projects.
2023-05-19 17:36:23 Go to the project directory and run:
2023-05-19 17:36:23 
2023-05-19 17:36:23   idf.py build
2023-05-19 17:36:23
```

### More Information.

If you need anything else, Ill see what I can do!

Josh-TapNoa, 2023-05-22:
Built on the esp-idf official docker image, no such error - Happy to blame docker and close if you guys are! Apologies in advance!

laukik-hase, 2023-05-24:
Hello, @Josh-TapNoa!

Closing this issue, feel free to reopen with more updates. Thanks.

closed 2023-05-24 by laukik-hase

benmelville, 2024-10-12:
Hello, @Josh-TapNoa 

I faced a similar issue and I have added it below.

```bash
Could not create symbolic link for: /Users/benjaminmelville/projects/esp/esp-idf/components/mbedtls/mbedtls/library/error.c
--> ln: /Users/benjaminmelville/projects/esp/esp-idf/examples/get-started/hello_world/cmake-build-debug/esp
idf/mbedtls/mbedtls/library/error.c: File exists
```

#### Solution
I was able to resolve this issue by wiping away my `cmake-build-debug` folder by running 
```bash
rm -fr cmake-build-debug
```
I then rebuilt the project and it worked. I hope this helps someone who comes across this issue in the future.