espressif/esp-idf #11446: install.bat fails: [WinError 5] Access is denied: libcrypto-1_1.dll

derrick-senva, 2023-05-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.2

### Operating System used.

Windows

### How did you build your project?

Other (please specify in More Information)

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

I expect the installation to finish without errors.

### What is the actual behavior?

The installation is halted when it encounters an "Access is denied" error.

### Steps to reproduce.

1. Clone ESP-IDF tag v5.0.2 to an esp-idf\ folder.
2. Run Windows cmd.exe as administrator.
3. Run install.bat within the esp-idf\ folder.

### Build or installation Logs.

```plain
c:\git\esp-idf>install.bat
Installing ESP-IDF tools
Current system platform: win64
Selected targets are: esp32s3, esp32c3, esp32, esp32h2, esp32s2, esp32c2
Installing tools: xtensa-esp-elf-gdb, riscv32-esp-elf-gdb, xtensa-esp32-elf, xtensa-esp32s2-elf, xtensa-esp32s3-elf, riscv32-esp-elf, esp32ulp-elf, cmake, openocd-esp32, ninja, idf-exe, ccache, dfu-util, esp-rom-elfs
Skipping xtensa-esp-elf-gdb@11.2_20220823 (already installed)
Skipping riscv32-esp-elf-gdb@11.2_20220823 (already installed)
Skipping xtensa-esp32-elf@esp-2022r1-11.2.0 (already installed)
Skipping xtensa-esp32s2-elf@esp-2022r1-11.2.0 (already installed)
Skipping xtensa-esp32s3-elf@esp-2022r1-11.2.0 (already installed)
Skipping riscv32-esp-elf@esp-2022r1-11.2.0 (already installed)
Skipping esp32ulp-elf@2.35_20220830 (already installed)
Skipping cmake@3.24.0 (already installed)
Skipping openocd-esp32@v0.11.0-esp32-20221026 (already installed)
Skipping ninja@1.10.2 (already installed)
Skipping idf-exe@1.0.3 (already installed)
Skipping ccache@4.6.2 (already installed)
Skipping dfu-util@0.9 (already installed)
Skipping esp-rom-elfs@20220823 (already installed)
Setting up Python environment
Creating a new Python environment in C:\Users\myuser\AppData\Local\Programs\Python\Python310\
Error: [WinError 5] Access is denied: 'C:\\Users\\myuser\\AppData\\Local\\Programs\\Python\\Python310\\DLLs\\libcrypto-1_1.dll'
Traceback (most recent call last):
  File "c:\git\esp-idf\tools\idf_tools.py", line 2610, in <module>
    main(sys.argv[1:])
  File "c:\git\esp-idf\tools\idf_tools.py", line 2602, in main
    action_func(args)
  File "c:\git\esp-idf\tools\idf_tools.py", line 2022, in action_install_python_env
    subprocess.check_call([sys.executable, '-m', 'venv',
  File "C:\Users\myuser\AppData\Local\Programs\Python\Python310\lib\subprocess.py", line 369, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['C:\\Users\\myuser\\AppData\\Local\\Programs\\Python\\Python310\\python.exe', '-m', 'venv', '--clear', '--upgrade-deps', 'C:\\Users\\myuser\\AppData\\Local\\Programs\\Python\\Python310\\']' returned non-zero exit status 1.
```

### More Information.

I'm running this on Windows 10. I originally had my Python 3.10 installation under "Program Files", but uninstalled it and reinstalled it for just my Windows user account so that it installed under Users\myuser\AppData\Local\Programs\Python\Python310\.

If I revert my ESP-IDF checkout to tag v5.0.1 and rerun install.bat, it works just fine.

dobairoland, 2023-05-22:
Do you have set `IDF_PYTHON_ENV_PATH` environment variable? Because the following doesn't seem right.

> Creating a new Python environment in C:\Users\myuser\AppData\Local\Programs\Python\Python310\

BTW, the recommended way is to use the Windows Tool Installer from https://dl.espressif.com/dl/esp-idf/ which would install a Python environment locally for ESP-IDF (not even the current Windows user will be affected). You will get the least number of potential issues with this one.

dobairoland, 2023-05-22:
> If I revert my ESP-IDF checkout to tag v5.0.1 and rerun install.bat, it works just fine.

That works just because the environment already exists and doesn't have to be created.

closed 2023-05-30 by derrick-senva

dobairoland, 2023-05-30:
I'm happy that it got solved. Thanks for sharing the results.