espressif/esp-idf #11448: ESP32-S3 GPIO3 high + GPIO0 high on boot enters `UART0_BOOT` mode

wuyuanyi135, 2023-05-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.1

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32-S3-WROOM-1-N16R2

### Power Supply used.

USB

### What is the expected behavior?

When the efuses are at default (confirmed by `espfuse summary`)
[image]

GPIO3 level should not affect boot.

### What is the actual behavior?

When GPIO3 and GPIO0 are high on boot, the device did not go into flashable bootloader, instead it prints
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x27 (UART0_BOOT)
Saved PC:0x40041a79
wait uart0 download
```
The boot register was decoded to: `GPIO3:1, GPIO45:0, GPIO0:0, GPIO46:1`

When GPIO3 is low, and GPIO0 is high on boot, it enters the correct bootloader state
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x3 (DOWNLOAD(USB/UART0))
Saved PC:0x40041a76
waiting for download
```
boot register: `GPIO3:0, GPIO45:0, GPIO0:0, GPIO46:1`.

Clearly, GPIO3:1 leads to the state I have never seen before (`UART0_BOOT`). I wonder
1. is there any documentation of `UART0_BOOT` for esp32-s3? is it a mode that only accept flashing from UART0? esptool will hang at `Connecting...` in this state.

2. why does the device enter this mode while the docs state that in this efuse configuration it should ignore GPIO3 on boot?

3. how to ignore GPIO3 on boot and enter USB-JTAG download mode whenever GPIO0 is low?

### Steps to reproduce.

1. pull the state of GPIO3 to high or low
2. pull GPIO0 low
3. reset

igrr, 2023-05-20:
I'm sorry, i will close this issue because the ROM bootloader and chip strapping pins behavior is not part of esp-idf project.

Please consider submitting documentation feedback using the link provided in the datasheet or the TRM, instead.

(There are a few additional boot modes: UART-only, USB_SERIAL_JTAG-only and SPI-only, which are not currently listed in the strapping pins table in the docs.)

closed 2023-05-20 by igrr

wuyuanyi135, 2023-05-20:
I believe this is then somehow triggered the undocumented UART only mode. Thanks