espressif/esp-idf #11451: Camera stops working after esp_restart()

icepick3000, 2023-05-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.1

### Operating System used.

macOS

### How did you build your project?

Other (please specify in More Information)

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32Cam

### Power Supply used.

USB

### What is the expected behavior?

When you issue an esp_Restart you expect the webcam to be still operational. 

### What is the actual behavior?

The webcam will no longer be functional until you recycle power. 

### Steps to reproduce.

1. Create a project that captures a webcam image in a loop
2. issue esp_restart()
3. The webcam will no longer work (sometimes after the first reboot but for sure after 2-3 reboots)

### Debug Logs.

```plain
06:12:58.765 -> [E][camera.c:1113] camera_probe(): Detected camera not supported.
06:12:58.765 -> [E][camera.c:1379] esp_camera_init(): Camera probe failed with error 0x20004
```

me-no-dev, 2023-06-12:
Hi @icepick3000 ! Sorry for the late reply. Looks like you are using an old version of the camera driver. `camera.c` have not existed for a while. Could you update the driver and check again? Error generally means that I2C communication with the camera failed.

Alvin1Zhang, 2023-08-07:
Thanks for reporting, will close due to short of feedback, feel to reopen with more updates. Thanks.

closed 2023-08-07 by Alvin1Zhang