espressif/esp-idf #11453: ESP32-S3无法连接WIFI6路由器

king592, 2023-05-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

请问一下，ESP32-S3是否能连接WIFI6的路由器呢？我使用ESPTOUCH配网无法连接，使用WIFI5的路由器是可以成功连接的

xuxiao111, 2023-05-22:
@king592  可以提供一下连接失败的log嘛

king592, 2023-05-22:
没有抓取到日志，就是ESP32收不到配网信息的广播，正常收到广播我是会打印出WIFI的账号和密码的，但是连接WIFI6的路由上是没有日志内容的，也就是说没有收到广播信息

xuxiao111, 2023-05-22:
可以用 example/wifi/iperf 例程测试一下嘛，验证一下是否是S3连接不上WIFI6路由器。如果可以成功连上，可以通过无线网卡抓取一下手机发送的配网信息的广播，查看是否是没有发送出来导致的无法连接。

xuxiao111, 2023-06-15:
@king592  hello，问题还会复现嘛

closed 2023-08-07 by Alvin1Zhang