espressif/esp-idf #11456: ESP32-S3-WROOM-1-N8R8 External PSRAM error with some modules

misiek1994, 2023-05-20:
Hello,
We're almost done with writing application for our device with lcd screen. During this process we had a problem with one ESP32-S3-WROOM-1-N8R8 module (from total of 4 or 5 modules) - PSRAM wasn't working. I thought it's nothing special - maybe single module damaged by accident (soldering, esd etc.).
This week we bought 5pcs of exactly the same modules on one tape to start ultra-small batch for client. We were suprised by first device - PSRAM won't work. Today I managed to solder another device with a exactly same components and with a ESP32 module from the same tape - PSRAM works.
So I have two modules - A not working and B working. Everything what I'll describe next was tested on B and works! 

1. As our application won't start we get back to basics. New application - hello world. Compile without psram - works.
2. Same application - in SDK config we added only: support for external RAM, octal mode, 40MHz, integrate using malloc according to https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/flash_psram_config.html. PSRAM doesn't work.
3. Some test with: 80MHz, octal/quad, RAM access method, type of ram (change from auto-detect to esp-psram...) - PSRAM still doesn't work.
4. Tested on esp-idf 4.4.3 and 5.0.2.

Logs:
4.4.3:
I (24) boot: ESP-IDF v4.4.3-dirty 2nd stage bootloader
I (25) boot: compile time 20:09:17
I (25) boot: chip revision: 0
I (27) boot.esp32s3: Boot SPI Speed : 80MHz
I (31) boot.esp32s3: SPI Mode       : DIO
I (36) boot.esp32s3: SPI Flash Size : 2MB
I (41) boot: Enabling RNG early entropy source...
I (46) boot: Partition Table:
I (50) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (72) boot:  2 factory          factory app      00 00 00010000 00100000
I (80) boot: End of partition table
I (84) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=07fd8h ( 32728) map
I (98) esp_image: segment 1: paddr=00018000 vaddr=3fc912f0 size=029cch ( 10700) load
I (103) esp_image: segment 2: paddr=0001a9d4 vaddr=40374000 size=05644h ( 22084) load
I (114) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=178e0h ( 96480) map
I (135) esp_image: segment 4: paddr=00037908 vaddr=40379644 size=07ca8h ( 31912) load
I (142) esp_image: segment 5: paddr=0003f5b8 vaddr=50000000 size=00010h (    16) load
I (148) boot: Loaded app from partition at offset 0x10000
I (148) boot: Disabling RNG early entropy source...
I (163) opi psram: vendor id : 0x00 (UNKNOWN)
I (164) opi psram: dev id    : 0x03 (generation 4)
I (164) opi psram: density   : 0x07 (256 Mbit)
I (168) opi psram: good-die  : 0x01 (Pass)
I (173) opi psram: Latency   : 0x01 (Fixed)
I (177) opi psram: VCC       : 0x01 (3V)
I (182) opi psram: SRF       : 0x00 (Slow Refresh)
I (188) opi psram: BurstType : 0x01 ( Wrap)
I (192) opi psram: BurstLen  : 0x03 (1024 Byte)
I (198) opi psram: Readlatency  : 0x07 (20 cycles@Fixed)
I (204) opi psram: DriveStrength: 0x03 (1/8)
I (209) spiram: Found 256MBit SPI RAM device
I (213) spiram: SPI RAM mode: sram 40m
I (218) spiram: PSRAM initialized, cache is in normal (1-core) mode.

5.0.2:
I (24) boot: ESP-IDF v5.0.2-dirty 2nd stage bootloader
I (24) boot: compile time 20:29:31
I (24) boot: chip revision: v0.1
I (26) boot.esp32s3: Boot SPI Speed : 80MHz
I (31) boot.esp32s3: SPI Mode       : DIO
I (36) boot.esp32s3: SPI Flash Size : 8MB
I (41) boot: Enabling RNG early entropy source...
I (46) boot: Partition Table:
I (50) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (64) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (72) boot:  2 factory          factory app      00 00 00010000 00100000
I (79) boot: End of partition table
I (84) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=0981ch ( 38940) map
I (99) esp_image: segment 1: paddr=00019844 vaddr=3fc92100 size=02a84h ( 10884) load
I (103) esp_image: segment 2: paddr=0001c2d0 vaddr=40374000 size=03d48h ( 15688) load
I (112) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=19e20h (106016) map
I (136) esp_image: segment 4: paddr=00039e48 vaddr=40377d48 size=0a340h ( 41792) load
I (152) boot: Loaded app from partition at offset 0x10000
I (152) boot: Disabling RNG early entropy source...
E (163) octal_psram: PSRAM ID read error: 0x00000000, PSRAM chip not found or not supported, or wrong PSRAM line mode
E (164) cpu_start: Failed to init external RAM!

It's super-duper strange - no vendor ID, density 0x7 - 32MB. However to be honest for me these aren't random numbers! If anyone could quite fast tell me what the heck exactly happened I would be really grateful. Till the May 28th we have to send working devices to our client so it would be nice to know if they are reliable.

Waiting for tips and suggestions.

Icarus113, 2023-05-22:
**Difference between v4.4.3 and v5.0.2**

The log on v4.4.3 is because lack of this [change](https://github.com/espressif/esp-idf/commit/bf6ca71630e4e4ed5ac9f577a60725ce35df3ebd#diff-57f978ee52d6e6ebf9e2778c92bac383daf21e9853601ce949cab0f6eaaa283a) I think. The condition is the driver code doesn't recognize the expected vendor ID, but it doesn't fail at this time and the program runs further.

With this change on v5.0.2, the error log is expected and it fails at this time.

**About the failure**
So it's actually the same failure among these two releases.

This failure is weird as we've done tests for this ESP32-S3-WROOM-1 module. Are you using devboards from Espressif? Or do you muliplex the MSPI pins?

@misiek1994

misiek1994, 2023-05-22:
1. I know that logs are the same. I just want to make sure that on newest 5.0.2 versions error still occurs.
2. I think that 33-37 IOs should not be connected. And I did it correctly I guess. Part of schematic in attachments.
3. 2 out of 6 chips we tried had such a problem. The rest work fine. I will double check soldering (maybe resolder) but It was hand-soldered so I think there is really tiny chance that this is the problem.
[image]

Icarus113, 2023-05-22:
Apart from IO33-37, 
IO 26, 27, 28, 30, 31, 32 are also used for accessing the main SPI Flash and the PSRAM. You may check these as well.

Icarus113, 2023-06-01:
@misiek1994 Hi, may I know if there's any update?

misiek1994, 2023-06-05:
Yes, it was soldering issue. However I'm not sure if it was internal in module or in our PCB, because we had to use preheater + hotair. We haven't seen any obvious shorts.
At the moment we are testing back other PCBs, because we are pretty sure there was another PSRAM error.
It is quite funny that Flash worked correctly, but not RAM. The reason is ofc quad/octal :)
I think that we can close this topic.

closed 2023-06-05 by misiek1994

zhaochengwei, 2024-04-27:
打开menuconfig取消PSRAM配置