espressif/esp-idf #11458: 请教关于ESP32 ble mesh provisioner示例给onoff cilent配网的问题

zyfl88, 2023-05-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.1-dirty

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32-DevKitC

### Power Supply used.

USB

### What is the expected behavior?

我想通过ble_mesh_provisioner示例给onoff_cilent和onoff_sever配网，以实现onoff_cilent给onoff_sever发送控制指令消息的功能。

### What is the actual behavior?

通过对以上示例的烧录运行后发现ble_mesh_provisioner示例给onoff_cilent配网后，onoff_cilent无法成功发送消息，报错：BLE_MESH: Model not bound to AppKey 0x0001，可是我使用手机APP EspBleMesh或者nRF Mesh给其配网时，消息均能发送成功。

### Steps to reproduce.

1. Step
[ble_mesh_provisioner.zip](https://github.com/espressif/esp-idf/files/11523631/ble_mesh_provisioner.zip)

2. Step
[onoff_client.zip](https://github.com/espressif/esp-idf/files/11523635/onoff_client.zip)

3. Step
...

### Debug Logs.

```plain
I (621) EXAMPLE: Initializing...
I (621) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1341) EXAMPLE_NVS: Restore, key "onoff_client", length 6
I (1341) EXAMPLE_NVS: Restore, data: 00 00 01 00 01 47
I (1351) EXAMPLE: Restore, net_idx 0x0000, app_idx 0x0001, onoff 1, tid 0x47
I (1361) EXAMPLE: BLE Mesh Node initialized
I (1371) EXAMPLE: ESP_BLE_MESH_NODE_PROV_ENABLE_COMP_EVT, err_code 0
I (1471) EXAMPLE: ESP_BLE_MESH_NODE_PROV_LINK_OPEN_EVT, bearer PB-ADV
I (680241) BOARD: tap cb (RELEASE)
E (680241) BLE_MESH: Model not bound to AppKey 0x0001
E (680241) BLE_MESH: Failed to send client message 0x00008203
I (680251) EXAMPLE_NVS: Store, key "onoff_client", length 6
I (680251) EXAMPLE_NVS: Store, data: 00 00 01 00 00 48
I (680261) EXAMPLE: Generic client, event 1, error code -22, opcode is 0x8203
I (680271) EXAMPLE: ESP_BLE_MESH_GENERIC_CLIENT_SET_STATE_EVT
```

zyfl88, 2023-05-23:
哪位大佬帮忙指导下呢？感激不尽啊

ghoshaljayant, 2023-10-06:
Hi I am facing the similar problem here.

I burned 
2 esp32-devkit with [onoff_client](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/esp_ble_mesh/onoff_models/onoff_client)
1 esp32-devkit with [onoff_server](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/esp_ble_mesh/onoff_models/onoff_server)
1 esp32-devkit with [provisioner](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/esp_ble_mesh/provisioner)

now i boot onoff_client and onoff_server first
secondly i boot the provisioner.

Observation in onoff_server logs
- provision completed and app key also got bind        :  ESP_BLE_MESH_MODEL_OP_MODEL_APP_BIND

Observation in onoff_client logs
- provision completed but no app key got bind

Observation in provisioner logs
- it provisioned 3 devices.

now if i press on/off button from onoff_client, logs says
`BLE_MESH: Model not bound to AppKey 0x0001`