espressif/esp-idf #11459: Problem of high ping value when connecting TCP clien

krxiaoyu, 2023-05-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Official Code: D:\Espressif\frameworks\esp-idf-v5.0.2\examples\protocols\sockets\tcp_client
Device: ESP32C3
Router: ASUS router, only 1 meter away from ESP32C3
Demand:
The ping value is very high.Delay exceeded 300ms. How to optimize it to reduce the ping value? For example, reach within 100ms or 50ms.
[image]

david-cermak, 2023-10-10:
@krxiaoyu You can check the IDF documentation on performance optimization and WiFi buffer usage, and `WIFI_PS_NONE` 

* https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/lwip.html#performance-optimization
* https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html#wifi-buffer-usage
* https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html#esp32-wi-fi-power-saving-mode

MaxwellAlan, 2024-03-04:
Hi @krxiaoyu 

We will close this issue since so long no response.

Please feel free to reopen if still have this issue.

Thanks.

closed 2024-03-04 by MaxwellAlan