espressif/esp-idf #11460: idf,py gdbgui does not work

Mair, 2023-05-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.x

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

gdbgui should launch

### What is the actual behavior?

an error message states
`Unfortunately, gdbgui is supported only with Python 3.10 or older. See: https://github.com/espressif/esp-idf/issues/10116. Please use "idf.py gdb" or debug in Eclipse/Vscode instead`

### Steps to reproduce.

download version 5
flash 
idf.py openocd
idf.py gdbgui

### Build or installation Logs.

```plain
PS C:\esp\esp-idf-v5.0\examples\get-started\hello_world> idf.py openocd gdbgui
Executing action: openocd
Note: OpenOCD cfg not found (via env variable OPENOCD_COMMANDS nor as a --openocd-commands argument)
OpenOCD arguments default to: "-f board/esp32-wrover-kit-3.3v.cfg"
OpenOCD started as a background task 9388
Executing action: gdbgui
['gdbgui', '-g', 'xtensa-esp32-elf-gdb', '--gdb-args', '"-x=C:/esp/esp-idf-v5.0/examples/get-started/hello_world/build\\gdbinit\\gdbinit"']
[WinError 2] The system cannot find the file specified
Unfortunately, gdbgui is supported only with Python 3.10 or older. See: https://github.com/espressif/esp-idf/issues/10116. Please use "idf.py gdb" or debug in Eclipse/Vscode instead.
PS C:\esp\esp-idf-v5.0\examples\get-started\hello_world>
```

dobairoland, 2023-05-21:
In addition to the above message and link, I can offer https://github.com/cs01/gdbgui/issues/447 as well. Unfortunately, nothing can be done at this moment.

dobairoland, 2023-05-21:
> https://www.gdbgui.com/installation/#windows-dependencies
> Note that windows is only supported for gdbgui versions less than 0.14.

Therefore, I don't think there is a chance that gdbgui would work again on Windows especially with newer Python versions.

Mair, 2023-05-21:
that sucks 😞

dobairoland, 2023-05-22:
Here are some of your options:
- Use an older version of the Windows Tool Installer which creates a 3.8 Python environment. You can use the latest version of any supported ESP-IDF branches and `idf.py gdbgui` will work for you.
- Use `idf.py gdb`.
- Use the debugging features of Eclipse provided by https://github.com/espressif/idf-eclipse-plugin.
- Use the debugging features of Vscode provided by https://github.com/espressif/vscode-esp-idf-extension.

closed 2023-05-22 by dobairoland