espressif/esp-idf #11461: [v5.1] "task_wdt: esp_task_wdt_reset(783): task not found" logged a lot at boot time

chipweinberger, 2023-05-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

release/v5.1

### Operating System used.

macOS

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3 Dev Kit C

### Power Supply used.

USB

### What is the expected behavior?

not sure why this is logged so many times

### What is the actual behavior?

lots of logs

### Steps to reproduce.

fdfdf

### Debug Logs.

```plain
E (4678) task_wdt: esp_task_wdt_reset(783): task not found
E (4678) task_wdt: esp_task_wdt_reset(783): task not found
E (4678) task_wdt: esp_task_wdt_reset(783): task not found
E (4678) task_wdt: esp_task_wdt_reset(783): task not found
E (4678) task_wdt: esp_task_wdt_reset(783): task not found
E (4688) task_wdt: esp_task_wdt_reset(783): task not found
E (4688) task_wdt: esp_task_wdt_reset(783): task not found
E (4688) task_wdt: esp_task_wdt_reset(783): task not found
E (4688) task_wdt: esp_task_wdt_reset(783): task not found
E (4688) task_wdt: esp_task_wdt_reset(783): task not found
E (4688) task_wdt: esp_task_wdt_reset(783): task not found
E (4698) task_wdt: esp_task_wdt_reset(783): task not found
E (4698) task_wdt: esp_task_wdt_reset(783): task not found
E (4698) task_wdt: esp_task_wdt_reset(783): task not found
E (4698) task_wdt: esp_task_wdt_reset(783): task not found
E (4698) task_wdt: esp_task_wdt_reset(783): task not found
E (4698) task_wdt: esp_task_wdt_reset(783): task not found
E (4698) task_wdt: esp_task_wdt_reset(783): task not found
E (4708) task_wdt: esp_task_wdt_reset(783): task not found
E (4708) task_wdt: esp_task_wdt_reset(783): task not found
```

ESP-Marius, 2023-05-23:
@chipweinberger do you have any way of helping us reproduce this? E.g. a simple example or something similar?

chipweinberger, 2023-05-23:
Not yet, but I'll investigate it =) Thanks for the reply.

ESP-Marius, 2023-05-23:
@chipweinberger Thanks! I remember seeing this before in https://github.com/espressif/esp-idf/issues/10314, but at that time we never traced it to an actual bug.

Lets see if we can track it down when you have some time for investigating it.

chipweinberger, 2023-05-23:
heres the backtrace. idle task hook.

```
E (5499) task_wdt: esp_task_wdt_reset(797): task not found

Backtrace: 0x420C956F:0x3FCB77D0 0x420C95E5:0x3FCB7810 0x42006DA6:0x3FCB7840 0x40387975:0x3FCB7870 0x40386092:0x3FCB78A0
0x420c956f: esp_task_wdt_reset at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esp_system/task_wdt/task_wdt.c:794

0x420c95e5: idle_hook_cb at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esp_system/task_wdt/task_wdt.c:497

0x42006da6: esp_vApplicationIdleHook at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esp_system/freertos_hooks.c:47 (discriminator 1)

0x40387975: prvIdleTask at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:4327

0x40386092: vPortTaskWrapper at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162
```

chipweinberger, 2023-05-23:
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/11540390/sdkconfig.txt)

o-marshmallow, 2023-05-25:
Hi @chipweinberger ,

Does this error often reproduce or is it random? By any chance, is your code calling inadvertently one of these functions: `esp_task_wdt_deinit`, `esp_task_wdt_reconfigure` or `esp_task_wdt_delete`? Could you try tracing/logging these functions when called?

The error happens when the Idle task address cannot be found anymore in the TWDT entry list `p_twdt_obj->entries_slist`. If none of the functions above are called, this could also be caused by a memory corruption.

chipweinberger, 2023-05-25:
Yes I do call both:
- esp_task_wdt_delete(xTaskGetIdleTaskHandleForCPU(0)));
- esp_task_wdt_delete(xTaskGetIdleTaskHandleForCPU(1)));
- esp_task_wdt_add(xTaskGetIdleTaskHandleForCPU(0)));
- esp_task_wdt_add(xTaskGetIdleTaskHandleForCPU(1)));

I call them on purpose to avoid WDT during long running operations. For example:
- esp_ds_sign
- mbedtls_rsa_gen_key
- mount_sdmmc
- f_getfree

Undoubtedly, this happens when I mount my sd card at boot time. 

But why does it log in v5.1 and not v4.4?

And if I disabled the idle twdt, why is it still running and logging?

o-marshmallow, 2023-05-25:
@chipweinberger Thanks for the details.

It is clearer now, the problem comes from the fact that between the moment you call `esp_task_wdt_delete(xTaskGetIdleTaskHandleForCPU(x))` and the moment you call back `esp_task_wdt_add(xTaskGetIdleTaskHandleForCPU(x)));`, if the Idle task is run, you will get this issue because it tries to run its callback that resets the WDT (`idle_hook_cb`).

This happens now, in v5.1, since some changes in mmc code make it blocking by calling `vTaskDelay`, which will give the hand to the Idle task. While this may not be present in former versions, such as v4.4, you should **not** make the assumption that IDF functions (including `esp_ds_sign`, `mbedtls_rsa_gen_key`, `mount_sdmmc`, `f_getfree`) won't give back the hand to the Idle task by calling `vTaskDelay` or any other blocking operation.

I see three solutions here:
* Instead of deleting the Idle taks from the TWDT, reconfigure the latter so that it won't watch the Idle tasks, and unsubscribe the callbacks:
```
esp_task_wdt_config_t config = {
    .timeout_ms = 60000,
    .trigger_panic = true,
    .idle_core_mask = 0, // i.e. do not watch any idle task
};
esp_err_t err = esp_task_wdt_reconfigure(&config);
```
* Disable watching Idle core taks from the `menuconfig` and register your own task to watch thanks to `esp_task_wdt_add()` and `esp_task_wdt_reset()` (to feed), or your own entities with `esp_task_wdt_add_user()` and `esp_task_wdt_reset_user()`
* Simply disable the TWDT from the menuconfig: if you cannot guarantee that your tasks will run within a certain delay, how to tell if the chip is stuck or if a heavy task is going on.

chipweinberger, 2023-05-25:
@o-marshmallow , thank you for such a detailed reply!!!

> reconfigure the latter so that it won't watch the Idle tasks

I will do this approach!

chipweinberger, 2023-05-26:
That fixed it!

closed 2023-05-26 by chipweinberger

o-marshmallow, 2023-05-26:
@chipweinberger Thanks for coming back with the confirmation, happy to hear it worked for you!

chriscckan, 2023-09-25:
hi, I have the same error "E (12377) task_wdt: esp_task_wdt_reset(783): task not found" when I start my changed in code to load SD card wave file for continuous playback.  Previously I play audio data from ROM, recently I put wave file in SD card and load the file from SD to playback and it would have this error randomly and then reboot.
I have a separate task (not inside app_main) for accessing wave file on SD card and each time I load 2048byte to a global variable area (not malloc) and write it to i2s buffer, vtaskdelay for 30ms and then load next 2048bytes again, how can I further debug this problem and avoid trigger the error?

norventa, 2024-02-27:
Hi, I think there may be a small bug in esp_task_wdt_reset. I added a bit of debug code to the bottom of this function and it reported the offending task as: **IDLE1**(3FFC1EAC). I don't think IDLE1 should reset TWDT or am I missing something.

`esp_err_t
esp_task_wdt_reset(void)
{
    ESP_RETURN_ON_FALSE(p_twdt_obj != NULL, ESP_ERR_INVALID_STATE, TAG, "TWDT was never initialized");
    esp_err_t ret;
    TaskHandle_t handle = xTaskGetCurrentTaskHandle();

    portENTER_CRITICAL(&spinlock);
    // Find entry from task handle
    bool all_reset;
    twdt_entry_t *entry;
    entry = find_entry_from_task_handle_and_check_all_reset(handle, &all_reset);
    ESP_GOTO_ON_FALSE_ISR((entry != NULL), ESP_ERR_NOT_FOUND, err, TAG, "task not found");
    // Mark entry as reset and issue timer reset if all entries have been reset
    entry->has_reset = true; // Reset the task if it's on the task list
    if (all_reset)
    { // Reset if all other tasks in list have reset in
        task_wdt_timer_feed();
    }
    ret = ESP_OK;

    // My modification to get the name of the task.
    portEXIT_CRITICAL(&spinlock);

    return ret;
err:

    TaskStatus_t status;

    vTaskGetInfo(handle, &status, pdFALSE, eInvalid);
    ESP_EARLY_LOGE(TAG, "Task: %s(%08lX)", status.pcTaskName, handle);
    portEXIT_CRITICAL(&spinlock);

    return ret;
}`

norventa, 2024-02-28:
That is interesting. So why does find_entry_from_task_handle_and_check_all_reset return ESP_ERR_NOT_FOUND for IDLE1?

 

Both idle tasks should reset that TWDT yes, this is how we monitor if any task is starving the system. More info here <https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/system/wdts.html#task-watchdog-timer-twdt>

norventa, 2024-03-01:
So by disabling the idle tasks as suggested, (esp_task_wdt_config_t.idle_core_mask = 0) can cause problems as I have just discovered. I have a task that updates a fat partition using, among other things, fopen, fwrite and unlink. It sets esp_task_wdt_config_t.idle_core_mask = 0, as it takes a bit of time to complete. With all the other tasks used blocking and, I am guessing, fwrite blocking, the function prvSelectHighestPriorityTaskSMP issues a panic as there are no tasks ready to run.

ESP-Marius, 2024-03-01:
> I should have said that esp_task_wdt_config_t.idle_core_mask has been set to 0 when this happens.

Ah, yeah then I agree that it should not be monitoring the idle task. 

Can you explain a bit more about how you are doing this? Are you able to provide an small example to us that can reproduce the issue you are experiencing?

norventa, 2024-03-04:
This happens in an app I am writing, which uses at least 20 tasks, so isn't easy to use to demostrate the problem. However I have put together a bit of code that exhibits the fault. At least I hope it does:

`// Test disabling idle tasks' WDT.
//

#include <esp_log.h>
#include <esp_timer.h>
#include <esp_task_wdt.h>
#include <freertos/freertos.h>

static const char c_szTag[] = "WDT_TEST";

static esp_task_wdt_config_t s_WDTConfig = {
.timeout_ms     = 5 * 1000,
.idle_core_mask = 3,
.trigger_panic  = true,
};

static int64_t s_NextUpdate;

static void
taskFn(void *)
{
    s_NextUpdate = esp_timer_get_time();

    while (true)
    {
        if (esp_timer_get_time() >= s_NextUpdate)
        {
            s_NextUpdate += 1000000;
        }
    }
}

// Disable the idle tasks' watchdog timers.
//
static void
watchdogDisable()
{
    if (s_WDTConfig.idle_core_mask == 0)
        return;

    s_WDTConfig.idle_core_mask = 0;
    esp_task_wdt_reconfigure(&s_WDTConfig);
}

// Enable the idle tasks' watchdog timers.
//
static void
watchdogEnable()
{
    if (s_WDTConfig.idle_core_mask != 0)
        return;

    s_WDTConfig.idle_core_mask = 3;
    esp_task_wdt_reconfigure(&s_WDTConfig);
}

void
app_main(void)
{
    TaskHandle_t taskHandles[50];

    for (int t = 0; t < 100; ++t)
    {
        watchdogDisable();

        for (int i = 0; i < 50; ++i)
        {
            char szTag[8];

            sprintf(szTag, "Task%d", i + 1);
            xTaskCreatePinnedToCore(taskFn, szTag, 1024, 0, 15, &taskHandles[i], 1);
        }

        watchdogEnable();

        for (int i = 49; i >= 0; --i)
            vTaskDelete(taskHandles[i]);
    }

    ESP_LOGI(c_szTag, "Count: %lld", s_NextUpdate);
}
`

norventa, 2024-03-04:
I believe there is a problem with the function vTaskSwitchContext in tasks.c. The line at the end:
`configASSERT( xTaskScheduled == pdTRUE ); /* At this point, a task MUST have been scheduled */`
can throw a panic when the idle tasks are disabled. All other tasks could be blocking. So it is valid to have no task ready to run.

A suggestion is to start a no affinity task that does nothing when disabling the idle tasks' WDTs of priority 0, and deleting it when enabling them. At least, that has worked for me. I can't be sure esp_task_wdt_reset won't complain about the idle tasks but it hasn't complained yet for me.

ESP-Marius, 2024-03-05:
> can throw a panic when the idle tasks are disabled.

But why would the idle task be disabled? I dont think there is any case where this is true is there? 

> a bit of code that exhibits the fault. 

Hmm, seemed to run fine when testing on S3 for me

> In the function esp_task_wdt_reconfigure all 'goto err' skip the call to portEXIT_CRITICAL, which is wrong.

Good find, this is indeed a bug! We'll fix that.

norventa, 2024-03-05:
Also in esp_task_wdt_reconfigure, should err be returned rather than ESP_OK?
And should the timer be restarted?

norventa, 2024-03-05:
<blockquote>
<p dir="auto">a bit of code that exhibits the fault.</p>
</blockquote>

Yes, it ran fine today but I upped the outer loop from 100 to 1000 and below was output on the serial monitor (ESP32-WROOM). I ran it a few times and a different number of `esp_task_wdt_reset(801): task not found` were output.

I (357) esp_apptrace: Initialized TRAX on CPU0
I (101) esp_apptrace: Initialized TRAX on CPU1
I (362) main_task: Started on CPU0
I (371) main_task: Calling app_main()
E (583) task_wdt: esp_task_wdt_reset(801): task not found
E (2846) task_wdt: esp_task_wdt_reset(801): task not found
I (3621) WDT_TEST: Count: 4273011
I (3622) main_task: Returned from app_main()

norventa, 2024-03-05:
> can throw a panic when the idle tasks are disabled.

What I mean there is esp_task_wdt_config_t.idle_core_mask is set to zero. The idle tasks aren't really tasks are they? They are callbacks from a hardware timer. Or at least that is what I think they are. I'm hoping when the 'task not found' problem is cured this will also be.
It doesn't happen when I introduce a 'do nothing' task that simply calls taskYIELD() within a forever loop.
For prvSelectHighestPriorityTaskSMP in tasks.c to not reach the configASSERT(xTaskScheduled == pdTRUE ) at its end, there must be at least one task that doesn't block. That is true isn't it?

norventa, 2024-03-05:
That is peculiar. My app throws a panic from prvSelectHighestPriorityTaskSMP. How is that possible?

ESP-Marius, 2024-03-05:
> How is that possible?

Dont think it should ever happen, hence there being an assert there.

Could it simply be a stackoverflow that is messing things up? I noticed in the example you showed you used `1024`, which is probably too small

norventa, 2024-03-05:
1024 was just for the example. I have canarys to catch stack overflow. If I have a 'do nothing' task the panic goes away.

norventa, 2024-03-06:
Back to prvSelectHighestPriorityTaskSMP in tasks.c. If a task is currently being executed it won't be scheduled to run. When my app panics at:
configASSERT(xTaskScheduled == pdTRUE); /* At this point, a task MUST have been scheduled */ 
Adding a call to vTaskList just before that line shows IDLE0 is being executed (X) and xCurCoreID == 0.
So my question is: what prevents this situation happening?

ESP-Marius, 2024-03-07:
@norventa do you think you could split out another issue for this assert problem? Then it will be easier for any other user that has the same problem to find it.

dzungpv, 2024-03-07:
@norventa I have the same issue with Arduino Core 3.0 and IDF 5.1, do you have any solution?. This should keep open.

norventa, 2024-03-07:
Yes Marius, I will file this issue in a new thread. Thanks for your help.