espressif/esp-idf #11462: [v5.1][USB Host] usb_host_install returns ESP_ERR_NOT_FOUND

chipweinberger, 2023-05-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

release/v5.1

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

This used to work on v4.4.4 without error. 

### What is the actual behavior?

usb_host_install() returns ESP_ERR_NOT_FOUND

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Build or installation Logs.

```plain
I (12748) pd midi usb: init
I (12748) pd midi usb: xTaskCreate midi_usb_open_loop
I (12748) pd midi usb: midi_usb_open_loop task started
I (12748) pd usbh: oneTimeInit()
I (12748) pd usbh: [oneTimeInit] create xUsbhInitializeMutex
I (12748) pd usbh: [oneTimeInit] create xOpenedDeviceHandlesMutex
I (12748) pd usbh: [oneTimeInit] create queue xDeviceConnectedEvents
I (12748) pd main: Waiting 1.5 seconds for midi usb task to start
I (13748) pd usbh: [openSingleDevice] called
I (13748) pd usbh: [openSingleDevice] all devices closed. continuing...
I (13748) pd usbh: [openSingleDevice] initializing usbh
I (13748) pd usbh: [init] initialize
I (13748) pd usbh: [init] got mutex
I (13748) pd usbh: [init] xSemaphoreCreateRecursiveMutex() xClaimedInterfacesMutex
I (13748) pd usbh: [init] usb_host_install()
W (13748) pd usbh: [init] failed
I (13758) pd alerts: notify issue
W (13758) pd issues: RAISE - [Usb Device] Init Failure | None
E (13758) : e1 : [pd usbh] usb_host_install() failed. ESP_ERR_NOT_FOUND
e2 : [pd usbh] initialize failed
e3 : [pd midi usb] openSingleDevice() failed
```

chipweinberger, 2023-05-21:
The error happens here:

```
    esp_err_t intr_alloc_ret = esp_intr_alloc(ETS_USB_INTR_SOURCE,
                                              config->intr_flags | ESP_INTR_FLAG_INTRDISABLED,  //The interrupt must be disabled until the port is initialized
                                              intr_hdlr_main,
                                              (void *)p_hcd_obj_dmy->port_obj,
                                              &p_hcd_obj_dmy->isr_hdl);
```

chipweinberger, 2023-05-21:
These are the debug logs. I can see I am already using interrupts for:

- i2c
- gpio_install_isr_service
- sdcard detection (i think related to gpio_install_isr_service)
- something related to "Usb Console: Install Driver"
- uart

```
W (14810) intr_alloc: get_available_int: try to find existing. Cpu: 0, Source: 38
W (14810) intr_alloc: get_free_int: start looking. Current cpu: 0
W (14810) intr_alloc: Int 0 reserved 2 priority 1 LEVEL hasIsr 0
W (14810) intr_alloc: ....Unusable: reserved
W (14810) intr_alloc: Int 1 reserved 2 priority 1 LEVEL hasIsr 0
W (14810) intr_alloc: ....Unusable: reserved
W (14810) intr_alloc: Int 2 reserved 0 priority 1 LEVEL hasIsr 1
W (14810) intr_alloc: ....Unusable: already in (non-shared) use.
W (14810) intr_alloc: Int 3 reserved 0 priority 1 LEVEL hasIsr 1
W (14810) intr_alloc: ....Unusable: already in (non-shared) use.
W (14810) intr_alloc: Int 4 reserved 2 priority 1 LEVEL hasIsr 0
W (14810) intr_alloc: ....Unusable: reserved
W (14810) intr_alloc: Int 5 reserved 2 priority 1 LEVEL hasIsr 1
W (14810) intr_alloc: ....Unusable: reserved
W (14810) intr_alloc: Int 6 reserved 0 priority 1 EDGE hasIsr 0
W (14810) intr_alloc: ....Unusable: special-purpose int
W (14810) intr_alloc: Int 7 reserved 0 priority 1 EDGE hasIsr 0
W (14810) intr_alloc: ....Unusable: special-purpose int
W (14810) intr_alloc: Int 8 reserved 2 priority 1 LEVEL hasIsr 1
W (14810) intr_alloc: ....Unusable: reserved
W (14810) intr_alloc: Int 9 reserved 0 priority 1 LEVEL hasIsr 1
W (14810) intr_alloc: ....Unusable: already in (non-shared) use.
W (14810) intr_alloc: Int 10 reserved 0 priority 1 EDGE hasIsr 0
W (14810) intr_alloc: ....Unusable: incompatible trigger type
W (14810) intr_alloc: Int 11 reserved 0 priority 3 EDGE hasIsr 0
W (14810) intr_alloc: ....Unusable: special-purpose int
W (14810) intr_alloc: Int 12 reserved 0 priority 1 LEVEL hasIsr 1
W (14810) intr_alloc: ....Unusable: already in (non-shared) use.
W (14810) intr_alloc: Int 13 reserved 0 priority 1 LEVEL hasIsr 1
W (14810) intr_alloc: ....Unusable: already in (non-shared) use.
W (14810) intr_alloc: Int 14 reserved 2 priority 7 LEVEL hasIsr 0
W (14810) intr_alloc: ....Unusable: reserved
W (14810) intr_alloc: Int 15 reserved 0 priority 3 EDGE hasIsr 0
W (14810) intr_alloc: ....Unusable: special-purpose int
W (14810) intr_alloc: Int 16 reserved 0 priority 5 EDGE hasIsr 0
W (14810) intr_alloc: ....Unusable: special-purpose int
W (14810) intr_alloc: Int 17 reserved 0 priority 1 LEVEL hasIsr 1
W (14810) intr_alloc: ....Unusable: already in (non-shared) use.
W (14810) intr_alloc: Int 18 reserved 0 priority 1 LEVEL hasIsr 1
W (14810) intr_alloc: ....Unusable: already in (non-shared) use.
W (14810) intr_alloc: Int 19 reserved 0 priority 2 LEVEL hasIsr 1
W (14810) intr_alloc: ....Unusable: incompatible priority
W (14810) intr_alloc: Int 20 reserved 0 priority 2 LEVEL hasIsr 1
W (14810) intr_alloc: ....Unusable: incompatible priority
W (14810) intr_alloc: Int 21 reserved 0 priority 2 LEVEL hasIsr 1
W (14810) intr_alloc: ....Unusable: incompatible priority
W (14810) intr_alloc: Int 22 reserved 2 priority 3 EDGE hasIsr 0
W (14810) intr_alloc: ....Unusable: reserved
W (14810) intr_alloc: Int 23 reserved 0 priority 3 LEVEL hasIsr 0
W (14810) intr_alloc: ....Unusable: incompatible priority
W (14810) intr_alloc: Int 24 reserved 2 priority 4 LEVEL hasIsr 0
W (14810) intr_alloc: ....Unusable: reserved
W (14810) intr_alloc: Int 25 reserved 2 priority 4 LEVEL hasIsr 0
W (14810) intr_alloc: ....Unusable: reserved
W (14810) intr_alloc: Int 26 reserved 0 priority 5 LEVEL hasIsr 0
W (14810) intr_alloc: ....Unusable: incompatible priority
W (14810) intr_alloc: Int 27 reserved 2 priority 3 LEVEL hasIsr 0
W (14810) intr_alloc: ....Unusable: reserved
W (14810) intr_alloc: Int 28 reserved 0 priority 4 EDGE hasIsr 0
W (14810) intr_alloc: ....Unusable: incompatible priority
W (14810) intr_alloc: Int 29 reserved 0 priority 3 EDGE hasIsr 0
W (14810) intr_alloc: ....Unusable: special-purpose int
W (14810) intr_alloc: Int 30 reserved 2 priority 4 EDGE hasIsr 0
W (14810) intr_alloc: ....Unusable: reserved
W (14810) intr_alloc: Int 31 reserved 2 priority 5 LEVEL hasIsr 0
W (14810) intr_alloc: ....Unusable: reserved
W (14810) intr_alloc: get_available_int: using int -1
```

chipweinberger, 2023-05-21:
What I dont understand: **why does v5.1 use more interrupts?** 

I've tried `ESP_INTR_FLAG_LEVEL1` to `ESP_INTR_FLAG_LEVEL5`, and none are sufficient. I don't have enough interrupts anymore on v5.1.

igrr, 2023-05-22:
According to the log, you have a 4 more interrupts available:
```
W (14810) intr_alloc: Int 19 reserved 0 priority 2 LEVEL hasIsr 1
W (14810) intr_alloc: ....Unusable: incompatible priority
W (14810) intr_alloc: Int 20 reserved 0 priority 2 LEVEL hasIsr 1
W (14810) intr_alloc: ....Unusable: incompatible priority
W (14810) intr_alloc: Int 21 reserved 0 priority 2 LEVEL hasIsr 1
W (14810) intr_alloc: ....Unusable: incompatible priority
...
W (14810) intr_alloc: Int 23 reserved 0 priority 3 LEVEL hasIsr 0
W (14810) intr_alloc: ....Unusable: incompatible priority
```

You can use ESP_INTR_FLAG_LOWMED for one or more of the drivers, allowing them to use these level 2 and 3 interrupts.

chipweinberger, 2023-05-23:
Thanks for all the suggestions Ivan. I gave them all a try.

- using `ESP_INTR_FLAG_LEVEL2 | ESP_INTR_FLAG_LEVEL3` for uart & i2c, and `LEVEL1` for usb
- using `ESP_INTR_FLAG_SHARED ` for uart, and `gpio_install_isr_service`

None of those worked for whatever reason.

🎉🎉 However, what **did work**, was moving my usb init to CPU1 using `xTaskCreatePinnedToCore`, as you suggest. 🎉🎉

**Thanks for the help and the suggestion!** It feels good to update!

closed 2023-05-23 by chipweinberger