espressif/esp-idf #11464: [Feature Request][Nimble] detect when a user tries to use a os_mbuf that was already freed

chipweinberger, 2023-05-21:
### Is your feature request related to a problem?

See issue: https://github.com/espressif/esp-idf/issues/11450

If the user tries to submit a block that was already freed, _it causes an incredibly trick to debug infinite loop_. This is so hard to debug, and easy to for the user to accidentally trigger, that we ought to prevent this issue. 

**Below is some code that triggers such an issue**

**BAD CODE. DONT USE.** (does not re-call `ble_hs_mbuf_from_flat` after `ble_gattc_indicate_custom`)
```
    // loop get txom
    do{
        txom = ble_hs_mbuf_from_flat(msg, msgLen);
        if (txom == NULL){
            vTaskDelay(10 / portTICK_PERIOD_MS);
            if (millis() - tStart > 4000){
                pd_err_fail(error, TAG , "timed out (4 secs) waiting for BLE resources - ""B");
                return;
            }
        }
    } while (txom == NULL);
    
    // loop send indication
    int rc = 0;
    do {
        rc = ble_gattc_indicate_custom(gapConnHandle, attrHandle, txom);
        if (rc == BLE_HS_ENOMEM){
            vTaskDelay(10 / portTICK_PERIOD_MS);
            if (millis() - tStart > 4000){
                pd_err_fail(error, TAG , "timed out (4 secs) waiting for BLE resources - ""C");
                return;
            }
        }
    }while (rc == BLE_HS_ENOMEM);
```

### Describe the solution you'd like.

`ble_gattc_indicate_custom` should check if the os_mbuf is not currently in the free pool. 

A simple way to do this would be a flag in os_mbuf. i.e. `os_mbuf->is_in_freepool`, which gets set to `true` when freed with `os_mbuf_free` and `false` when returned by `os_msys_get_pkthdr` or similar. In `ble_gattc_indicate_custom`, we'd check that the `om->is_in_freepool` is false.

chipweinberger, 2023-05-22:
@rahult-github @IshaESP @RoshanESP

therealergo, 2024-05-10:
@chipweinberger This might be worth reporting to the Apache Nimble team [here](https://github.com/apache/mynewt-nimble) instead (although there are a lot of open unanswered bugs there).

mickeyl, 2025-04-28:
Could that also be responsible for a hang in `os_mbuf_append` like in https://github.com/apache/mynewt-nimble/issues/2027 ?

sjanc, 2025-12-09:
Hi,

There is OS_MEMPOOL_POISON option which should help in debugging "use-after-free"