espressif/esp-idf #11466: ESP-IDF V5.0.1 does not support ESP32-C6 DEVKITC-1. Any upgrade to resolve the issue?

loraadh, 2023-05-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello,
I am not able to use my ESP32-C6 DEVKITC-1. That is because ESP-IDF V5.0.1 still does not supports C6 targets. Please find attached the error message.
Is there an upgrade that support C6 targets? any workaround?
[image]

dobairoland, 2023-05-22:
See https://github.com/espressif/esp-idf/issues/10423. ESP32-C6 won't be supported by any of the v5.0.x releases. The initial support will be in the upcoming v5.1 release.

closed 2023-05-22 by dobairoland

Hedda, 2023-05-22:
> See #10423. ESP32-C6 won't be supported by any of the v5.0.x releases. The initial support will be in the upcoming v5.1

Also keep an eye on https://github.com/espressif/arduino-esp32/issues/7852 and https://github.com/espressif/arduino-esp32/issues/7713 

You can until then still use the unreleased "release/5.1" branch:

https://github.com/espressif/esp-idf/tree/release/v5.1

If have questions then post them to the esp32.com forum:

https://www.esp32.com