espressif/esp-idf #11468: ESP-IDF for platformio V1.6.2: ESP32C3 RMT transmit fails when data size is exactly SOC_RMT_MEM_WORDS_PER_CHANNEL

Straubm, 2023-05-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF for platformio V1.6.2

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

Custom board ESP32C3

### Power Supply used.

USB

### What is the expected behavior?

Data packet should be app. 57 µS.

### What is the actual behavior?

Data packet of insane length. Fails for all data sizes of SOC_RMT_MEM_WORDS_PER_CHANNEL or multiples thereof

[image]

### Steps to reproduce.

1. Step
2. Step
3. Step
...

suda-morris, 2023-05-22:
could you please provide the minimal code that can reproduce the issue? And which IDF version are you using? (by checking the `esp_idf_version.h`). @Straubm

Straubm, 2023-05-22:
Thanks for the quick reply.
The version seems to be V5.0.1
As for he code: I would have to write code to demonstrate the effect (or cut down existing code). I am pretty sure it is in the rmt component as I am simply using rmt_transmit and the bytes encoder (led_strip).
But I see you need something to work with, so bear with me, it will take a little.

suda-morris, 2023-05-23:
Hi @Straubm I can provide my test code as follows:

```c
TEST_CASE("rmt bytes encoder", "[rmt]")
{
    rmt_tx_channel_config_t tx_channel_cfg = {
        .mem_block_symbols = SOC_RMT_MEM_WORDS_PER_CHANNEL,
        .clk_src = RMT_CLK_SRC_DEFAULT,
        .resolution_hz = 1000000, // 1MHz, 1 tick = 1us
        .trans_queue_depth = 4,
        .gpio_num = 0,
    };
    printf("install tx channel\r\n");
    rmt_channel_handle_t tx_channel = NULL;
    TEST_ESP_OK(rmt_new_tx_channel(&tx_channel_cfg, &tx_channel));
    printf("install bytes encoder\r\n");
    rmt_encoder_handle_t bytes_encoder = NULL;
    rmt_bytes_encoder_config_t bytes_enc_config = {
        .bit0 = {
            .level0 = 1,
            .duration0 = 3, // 3us
            .level1 = 0,
            .duration1 = 9, // 9us
        },
        .bit1 = {
            .level0 = 1,
            .duration0 = 9, // 9us
            .level1 = 0,
            .duration1 = 3, // 3us
        },
    };
    TEST_ESP_OK(rmt_new_bytes_encoder(&bytes_enc_config, &bytes_encoder));
    printf("enable tx channel\r\n");
    TEST_ESP_OK(rmt_enable(tx_channel));

    printf("start transaction\r\n");
    rmt_transmit_config_t transmit_config = {
        .loop_count = 0, // no loop
    };
    TEST_ESP_OK(rmt_transmit(tx_channel, bytes_encoder, (uint8_t[]) {
        0x00, 0x01, 0x02, 0x03, 0x04, 0x05
    }, 6, &transmit_config));
    // adding extra delay here for visualizing
    vTaskDelay(pdMS_TO_TICKS(500));
    TEST_ESP_OK(rmt_transmit(tx_channel, bytes_encoder, (uint8_t[]) {
        0x00, 0x01, 0x02, 0x03, 0x04, 0x05, 0x06
    }, 7, &transmit_config));
    vTaskDelay(pdMS_TO_TICKS(500));
    TEST_ESP_OK(rmt_transmit(tx_channel, bytes_encoder, (uint8_t[]) {
        0x00, 0x01, 0x02, 0x03, 0x04
    }, 5, &transmit_config));
    vTaskDelay(pdMS_TO_TICKS(500));

    printf("disable tx channel\r\n");
    TEST_ESP_OK(rmt_disable(tx_channel));
    printf("remove tx channel and led strip encoder\r\n");
    TEST_ESP_OK(rmt_del_channel(tx_channel));
    TEST_ESP_OK(rmt_del_encoder(bytes_encoder));
}
```
We set the memory block size to 48 words (i.e. SOC_RMT_MEM_WORDS_PER_CHANNEL). We're using the bytes encoder to transmit 48, 56, 40 words in three transactions. However,  I didn't see wrong in the waveform.

Can you increase the `mem_block_symbols` value on your side, to see if you can still see the issue? I guess the issue you met was related to "when handling the ping-pong interrupt, the RMT encoder costs much time, but the next ping-pong interrupt comes again, and missed"

Straubm, 2023-05-23:
Thx a lot. Will do. Right now I am AFK, so tonight.

Straubm, 2023-05-24:
Hi, so I tested with your code with modified resolution (0.1µS to match my use case) and mem_block_symbols = SOC_RMT_MEM_WORDS_PER_CHANNEL as well as data size SOC_RMT_MEM_WORDS_PER_CHANNEL (which fails in my application).
It worked.
So I am at a loss to see what's happening. Sure, my code has many more components running.
Seems I have to check/debug my code again or something.
I am afraid the thread will stay "In Progress" a little.
Meanwhile, thanks for your cooperation.

Straubm, 2023-05-24:
@suda-morris I think I have found the issue.

In my encoder I had some ESP_DRAM_LOGI entries for diagnostic messages as you cannot watch variables when you break in IRAM.
Removing them solved the problem.
Sorry for the possibly unnecessary noise.

Case closed as far as I am concerned.

suda-morris, 2023-05-26:
yes, wifi stuff has higher priority, can affect the RMT encoding session, because the RMT does the encoding in the ISR context. If the ISR is deferred somehow (mean it's not a hard realtime system), then the RMT hardware will continue transmitting the previous encoding artifacts.

What we can do, it to make the interrupt interval as bigger as possible and make the RMT ISR as shorter as possible. You can increase the value of `mem_block_symbols` to for example `SOC_RMT_MEM_WORDS_PER_CHANNEL*4`.

Straubm, 2023-05-26:
Ok, now I'm happy 🙂 . Increasing mem_block_symbols helped. Main reason for the problems was a blunder in my code, though.
So NOW it's done from my side.
Thanks for your help.

closed 2023-06-03 by espressif-bot