espressif/esp-idf #11471: ESP-IDF 5.1

loraadh, 2023-05-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello,
In previous versions I used to do the installations via the online installer: https://dl.espressif.com/dl/esp-idf/?idf=4.4

Version ESP-IDF 5.1 is still not supported by that installer. Is there a user guide on how to install ESP-IDF 5.1, flash/upload, and command the C6 (preferably using terra term)?
I use the ESP32-C6-DevKitC-1
thanks
BR

chipweinberger, 2023-05-22:
roughly

```
git clone, 

then run ./install.sh

. $HOME/esp/esp-idf/export.sh
```

see: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/

dobairoland, 2023-05-22:
As I pointed out in the duplicate issue (https://github.com/espressif/esp-idf/issues/11466), v5.1 is not yet released. Therefore, you cannot find a Windows Installer built for it. However, you can use the Windows Installer targeting an external ESP-IDF copy with checked out `release/v5.1` release branch as it was suggested by the previous post.

Please use the forums at https://esp32.com/ for asking questions instead of opening an issue for each of them. Thanks!

Alvin1Zhang, 2023-05-23:
Thanks for reporting, feel free to reopen.

closed 2023-05-23 by Alvin1Zhang

loraadh, 2023-05-23:
> roughly
> 
> ```
> git clone https://github.com/espressif/esp-idf.git
> 
> git checkout release/v5.1
> 
> then run ./install.sh
> 
> . $HOME/esp/esp-idf/export.sh
> ```
> 
> see: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/

Hello,
thanks @chipweinberger 
I did the git clone https://github.com/espressif/esp-idf.git
then unzip it under c/:
then double clicked the install.sh but it is not running; a black window opens then closes rapidly.
Knowing that I don't have experience in FW.
BR
Achraf

loraadh, 2023-05-23:
> As I pointed out in the duplicate issue (#11466), v5.1 is not yet released. Therefore, you cannot find a Windows Installer built for it. However, you can use the Windows Installer targeting an external ESP-IDF copy with checked out `release/v5.1` release branch as it was suggested by the previous post. For the most convenient solution you will have to wait for the release of v5.1.
> 
> Please use the forums at https://esp32.com/ for asking questions instead of opening an issue for each of them. Thanks!
Hello @dobairoland 
thanks 
any idea on the release date of release v5.1 ? coming days, weeks, months?
BR

chipweinberger, 2023-05-23:
@loraadh , just wait for the official release. too much for you to learn =)