espressif/esp-idf #11472: fail-over mechanism when use both wifi station and eth

gl-chenxiaosheng, 2023-05-22:
### Is your feature request related to a problem?

I see that the priority of wan can be set when initializing the network. Have you also completed the automatic switching feature for wan fail over ?

Xiehanxin, 2023-05-29:
hi, I want to know if you want to do this: When there are multiple interfaces (eth and wifi) working at the same time, if eth is suddenly disconnected,  you can directly switch to the sta interface to continue forwarding, or you can describe your application scenario in more detail

gl-chenxiaosheng, 2023-05-29:
yes, it is like as you decription.
> When there are multiple interfaces (eth and wifi) working at the same time, if eth is suddenly disconnected, you can directly switch to the sta interface to continue forwarding.

By the way, if I want to change the priority of the route, is there an interface that can be implemented.
Thanks.

david-cermak, 2024-03-11:
Hi @gl-chenxiaosheng 

There's no way to update the `route_prio` runtime, but it should automatically switch to the other interface if the one (with higher prio) was set down (for example for Ethernet link-down event).
It's also possible to override this setting and set one interface as default (regardless of the `route_prio`) -- this is useful when one interface looses a specific connectivity (e.g. the link is up, connection to local endpoints still working, but the remote endpoint is not available).
We have an example that demonstrate this kind of switching between WiFi, Ethernet and PPP interfaces:

https://github.com/espressif/esp-protocols/tree/master/examples/esp_netif/multiple_netifs

david-cermak, 2024-04-12:
Closing as this switch over mechanism between netifs has been implemented and an example give. Feel free to reopen if needed.

closed 2024-04-12 by david-cermak