espressif/esp-idf #11473: esp_log_level_set() seems to assume passed string names are fixed in memory

moefear85, 2023-05-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

latest mater

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32 rev1 module

### Power Supply used.

External 3.3V

### What is the expected behavior?

When I call esp_log_level_set(...), I expect it to immediately set the log level or save the name.

### What is the actual behavior?

Instead it sometimes takes effect, in others it doesn't.

### Steps to reproduce.

I don't currently have a minimal example, but I have a parser loop that takes statements from the serial line, and if a certain command tells it to set the log level for a specific tag to some value, then `esp_log_level_set()` is called on that tag name and the given level is set. However, I'm coding in C++, and the passed name and level only exist for a short while on the stack. I'm using `string_name.c_str()` when passing the name to `esp_log_level_set()`. However I realized it doesn't take effect when called from within that particular parser loop except for a single specific tag, which is the tag that the parser loop itself uses to log its behavior to serial, leading me to assume `esp_log_level_set()` isn't immediately making use of the char* passed to it, rather is deferring some action until later when said pointer is no longer valid.

It isn't crashing the system, but it's strange. I've placed many extensive logging statements within the parser loop before and after the call to `esp_log_level_set` to ensure that I am actually passing the correct tag name and level that I think I am, so I'm absolutely sure I'm not making any mistake. I'm passing valid tag pointers to the function, but not seeing any effect until a restart, when the main() function applies the same level settings to all the tags at start.

moefear85, 2023-05-22:
When `esp_log_level_set()` actually uses the passed name to set the level, does it actually compare the string literals themselves, or does it just look at the address of the pointer?

igrr, 2023-05-22:
This sounds very similar to https://github.com/espressif/esp-idf/issues/8567, which has been reported a while ago but we haven't found way of reproducing, IIRC.

(There has been an issue of comparing tag pointers rather than values, however it was fixed a long time ago in 750c8c00c89f3911692c37494c2eb8401cd5308e)

moefear85, 2023-05-22:
@igrr 

I modified the issue. Please read the updates.

moefear85, 2023-05-23:
fine, but please mark that as active & reproducible, as it is corroborated by more than one user

closed 2023-05-23 by igrr

igrr, 2023-05-23:
Yeah, that other one is "Selected for development", just the priority is relatively low because it relates to a non-standard use of ESP_LOG (dynamic tag strings).