espressif/esp-idf #11474: MQTT implementation error on mqtt_app_start()

fnandomaia, 2023-05-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

esp-idf-v5.0.1

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

WT32-SC01 Plus 8Bit Parallel 3.5

### Power Supply used.

USB

### What is the expected behavior?

I'm trying to implement the MQTT protocol inside an ESP32-TUX instance.

1. I enabled MQTT configuration inside idf.py menuconfig.

2. main.hpp I put the include of the local file for the mqtt library downloaded from Github (https://github.com/espressif/esp-mqtt) because that way it didn't give error of include not found.

3. main.hpp include mqtt_client.h file. I initialized the functions: mqtt_event_handler() and mqtt_app_start().
[image]

4. main.cpp I placed the respective functions from the previous step and initialized mqtt_app_start().
[image]

### What is the actual behavior?

It happens that when compiling it gives an error inside the mqtt_client.h library, as shown in the image below.
[image]

I wonder if anyone has an idea of where am I getting wrong with my code?
Looks like some C# C++ error. 

### Steps to reproduce.

As describle above.

euripedesrocha, 2023-05-26:
Hi @fnandomaia, we introduced a new feature in the mqtt client that breaks the compilation for C++, with the usage of this convenience macro. This should be fixed soon.

euripedesrocha, 2023-05-29:
Hi @fnandomaia taking a look again on your issue, while is there a problem on our side, I would also like to point that you may have issues on your project configuration. From the description of your usage of MQTT library, I think you are missing the mqtt component in your required dependencies. There is no need to clone the MQTT client separately as you are doing.

tshcherban, 2023-08-06:
Hi
any updates here?
Im using IDF v 5.1 and a simple MQTT example, compiling with C++. Compiles ok but fails at runtime with:
```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400556c0  PS      : 0x00060f30  A0      : 0x8005562d  A1      : 0x3fca7bb0
0x400556c0: strlen in ROM
```
seems it has something to do with a C-style strings which are empty by default? 
The same example works OK under C language.

euripedesrocha, 2023-08-07:
@tshcherban could you please open a new issue, providing the information to verify(Mqtt client configuration, sdkconfig, mqtt version) ? The build problem I mentioned is already solved and not present in 5.1.

closed 2023-11-07 by euripedesrocha