espressif/esp-idf #11475: ESP32-C3 Incorrect Wakeup Cause with GPIO, Resulting in RTC Reset

Pavelovec, 2023-05-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.2

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-C3 rev v0.3 on custom board

### Power Supply used.

External 3.3V

### What is the expected behavior?

The ESP32-C3 board should reliably wake up from deep sleep when a connected button (with external pull-up) on GPIO5 is pressed. This should be consistently signaled with a wakeup cause of rst:0x5 (DSLEEP). The RTC time and memory should be maintained across deep sleep cycles.

### What is the actual behavior?

The ESP32-C3 board does not reliably wake up from deep sleep when the connected button on GPIO5 is pressed. Occasionally, instead of waking up from deep sleep, the system performs a full power-on reset (signaled by rst:0x1 (POWERON)) which resets the RTC time and memory.

### Steps to reproduce.

1.  Connect a button with an external pull-up resistor to GPIO5.
2.  Flash the provided code onto an ESP32-C3 board.

```
#define WAKETEST_GPIO_NUM 5

void app_main(void)
{

    printf("*  Reset reason:  ");
    switch (rtc_get_reset_reason(0))
    {
    case 1:
        printf("POWERON_RESET");
        break; /**<1,  Vbat power on reset*/
    case 3:
        printf("SW_RESET");
        break; /**<3,  Software reset digital core*/
    case 5:
        printf("DEEPSLEEP_RESET");
        break; /**<5,  Deep Sleep reset digital core*/
    case 7:
        printf("TG0WDT_SYS_RESET");
        break; /**<7,  Timer Group0 Watch dog reset digital core*/
    case 8:
        printf("TG1WDT_SYS_RESET");
        break; /**<8,  Timer Group1 Watch dog reset digital core*/
    case 9:
        printf("RTCWDT_SYS_RESET");
        break; /**<9,  RTC Watch dog Reset digital core*/
    case 10:
        printf("INTRUSION_RESET");
        break; /**<10, Instrusion tested to reset CPU*/
    case 11:
        printf("TGWDT_CPU_RESET");
        break; /**<11, Time Group reset CPU*/
    case 12:
        printf("SW_CPU_RESET");
        break; /**<12, Software reset CPU*/
    case 13:
        printf("RTCWDT_CPU_RESET");
        break; /**<13, RTC Watch dog Reset CPU*/
    case 15:
        printf("RTCWDT_BROWN_OUT_RESET");
        break; /**<15, Reset when the vdd voltage is not stable*/
    case 16:
        printf("RTCWDT_RTC_RESET");
        break; /**<16, RTC Watch dog reset digital core and rtc module*/
    default:
        printf("NO_MEAN");
    }
    printf("\n");

    printf("*  WakeUp reason: ");
    switch (esp_sleep_get_wakeup_cause())
    {
    case ESP_SLEEP_WAKEUP_EXT0:
        printf("Wakeup caused by external signal using RTC_IO");
        break;
    case ESP_SLEEP_WAKEUP_EXT1:
        printf("Wakeup caused by external signal using RTC_CNTL");
        break;
    case ESP_SLEEP_WAKEUP_TIMER:
        printf("Wakeup caused by timer");
        break;
    case ESP_SLEEP_WAKEUP_TOUCHPAD:
        printf("Wakeup caused by touchpad");
        break;
    case ESP_SLEEP_WAKEUP_ULP:
        printf("Wakeup caused by ULP program");
        break;
    case ESP_SLEEP_WAKEUP_GPIO:
        printf("Wakeup caused by GPIO");
        break;
    default:
        printf("Wakeup was not caused by deep sleep: %d\n", esp_sleep_get_wakeup_cause());
        break; // printf(""); //
    }
    printf("\n");
    
    vTaskDelay(2000 / portTICK_PERIOD_MS);

    printf("Entering deep sleep...\n\n");

    fflush(stdout);

    const gpio_config_t config = {
        .pin_bit_mask = BIT(WAKETEST_GPIO_NUM),
        .mode = GPIO_MODE_INPUT,
    };
    ESP_ERROR_CHECK(gpio_config(&config));
    ESP_ERROR_CHECK(esp_deep_sleep_enable_gpio_wakeup(BIT(WAKETEST_GPIO_NUM), ESP_GPIO_WAKEUP_GPIO_LOW));
    esp_deep_sleep_start();
}

```

3.  Press the connected button to attempt to wake the board from deep sleep.
   
   
    
   

### Debug Logs.

```plain
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xd (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0x3ac
load:0x403cc710,len:0x6c0
load:0x403ce710,len:0x24d0
entry 0x403cc710
*  Reset reason:  POWERON_RESET
*  WakeUp reason: Wakeup was not caused by deep sleep: 0

Entering deep sleep...

ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xd (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0x3ac
load:0x403cc710,len:0x6c0
load:0x403ce710,len:0x24d0
entry 0x403cc710
*  Reset reason:  POWERON_RESET
*  WakeUp reason: Wakeup was not caused by deep sleep: 0

Entering deep sleep...
```

### More Information.

The issue was observed while using the Arduino core v2.0.9 and ESP-IDF v5.0.2. The same behavior occurred with a button connected to GPIO3 with external pull-down. This issue seems to occur when the button is clicked **after** entering deep sleep; if the button is held **before** entering deep sleep, the wakeup works as expected.

In contrast, waking up the ESP32-C3 from deep sleep using the timer works perfectly, indicating that the issue is specific to GPIO wakeups.

I have tried nearly every possible workaround that I could think of and searched extensively for solutions, but none has resolved the issue. The goal is for the system to not reset RTC time or memory on wakeup.

This issue has proven to be a significant stumbling block, and I would greatly appreciate any assistance in resolving it. This behavior limits the usability of the ESP32-C3 for my project, as maintaining RTC across deep sleep cycles is a key requirement.

Pavelovec, 2023-05-24:
I have recently tested this issue with ESP-IDF v5.2 and it appears to function perfectly, without experiencing the incorrect wakeup cause from GPIO or RTC reset. However, for my project requirements, I need this to work with ESP-IDF v4.4 (because of Arduino framework). It would be really helpful if someone could guide me on the necessary changes to make this work with v4.4. Any advice or insights into this would be greatly appreciated. Thank you!

10086loutianhao, 2023-06-19:
@Pavelovec Hello, I am unable to reproduce your issue. I am using the release/v4.4 branch with the commit: 8b94183c9cb47ede8f02df5598d8b9d68c754860.

AxelLin, 2023-06-19:
@Pavelovec
Are you using v4.4 release? If so, can you try if v4.4.5 release work?
If you are using v4.4 branch, what's your 'git describe' output?

Pavelovec, 2023-06-19:
> @Pavelovec Are you using v4.4 release? If so, can you try if v4.4.5 release work? If you are using v4.4 branch, what's your 'git describe' output?

Yes, I am using the v4.4 release. I've just tried the v4.4.5 release as you suggested, but unfortunately, the problem still exists.

bricomp, 2023-06-21:
I can conform that the same thing happens for me. I have tried waking up on pins #2 and #4. I am set to a LOW level wake-up.
Variables set in the RTC memory are reset and the wake up cause is always shown as 0
I am using the Arduino IDE by the way.

10086loutianhao, 2023-06-26:
@bricomp hello, can you please tell me the commit you working on and the menuconfig you modified or other details, I will take action to solve this problem.

10086loutianhao, 2023-06-27:
@Pavelovec  @bricomp Can you provide samples to help me reproduce the problem? Thank you.

bricomp, 2023-06-29:
Hi @10086loutianhao I have found the source of the problem and I am able to work arround it.
The problem is that if the SerialMonotor is open the ESP32C3 does NOT wake up properly from deep sleep. I am using the VISUAL MICRO (ARDUINO Based) IDE.
Now that I know where the problem lies I can work around it.
By the way I presented a sketch that showed the problem at the start of my submission!!!

Pavelovec, 2023-07-26:
@10086loutianhao Apologies for the delay in responding. I hadn't noticed your question earlier. The example code is indeed in the "Steps to reproduce" section. I haven't made any modifications in menuconfig; the issue seems to be specific to the ESP-IDF version, as it works perfectly with ESP-IDF v5.2.

Pavelovec, 2023-08-08:
Based on this commit: https://github.com/espressif/esp-idf/commit/1b04acf68f06995208c84f3cc6fbc475279ece10, I have implemented the changes in esp-idf v4.4. After thorough testing, it appears that these changes work perfectly and have resolved the initial issue.  My hope is that they won't introduce any new complications. I believe this issue can now be closed.

markwaters306, 2023-11-16:
I am still experiencing this issue with the esp32c3.

I've tried multiple idf versions and the issue remains. After deep sleep esp_sleep_get_wakeup_cause() will always return zero and rtc memory is lost.

I've tested v5.1, v5.1.2, 5.2 and 5.3.

asergunov, 2024-05-11:
I have the same issue but I think it's because I have interrupts attached to pin changes. So first wake cause is correct but then watchdog triggers reboot which cleans rtc memory.

10086loutianhao, 2024-05-11:
If you encounter any problems you wish to solve, please create a new issue and provide the necessary information for reproduction. Thank you