espressif/esp-idf #11476: [v5.1] @click.option('-C', '--project-dir', default=os.getcwd(), type=click.Path()) No such file or directory

chipweinberger, 2023-05-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

release/v5.1

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I am trying to run get_idf

### What is the actual behavior?

It completes, but then crashes.

```
Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

Traceback (most recent call last):
  File "/Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/tools/idf.py", line 767, in <module>
    main()
  File "/Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/tools/idf.py", line 695, in main
    cli = init_cli(verbose_output=checks_output)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/tools/idf.py", line 620, in init_cli
    @click.option('-C', '--project-dir', default=os.getcwd(), type=click.Path())
                                                 ^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory
charlesweinberger@Chips-Macbook-Pro-2 jamcorder_app % idf.py build    
Traceback (most recent call last):
  File "/Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/tools/idf.py", line 767, in <module>
    main()
  File "/Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/tools/idf.py", line 695, in main
    cli = init_cli(verbose_output=checks_output)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/tools/idf.py", line 620, in init_cli
    @click.option('-C', '--project-dir', default=os.getcwd(), type=click.Path())
```

### Steps to reproduce.

Not entirely sure. It was not doing this before. I've tried 

### Build or installation Logs.

```plain
charlesweinberger@Chips-Macbook-Pro-2 jamcorder_app % getidf
Detecting the Python interpreter
Checking "python3" ...
Python 3.11.3
"python3" has been detected
Checking Python compatibility
Checking other ESP-IDF version.
WARNING: tool cmake found in path, but returned non-zero exit code (1) with error message:
Current working directory cannot be established.

WARNING: tool cmake version 3.24.0 is installed, but returned non-zero exit code (1) with error message:
Current working directory cannot be established.

Adding ESP-IDF tools to PATH...
WARNING: tool cmake found in path, but returned non-zero exit code (1) with error message:
Current working directory cannot be established.

WARNING: tool cmake version 3.24.0 is installed, but returned non-zero exit code (1) with error message:
Current working directory cannot be established.

Checking if Python packages are up to date...
Constraint file: /Volumes/User/MBP-Google-Drive/jamcorder/firmware/.espressif/espidf.constraints.v5.1.txt
Requirement files:
 - /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/tools/requirements/requirements.core.txt
Python being checked: /Volumes/User/MBP-Google-Drive/jamcorder/firmware/.espressif/python_env/idf5.1_py3.11_env/bin/python
Python requirements are satisfied.
Updated PATH variable:
  /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/espcoredump:/Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/partition_table:/Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/app_update:/Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/espcoredump:/Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/partition_table:/Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/app_update:/Volumes/User/MBP-Google-Drive/jamcorder/firmware/.espressif/tools/xtensa-esp-elf-gdb/12.1_20221002/xtensa-esp-elf-gdb/bin:/Volumes/User/MBP-Google-Drive/jamcorder/firmware/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin:/Volumes/User/MBP-Google-Drive/jamcorder/firmware/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin:/Volumes/User/MBP-Google-Drive/jamcorder/firmware/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:/Volumes/User/MBP-Google-Drive/jamcorder/firmware/.espressif/tools/cmake/3.24.0/CMake.app/Contents/bin:/Volumes/User/MBP-Google-Drive/jamcorder/firmware/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230419/openocd-esp32/bin:/Volumes/User/MBP-Google-Drive/jamcorder/firmware/.espressif/python_env/idf5.1_py3.11_env/bin:/Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/tools:/usr/local/bin:/System/Cryptexes/App/usr/bin:/usr/bin:/bin:/usr/sbin:/sbin:/Library/Apple/usr/bin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/local/bin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/bin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/appleinternal/bin:/Volumes/User/MBP-Google-Drive/jamcorder/app/flutter/bin:/Volumes/User/MBP-Google-Drive/jamcorder/app/flutter/.pub-cache/bin

Detected installed tools that are not currently used by active ESP-IDF version.
For removing old versions of riscv32-esp-elf, cmake, xtensa-esp32-elf, esp32s2ulp-elf, openocd-esp32, esp32ulp-elf, esp-rom-elfs, xtensa-esp-elf-gdb, xtensa-esp32s3-elf, xtensa-esp32s2-elf, riscv32-esp-elf-gdb use command 'python /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/tools/idf_tools.py uninstall'
To free up even more space, remove installation packages of those tools. Use option 'python3 /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/tools/idf_tools.py uninstall --remove-archives'.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

Traceback (most recent call last):
  File "/Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/tools/idf.py", line 767, in <module>
    main()
  File "/Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/tools/idf.py", line 695, in main
    cli = init_cli(verbose_output=checks_output)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/tools/idf.py", line 620, in init_cli
    @click.option('-C', '--project-dir', default=os.getcwd(), type=click.Path())
                                                 ^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory
charlesweinberger@Chips-Macbook-Pro-2 jamcorder_app % idf.py build    
Traceback (most recent call last):
  File "/Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/tools/idf.py", line 767, in <module>
    main()
  File "/Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/tools/idf.py", line 695, in main
    cli = init_cli(verbose_output=checks_output)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/tools/idf.py", line 620, in init_cli
    @click.option('-C', '--project-dir', default=os.getcwd(), type=click.Path())
```

### More Information.

Every command is hitting this issue right now. I've cleaned /build manually, but still same issue.

chipweinberger, 2023-05-23:
Solution: Fixed by closing Terminal and reopening it.

I'll leave this issue open for other to find. Hopefully we can determine root cause.

dobairoland, 2023-05-23:
By `get_idf` you mean the alias suggested [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/linux-macos-setup.html#step-4-set-up-the-environment-variables)?

If you get this the next time I would suggest to run `. $IDF_PATH/export.sh` directly in the same terminal to see if that helps. That would confirm that this is a shell environment issue (or not).

If you get this again then we can re-open it. For now, I don't think others would be affected. We haven't changed anything related to this for quite long time. So I'm closing this.

igrr, 2023-05-23:
To reproduce:
```
$ mkdir /tmp/test
$ cd /tmp/test
# in another terminal:
$ rm -rf /tmp/test
# back to the first terminal
$ . /path/to/esp-idf/export.sh
```
not sure we need to fix this, though: a lot of other programs also misbehave the same way when you yank the CWD...

chipweinberger, 2023-05-23:
> By get_idf you mean the alias suggested here?

yes

> If you get this again then we can re-open it.

yes. I think that is fair. I don't think I would fix it either. But I always like to file an issue for google-ability!

dobairoland, 2023-05-23:
Yes, you are right @igrr. I can reproduce that way.

Than running anything produces the same issue:
```
cmake --version
Current working directory cannot be established.
```

chipweinberger, 2023-05-23:
cmake *does* log a nice error message though ;)

dobairoland, 2023-05-23:
It is interesting that `pwd` still works: 

```
.-(/tmp/test)---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
`--> python -c 'import os; print(os.getcwd())'
Traceback (most recent call last):
  File "<string>", line 1, in <module>
FileNotFoundError: [Errno 2] No such file or directory
.-(/tmp/test)---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
`--> pwd
/tmp/test
```

So I guess any fix could be done only at `idf.py`.

We will take a closer look at this.

closed 2023-06-10 by espressif-bot