espressif/esp-idf #11477: [v5.1][esp32-s3] logs show weird characters. did not happen on v4.4.

chipweinberger, 2023-05-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

release/v5.1

### Operating System used.

macOS

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3 Dev Kit C

### Power Supply used.

USB

### What is the expected behavior?

Logs should print normal characters.

### What is the actual behavior?

Here are the symptoms:

1. Sometimes there are bad logs: i.e. Å

```
I (9689) pd usbh: ÄinitÅ got mutex
I (9699) pd usbh: ÄinitÅ xSemaphoreCreateRecursiveMutex() xClaimedInterfacesMutex
I (9699) pd usbh: ÄinitÅ usb_host_install()
```

or 

```
I (5719) pd api: register Ä0Å /api/self-test/capabilities
I (5719) pd api: register Ä1Å /api/self-test/list
I (5729) pd api: register Ä2Å /api/self-test/run/single-test
I (5729) pd api: register Ä3Å /api/self-test/run/factory-test-suite
I (5729) pd api: register Ä4Å /webapp
I (5729) pd api: register Ä5Å /webapp/*
I (5729) pd api: register Ä6Å /sd/?*

```
2. sometimes there are even crazier logs. Nearly every character is bad. looking like wingdings.

3. restarting the device & the terminal usually fixes it

### Steps to reproduce.

Not sure what causes it. Seems random. Happens occasionally.

igrr, 2023-05-24:
Could you please check if running `idf.py --no-hints monitor` has the same behavior or not?

chipweinberger, 2023-05-24:
Sure I can try that. Why would `--no-hints` do anything? Seems strange!

igrr, 2023-05-24:
This just looks similar to the issue with control characters affecting terminal behavior when hints are enabled: https://github.com/espressif/esp-idf/issues/11351. The reason why disabling hints can change the result is that when hints are disabled, the mechanism used to send serial output to the terminal is different.

chipweinberger, 2023-05-30:
I have not been seeing this issue again lately! 

I saw it like 5+ times in 1 day, but not since.