espressif/esp-idf #11479: std::vector destructor causes random `LoadProhibited` or `StoreProhibited` crashes in tlsf.c

wuyuanyi135, 2023-05-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2-dev-703-gefe919275e

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32-S3-WROOM-1-N16R2

### Power Supply used.

USB

### What is the expected behavior?

Allocating/Deallocating vector should not cause crash.

### What is the actual behavior?

I have experienced several crashes when the system was running for ~24hrs. The coredump pointed to different final frames but the causing frame are the same: the destructor of a `std::vector`, and it leads to `LoadProhibited` or `StoreProhibited` exceptions in tlsf. I could confirm no stack overflow occurs. There was also no concurrent access (the vector is a local variable).

This vector was constructed/destructed periodically every 0.5 seconds by a `std::thread`. I know this is a bad design but I don't expect it to fail, especially when releasing the resource. 

See below for the stack trace and coredump/elf files. This is the latest `StoreProhibited` crash. I had another a few `LoadProhibited` but the coredumps were  not saved. Let me know if I need keep it running for another chance of crash.

### Steps to reproduce.

See below for the coredump/elf

### Debug Logs.

```plain
Here attaches the relevant crashing back trace

#0  insert_free_block (sl=15, fl=24, block=0x3fcae2d4, control=0x3fcaa7d4) at C:/Espressif/frameworks/esp-idf-master/components/heap/tlsf/tlsf.c:362
#1  block_insert (block=0x3fcae2d4, control=0x3fcaa7d4) at C:/Espressif/frameworks/esp-idf-master/components/heap/tlsf/tlsf.c:388
#2  tlsf_free (tlsf=0x3fcaa7d4, ptr=<optimized out>) at C:/Espressif/frameworks/esp-idf-master/components/heap/tlsf/tlsf.c:1123
#3  0x40384687 in multi_heap_free_impl (p=0x3fcae2dc, heap=0x3fcaa7c0) at C:/Espressif/frameworks/esp-idf-master/components/heap/multi_heap.c:231
#4  multi_heap_free_impl (heap=0x3fcaa7c0, p=0x3fcae2dc) at C:/Espressif/frameworks/esp-idf-master/components/heap/multi_heap.c:220
#5  0x40375f1c in heap_caps_free (ptr=0x3fcae2dc) at C:/Espressif/frameworks/esp-idf-master/components/heap/heap_caps.c:388
#6  0x40386074 in free (ptr=0x3fcae2dc) at C:/Espressif/frameworks/esp-idf-master/components/newlib/heap.c:39
#7  0x420cb3a0 in operator delete (ptr=0x3fcae2dc) at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32s3-elf/src/gcc/libstdc++-v3/libsupc++/del_op.cc:49
#8  0x420cb3c0 in operator delete (ptr=0x3fcae2dc) at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32s3-elf/src/gcc/libstdc++-v3/libsupc++/del_ops.cc:33
#9  0x4201ad3a in std::__new_allocator<float>::deallocate (this=<optimized out>, __n=<optimized out>, __p=<optimized out>)
--Type <RET> for more, q to quit, c to continue without paging--c
    at c:\espressif\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\xtensa-esp32s3-elf\include\c++\12.2.0\bits/new_allocator.h:158
#10 std::allocator<float>::deallocate (__n=<optimized out>, __p=<optimized out>, this=0x3fcdffcc) at c:\espressif\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\xtensa-esp32s3-elf\include\c++\12.2.0\bits/allocator.h:200
#11 std::allocator_traits<std::allocator<float> >::deallocate (__n=<optimized out>, __p=<optimized out>, __a=...) at c:\espressif\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\xtensa-esp32s3-elf\include\c++\12.2.0\bits/alloc_traits.h:496
#12 std::_Vector_base<float, std::allocator<float> >::_M_deallocate (this=0x3fcdffcc, __n=<optimized out>, __p=<optimized out>) at c:\espressif\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\xtensa-esp32s3-elf\include\c++\12.2.0\bits/stl_vector.h:387
#13 std::_Vector_base<float, std::allocator<float> >::~_Vector_base (this=0x3fcdffcc, __in_chrg=<optimized out>) at c:\espressif\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\xtensa-esp32s3-elf\include\c++\12.2.0\bits/stl_vector.h:366
#14 0x42021209 in std::vector<float, std::allocator<float> >::~vector (this=0x3fcdffcc, __in_chrg=<optimized out>) at c:\espressif\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\xtensa-esp32s3-elf\include\c++\12.2.0\bits/stl_vector.h:733
```
```

### More Information.

elf + coredump file: 
[unilab_stirrer_firmware.zip](https://github.com/espressif/esp-idf/files/11540167/unilab_stirrer_firmware.zip)

wuyuanyi135, 2023-05-23:
Update: after disabling the repeated heap allocation/deallocation in the abovementioned place, another crash occured by another place in my code, result in `StoreProhibited` crash.
```
#0  remove_free_block (sl=13, fl=0, block=0x3fcae430, control=0x3fcaa7d4) at C:/Espressif/frameworks/esp-idf-master/components/heap/tlsf/tlsf.c:332
#1  block_locate_free (size=<optimized out>, control=<optimized out>) at C:/Espressif/frameworks/esp-idf-master/components/heap/tlsf/tlsf.c:567
#2  tlsf_malloc (tlsf=0x3fcaa7d4, size=<optimized out>) at C:/Espressif/frameworks/esp-idf-master/components/heap/tlsf/tlsf.c:1004
#3  0x403846dc in multi_heap_malloc_impl (heap=0x3fcaa7c0, size=41) at C:/Espressif/frameworks/esp-idf-master/components/heap/multi_heap.c:207
#4  0x40375d85 in heap_caps_malloc_base (size=41, caps=6144) at C:/Espressif/frameworks/esp-idf-master/components/heap/heap_caps.c:176
#5  0x40375ddb in heap_caps_malloc_default (size=41) at C:/Espressif/frameworks/esp-idf-master/components/heap/heap_caps.c:233
#6  0x40386054 in malloc (size=41) at C:/Espressif/frameworks/esp-idf-master/components/newlib/heap.c:24
#7  0x420cc13c in operator new (sz=41) at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32s3-elf/src/gcc/libstdc++-v3/libsupc++/new_op.cc:50
#8  0x4201a88b in std::__new_allocator<char>::allocate (this=<optimized out>, __n=41)
    at c:\espressif\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\xtensa-esp32s3-elf\include\c++\12.2.0\bits/new_allocator.h:137
--Type <RET> for more, q to quit, c to continue without paging--c
#9  std::allocator<char>::allocate (__n=41, this=<optimized out>) at c:\espressif\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\xtensa-esp32s3-elf\include\c++\12.2.0\bits/allocator.h:188  
#10 std::allocator_traits<std::allocator<char> >::allocate (__a=..., __n=41) at c:\espressif\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\xtensa-esp32s3-elf\include\c++\12.2.0\bits/alloc_traits.h:464
#11 std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_create (this=0x3fcadf30, __old_capacity=0, __capacity=<synthetic pointer>: 40) at c:\espressif\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\xtensa-esp32s3-elf\include\c++\12.2.0\bits/basic_string.tcc:155
#12 std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_construct<char const*> (__end=0x3c116166 "", __beg=0x3c11613e "onewire_bus_write_bytes(ow_hdl, &cmd, 1)", this=0x3fcadf30) at c:\espressif\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\xtensa-esp32s3-elf\include\c++\12.2.0\bits/basic_string.tcc:225
#13 std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::basic_string<>(const char *, const std::allocator<char> &) (this=0x3fcadf30, __s=0x3c11613e "onewire_bus_write_bytes(ow_hdl, &cmd, 1)", __a=...) at c:\espressif\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\xtensa-esp32s3-elf\include\c++\12.2.0\bits/basic_string.h:642
#14 0x4201ab9a in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::basic_string<std::allocator<char> > (__a=..., __s=0x3c11613e "onewire_bus_write_bytes(ow_hdl, &cmd, 1)", this=0x3fcadf30) at c:\espressif\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\xtensa-esp32s3-elf\include\c++\12.2.0\bits/basic_string.h:643
```

elf+coredump attached 
[coredump.zip](https://github.com/espressif/esp-idf/files/11542046/coredump.zip)

chipweinberger, 2023-05-23:
This is usually caused by memory corruption, in your own code. 

 **My usual approach:** try to get the issue to happen as often as possible, and then comment out code until the issue goes away.

wuyuanyi135, 2023-05-23:
@chipweinberger Thanks for your suggestion! In my opinion, this is most probably a bug in the libc++ implementation in esp-idf as I have reported several issue related to the std cpp library before. I don't have time to dig into this problem right now and it seems v5.0.2 does not exihibit these crashes. I will stick to the release version right now.

MartyMcFlyInTheSky, 2023-05-24:
I remember that I had the same bug once, but as far as I can remember it didn't have something to do with vector in the end. If the problem is so easily reproducible, could you write a simple app containing only the relevant code? That would be an std::thread reallocating a vector every 0.5 seconds. 0.5 seconds is like eons in computer science so the repeatedness really shouldn't be the problem. The only thing I can think about right now is an uncaught ABA exception in the malloc implementation, but that would be pretty wicked.

0xjakob, 2023-05-24:
Hi @wuyuanyi135, have you tried any of the points listed in the [Finding Heap Corruption](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/heap_debug.html#finding-heap-corruption) section of the documentation? In particular, the [CONFIG_HEAP_CORRUPTION_DETECTION](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-heap-corruption-detection) might be of interest here.

wuyuanyi135, 2023-05-24:
@0xjakob IIRC, I had stack overflow checked but I don't remember if the heap corruption detection was also enabled. I will try to enable both to the maximum level and test it.

KaeLL, 2023-05-24:
> 5.0.2 does not suffer from this issue

Did you try bisecting it?

wuyuanyi135, 2023-05-24:
@0xjakob I may not be able to run the test in short time. Meanwhile did you notice anything suspecious in the coredump? i.e., anything that has changed since 5.0.2 that may be related to the issue (in libc++ or memory allocator)?

wuyuanyi135, 2023-05-25:
@KaeLL No, unfortunately. The time to crash is pretty random and therefore bisecting might need a lot of time. When I have a spare unit I could try this approach.

0xjakob, 2023-05-25:
> Meanwhile did you notice anything suspecious in the coredump? 

The only thing I noticed is that local variable `current` is null.
The coredump only writes out the stack and the task control block, basically.

The libstdc++ is the upstream version from GCC, so far we hadn't had any memory corruption bugs located in there. They would have probably been found by users on other systems like Linux already. The code is fairly generic. Besides, the heap corruption can have many reasons, e.g. a buffer overflow. In this case, the stack trace will still show the `std::vector` code, but the cause would be somewhere else.

If your code is generic enough to run on Linux, you could try the linux target and run it inside one of the popular Linux memory debugging tools, e.g. `valgrind`.

0xjakob, 2023-06-25:
@wuyuanyi135 Any new observations? Did you find any minimal reproducing example?

wuyuanyi135, 2023-06-30:
Hi, @0xjakob. Sorry for the late response. Unfortunately, I don't have time yet to dig into this problem. I did switched on all stack smear and heap corruption detection but the crash still occured once due to `std::mutex` issue. However, I missed the core dump file so it was not very helpful. The crash probability was much lower than before though (from about within 24hr to about a week), which also makes the debugging even more tedious.

0xjakob, 2023-08-01:
@wuyuanyi135 Let's close this issue for now. If you encounter any trouble again, please feel free to re-open it.

closed 2023-08-01 by 0xjakob

muhammetaliakbay, 2025-01-26:
I am also having the same issue, it happens on every run of my program.

@0xjakob Have you found a solution so far?
If not I will try to ship the code as an example to reproduce.

chipweinberger, 2025-01-26:
@muhammetaliakbay , it's usually memory corruption in your own code.

muhammetaliakbay, 2025-01-27:
What kind of corruption could it be if the program is completely in Rust and with no unsafe blocks?