espressif/esp-idf #11480: iPhone can find bluetooth, but can't connect

Z-KY, 2023-05-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.2-dirty

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

Previously, when using the ESP32 or ESP32C3 module for testing, both iPhones and Android devices were able to connect to Bluetooth and Wi-Fi without any issues. However, with the current use of the **ESP32-C3FN4** chip, it has been observed that iPhones fail to connect to the device's Bluetooth, as indicated by the log. On the other hand, Android phones can connect successfully. Additionally, there have been continuous failures in connecting to Wi-Fi during testing. What could be the possible reasons behind these issues?

### What is the actual behavior?

iPhone can find bluetooth, but can't connect

### Steps to reproduce.

Run the example ”\examples\bluetooth\bluedroid\ble\gatt_server_service_table“ on the ESP32-C3FN4 chip 

### Debug Logs.

```plain
[15:06:09.293]ESP-ROM:esp32c3-api1-20210207
[15:06:09.293]Build:Feb  7 2021
[15:06:09.311]rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
[15:06:09.311]SPIWP:0xee
[15:06:09.311]mode:DIO, clock div:1
[15:06:09.311]load:0x3fcd5820,len:0x16fc
[15:06:09.316]load:0x403cc710,len:0x95c
[15:06:09.316]load:0x403ce710,len:0x2dbc
[15:06:09.316]entry 0x403cc710
[15:06:09.328][0;32mI (30) boot: ESP-IDF v5.1-dev-862-g09f7589ef2-dirty 2nd stage bootloader[0m
[15:06:09.328][0;32mI (30) boot: compile time May 23 2023 14:38:41[0m
[15:06:09.339][0;32mI (31) boot: chip revision: V004[0m
[15:06:09.339][0;32mI (35) boot.esp32c3: SPI Speed      : 80MHz[0m
[15:06:09.350][0;32mI (40) boot.esp32c3: SPI Mode       : DIO[0m
[15:06:09.350][0;32mI (45) boot.esp32c3: SPI Flash Size : 4MB[0m
[15:06:09.350][0;32mI (49) boot: Enabling RNG early entropy source...[0m
[15:06:09.356][0;32mI (55) boot: Partition Table:[0m
[15:06:09.375][0;32mI (58) boot: ## Label            Usage          Type ST Offset   Length[0m
[15:06:09.375][0;32mI (66) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[15:06:09.375][0;32mI (73) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[15:06:09.398][0;32mI (81) boot:  2 factory          factory app      00 00 00010000 00177000[0m
[15:06:09.398][0;32mI (88) boot: End of partition table[0m
[15:06:09.398][0;32mI (92) esp_image: segment 0: paddr=00010020 vaddr=3c110020 size=39e70h (237168) map[0m
[15:06:09.403][0;32mI (136) esp_image: segment 1: paddr=00049e98 vaddr=3fc93a00 size=0366ch ( 13932) load[0m
[15:06:09.436][0;32mI (139) esp_image: segment 2: paddr=0004d50c vaddr=40380000 size=02b0ch ( 11020) load[0m
[... 116 lines cut ...]
[15:06:11.828]rx(66): []
[15:06:21.544][0;33mW (14797) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x3e[0m
[15:06:24.086][0;33mW (15638) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x3e[0m
[15:06:24.923][0;33mW (16837) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x3e[0m
[15:06:26.123][0;33mW (17857) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x3e[0m
[15:06:27.146][0;33mW (18307) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x3e[0m
[15:06:27.597][0;33mW (18697) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x3e[0m
[15:06:27.986][0;32mI (29376) GATTS_TABLE_DEVICE: ESP_GATTS_CONNECT_EVT, conn_id = 0[0m
[15:06:38.670][0;32mI (29377) GATTS_TABLE_DEVICE: 6f a6 e5 9d e5 99 [0m
[15:06:38.670][0;33mW (30096) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x8[0m
[15:06:39.396][0;32mI (30097) GATTS_TABLE_DEVICE: ESP_GATTS_DISCONNECT_EVT, reason = 0x8[0m
[15:06:39.396][0;32mI (30100) GATTS_TABLE_DEVICE: advertising start successfully[0m
[15:06:39.396][0;32mI (36997) GATTS_TABLE_DEVICE: ESP_GATTS_CONNECT_EVT, conn_id = 0[0m
[15:06:46.290][0;32mI (36998) GATTS_TABLE_DEVICE: 6f a6 e5 9d e5 99 [0m
[15:06:46.290][0;32mI (37999) GATTS_TABLE_DEVICE: update connection params status = 16, min_int = 16, max_int = 32,conn_int = 24,latency = 0, timeout = 72[0m
[15:06:47.295][0;33mW (38106) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x8[0m
[15:06:47.395][0;32mI (38107) GATTS_TABLE_DEVICE: ESP_GATTS_DISCONNECT_EVT, reason = 0x8[0m
[15:06:47.406][0;32mI (38110) GATTS_TABLE_DEVICE: advertising start successfully[0m
[15:06:47.406]
```

### More Information.

Part of the schematic diagram is as follows
[image]

Alvin1Zhang, 2023-05-23:
Thanks for reporting.

Z-KY, 2023-05-23:
I also tried ESP-IDF v5.1-dev-862-g09f7589ef2-dirty but it didn't work. How can I solve this problem?

Weijian-Espressif, 2023-05-23:
@Z-KY  please test IDF commit `e8dba711ff93e4d93febcb7155e3b09f96ede1cb`  and let me know the results

Z-KY, 2023-05-24:
> @Z-KY please test IDF commit `e8dba711ff93e4d93febcb7155e3b09f96ede1cb` and let me know the results

I tried IDF commit e8dba711ff93e4d93febcb7155e3b09f96ede1cb, but it's even worse, even android phones can't connect.The log information does not seem to have changed.Is this what is the reason？

Z-KY, 2023-05-24:
> > @Z-KY please test IDF commit `e8dba711ff93e4d93febcb7155e3b09f96ede1cb` and let me know the results
> 
> I tried IDF commit [e8dba71](https://github.com/espressif/esp-idf/commit/e8dba711ff93e4d93febcb7155e3b09f96ede1cb), but it's even worse, even android phones can't connect.The log information does not seem to have changed.Is this what is the reason？

idf version: v5.2-dev-703-gefe919275e

Weijian-Espressif, 2023-05-24:
@Z-KY  

> @Z-KY please test IDF commit `e8dba711ff93e4d93febcb7155e3b09f96ede1cb` and let me know the results

I used iphone 6, iphone 11 and ipad air to test the C3 module, and the problem you mentioned was not reproduced. In addition, I don't know the circuit design, so I'm not sure if there is a problem with this circuit design.

Z-KY, 2023-05-24:
> @Z-KY
>
>
> I used iphone 6, iphone 11 and ipad air to test the C3 module, and the problem you mentioned was not reproduced. In addition, I don't know the circuit design, so I'm not sure if there is a [...]

It runs normally on the c3 development board, but there is a problem with this ESP32-C3FN4 chip (the iPhone cannot connect to bluetooth and wifi)

Weijian-Espressif, 2023-05-24:
1. Using ESP32-C3FN4 chip, is this circuit designed by you?
2. Please provide the sdkconfig file you use

Z-KY, 2023-05-24:
Yes，I designed this circuit, and this circuit can run on the esp32c2 board I designed before.
Here is the sdkconfig i use. [sdkconfig.txt](https://github.com/espressif/esp-idf/files/11551903/sdkconfig.txt)
I don't have the capture device.

Weijian-Espressif, 2023-05-25:
[image]
[image]
@Z-KY  
Here is our preliminary analysis result. If you haven't found the cause of the problem, you can follow the formal business docking process: https://www.espressif.com/zh-hans/contact-us/circuit-schematic-pcb-design-review, fill in form business will be docking

Alvin1Zhang, 2023-08-10:
Thanks for reporting, will close due to short of feedback, feel free to reopen with more updates.

closed 2023-08-10 by Alvin1Zhang

ffamar, 2024-04-16:
I have the same problem with ESP32 C3, idf 5.0.6 using BLE 5.0.

Android can connect without any problems, but if I try to connect with iPhone I get this warning and the device stops working
**BT_HCI: hcif disc complete: hdl 0x1, rsn 0x13**

Using the same circuit with idf 5.0.6 and BLE 4.2 there are no problems.

Z-KY, 2024-04-30:
The problem I found later is that the capacitor used by the crystal oscillator of the chip is not the right size. After replacing the two 10pf capacitors with 20pf capacitors, the iPhone can also connect to Bluetooth normally.
The TX performance of RF is not only affected by the power ripple, but also by the crystal vibration. The quality of the crystal oscillator itself is not good, and the frequency deviation is too large to affect the TX performance of the RF.