espressif/esp-idf #11482: bootloader version check?

hzhh110, 2023-05-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

esp32-c3-devkitm-1

### Operating System used.

macOS

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

How to view from bootloader.bin to check the version of the bootloader?

### What is the actual behavior?

How to view from bootloader.bin to check the version of the bootloader?

### Steps to reproduce.

How to view from bootloader.bin to check the version of the bootloader?

chipweinberger, 2023-05-23:
i'd probably run "strings bootload.bin | grep ESP-IDF v" in the terminal

dobairoland, 2023-05-23:
Just for future readers I mention that this information will be available by esptool starting with the upcoming v4.6 release.

```sh
esptool.py --chip esp32c3 image_info -v 2 bootloader.bin
```

KonstantinKondrashov, 2023-05-24:
Hi @chipweinberger @hzhh110!
In the master, recently, a new feature was added - https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/bootloader_image_format.html#bootloader-description, using this you can extract information (including ESP-IDF version) about the bootloader from app.

closed 2023-05-24 by KonstantinKondrashov

KaeLL, 2023-05-24:
@dobairoland Does it work with IDF 3.0 bootloaders?

dobairoland, 2023-05-25:
> Does it work with IDF 3.0 bootloaders?

The command itself will work but unfortunately, bootloader information won't show up for older bootloaders. I think that information will be present just in v5.2+.