espressif/esp-idf #11483: hid_host can't connect hid_device when hid_host only support bt and iocap is ESP_BT_IO_CAP_IO

jjj19950710, 2023-05-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.1

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32

### Power Supply used.

USB

### What is the expected behavior?

I expect the hid_host to be able to connect to the hid_device when I turn off LE Bluetooth and set iocap to ESP_BT_IO_CAP_IO.

### What is the actual behavior?

Connection stopped during password verification phase，It looks like calling esp_bt_gap_ssp_confirm_reply didn't take effect.

### Steps to reproduce.

1. esp_hid_host  turn off Bluetooth Low Energy
2. chang esp_hid_gap.c:640   esp_bt_io_cap_t iocap = ESP_BT_IO_CAP_IO;
3. build and run 

### Debug Logs.

```plain
esp_hid_host log:

I (27) boot: ESP-IDF v5.0.1-dirty 2nd stage bootloader
I (27) boot: compile time 17:43:27
I (28) boot: chip revision: v3.0
I (31) boot.esp32: SPI Speed      : 40MHz
I (36) boot.esp32: SPI Mode       : DIO
I (40) boot.esp32: SPI Flash Size : 2MB
I (45) boot: Enabling RNG early entropy source...
I (50) boot: Partition Table:
I (54) boot: ## Label            Usage          Type ST Offset   Length
I (61) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (68) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (76) boot:  2 factory          factory app      00 00 00010000 00100000
I (83) boot: End of partition table
I (87) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=19c40h (105536) map
I (134) esp_image: segment 1: paddr=00029c68 vaddr=3ffbdb60 size=04184h ( 16772) load
I (141) esp_image: segment 2: paddr=0002ddf4 vaddr=40080000 size=02224h (  8740) load
I (145) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=6abd8h (437208) map
I (306) esp_image: segment 4: paddr=0009ac00 vaddr=40082224 size=156cch ( 87756) load
I (354) boot: Loaded app from partition at offset 0x10000
I (354) boot: Disabling RNG early entropy source...
I (366) cpu_start: Pro cpu up.
[... 143 lines cut ...]
W (36913) BT_HCI: hcif link supv_to changed: hdl 0x80, supv_to 8000
I (36933) BT_L2CAP: L2CA_SetDesireRole() new:x1, disallow_switch:0
I (36943) BT_L2CAP: L2CAP - Calling Connect_Ind_Cb(), CID: 0x0040
I (36943) BT_L2CAP: L2CA_ErtmConnectRsp()  CID: 0x0040  Result: 0  Status: 0  BDA: 94b555fc216e  p_ertm_info:0x0
I (36953) BT_L2CAP: L2CA_ConfigReq()  CID 0x0040: fcr_present:0 (mode 0) mtu_present:1 (672)
I (36973) BT_L2CAP: L2CA_ConfigRsp()  CID: 0x0040  Result: 0 MTU present:0 Flush TO:0 FCR:0 FCS:0
I (36983) BT_L2CAP: L2CAP - Calling Config_Rsp_Cb(), CID: 0x0040
I (37003) BT_L2CAP: L2CA_DataWrite()  CID: 0x0040  Len: 10
I (37033) BT_L2CAP: L2CAP - Calling Disconnect_Ind_Cb(), CID: 0x0040  Conf Needed
I (37033) BT_L2CAP: L2CA_DisconnectRsp()  CID: 0x0040
I (37073) BT_L2CAP: L2CAP - Calling Connect_Ind_Cb(), CID: 0x0041
I (37073) BT_L2CAP: L2CA_ErtmConnectRsp()  CID: 0x0041  Result: 0  Status: 0  BDA: 94b555fc216e  p_ertm_info:0x0
I (37083) BT_L2CAP: L2CA_ConfigReq()  CID 0x0041: fcr_present:0 (mode 0) mtu_present:1 (672)
I (37113) BT_L2CAP: L2CA_ConfigRsp()  CID: 0x0041  Result: 0 MTU present:0 Flush TO:0 FCR:0 FCS:0
I (37123) BT_L2CAP: L2CAP - Calling Config_Rsp_Cb(), CID: 0x0041
I (37133) BT_L2CAP: L2CA_DataWrite()  CID: 0x0041  Len: 14
I (37163) BT_L2CAP: L2CA_DataWrite()  CID: 0x0041  Len: 283
I (37203) BT_L2CAP: L2CAP - Calling Disconnect_Ind_Cb(), CID: 0x0041  Conf Needed
I (37203) BT_L2CAP: L2CA_DisconnectRsp()  CID: 0x0041
W (67373) BT_HCI: hci cmd send: disconnect: hdl 0x80, rsn:0x5
W (67463) BT_HCI: hcif disc complete: hdl 0x80, rsn 0x22
I (67463) BT_L2CAP: L2CA_SetDesireRole() new:x1, disallow_switch:0
E (67463) BT_BTC:  btc_dm_auth_cmpl_evt() Authentication fail reason 5
```
```

boblane1, 2023-05-25:
Hi @jjj19950710，

I guess you change the esp_hid_gap.c in the example of esp_hid_device. If you change the `iocap` to `ESP_BT_IO_CAP_IO`, you shall send confirmation to the peer like the following code.

```c
static void bt_gap_event_handler(esp_bt_gap_cb_event_t event, esp_bt_gap_cb_param_t *param)
{
    ...
#if (CONFIG_BT_SSP_ENABLED == true)
    case ESP_BT_GAP_CFM_REQ_EVT:
        ESP_LOGI(SPP_TAG, "ESP_BT_GAP_CFM_REQ_EVT Please compare the numeric value: %"PRIu32, param->cfm_req.num_val);
        esp_bt_gap_ssp_confirm_reply(param->cfm_req.bda, true);
        break;
    case ESP_BT_GAP_KEY_NOTIF_EVT:
        ESP_LOGI(SPP_TAG, "ESP_BT_GAP_KEY_NOTIF_EVT passkey:%"PRIu32, param->key_notif.passkey);
        break;
    case ESP_BT_GAP_KEY_REQ_EVT:
        ESP_LOGI(SPP_TAG, "ESP_BT_GAP_KEY_REQ_EVT Please enter passkey!");
        break;
#endif
    ...
}
```

jjj19950710, 2023-05-26:
I  change the esp_hid_gap.c in the example of esp_hid_device.and then it run well.Thanks very much!

Alvin1Zhang, 2023-05-29:
Thanks for reporting and sharing the updates. Good to hear that the issue is resolved, feel free to reopen.

closed 2023-05-29 by Alvin1Zhang