espressif/esp-idf #11485: UART RX/TX Interrupt Example

moefear85, 2023-05-23:
### Is your feature request related to a problem?

It would be nice to have an example of how to turn on tx interrupts for the uart, register a tx interrupt handler, and to determine when there is space to write, to avoid polling, and set the threshold to an arbitrary value, such as 1.

moefear85, 2023-05-28:
I'm figuring this one out. I've managed to bypass uart_driver_install() and setup interrupts on my own. It requires some private headers though, so I realize why there hasn't been an example yet. I'll upload it to my github once its done, in case anyone is looking.

ScumCoder, 2024-02-04:
Same here.
I'll go as far as to say that the OP's wall of text looks as if it was written by me. The same tasks, the same glaring problems, the same screaming ways of solving them.

ScumCoder, 2024-02-04:
> I'll upload it to my github once its done, in case anyone is looking.

Could you please post a link? The only fork I see among your repos states that its "sole purpose is to provide patches [...] to enable multi-CDC Channels in tinyusb"