espressif/esp-idf #11487: RMT Example (led_strip) doesn't work properly

aquiot-inc, 2023-05-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.2

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32-WROOM

### Power Supply used.

USB

### What is the expected behavior?

The LED rainbow chase pattern is expected on the LEDs

### What is the actual behavior?

The LEDs have a pattern that seems random. Sometimes RGB=[0,0,0] is sent to all LEDs. This was verified by a RIGOL scope.

### Steps to reproduce.

1. Using https://github.com/espressif/esp-idf/tree/master/examples/peripherals/rmt/led_strip
2. The GPIO was changed to 13 on line 15
~~~C
#define RMT_LED_STRIP_GPIO_NUM      0
~~~

### Debug Logs.

```plain
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6940
ho 0 tail 12 room 4
load:0x40078000,len:15500
load:0x40080400,len:3844
0x40080400: _init at ??:?

entry 0x4008064c
I (29) boot: ESP-IDF v5.0.2 2nd stage bootloader
I (29) boot: compile time 16:48:32
I (29) boot: chip revision: v1.0
I (32) boot.esp32: SPI Speed      : 40MHz
I (36) boot.esp32: SPI Mode       : DIO
I (41) boot.esp32: SPI Flash Size : 2MB
I (45) boot: Enabling RNG early entropy source...
I (51) boot: Partition Table:
I (54) boot: ## Label            Usage          Type ST Offset   Length
I (62) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (69) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (77) boot:  2 factory          factory app      00 00 00010000 00100000
I (84) boot: End of partition table
I (88) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=09ed4h ( 40660) map
I (112) esp_image: segment 1: paddr=00019efc vaddr=3ffb0000 size=01f24h (  7972) load
I (115) esp_image: segment 2: paddr=0001be28 vaddr=40080000 size=041f0h ( 16880) load
I (124) esp_image: segment 3: paddr=00020020 vaddr=400d0020 size=18ca0h (101536) map
I (163) esp_image: segment 4: paddr=00038cc8 vaddr=400841f0 size=0806ch ( 32876) load
I (182) boot: Loaded app from partition at offset 0x10000
I (183) boot: Disabling RNG early entropy source...
I (194) cpu_start: Pro cpu up.
I (194) cpu_start: Starting app cpu, entry point is 0x4008106c
0x4008106c: call_start_cpu1 at C:/apps/ESP32/ESP-IDF-Tools-2.9.1/frameworks/esp-idf-v5.0.2/components/esp_system/port/cpu_start.c:141

I (0) cpu_start: App cpu up.
I (208) cpu_start: Pro cpu start user code
I (208) cpu_start: cpu freq: 160000000 Hz
I (208) cpu_start: Application information:
I (213) cpu_start: Project name:     led_strip
I (218) cpu_start: App version:      1
I (223) cpu_start: Compile time:     May 23 2023 16:48:23
I (229) cpu_start: ELF file SHA256:  941087271fbc3f34...
I (235) cpu_start: ESP-IDF:          v5.0.2
I (240) cpu_start: Min chip rev:     v0.0
I (244) cpu_start: Max chip rev:     v3.99
I (249) cpu_start: Chip rev:         v1.0
I (254) heap_init: Initializing. RAM available for dynamic allocation:
I (261) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (267) heap_init: At 3FFB2870 len 0002D790 (181 KiB): DRAM
I (273) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (280) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (286) heap_init: At 4008C25C len 00013DA4 (79 KiB): IRAM
I (294) spi_flash: detected chip: generic
I (297) spi_flash: flash io: dio
W (301) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (315) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (325) example: Create RMT TX channel
I (335) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (335) example: Install led strip encoder
I (345) example: Enable RMT TX channel
I (345) example: Start LED rainbow chase
```

### More Information.

I have also tried with ESP-IDF v5.0.1, same result

I tried the example when my own code wasn't working as expected. 

I was previously using ESP-IDF v4.4 with the old driver `#include "driver/rmt.h"` and it worked fine. The old deprecated driver will also work properly in v5.0.x. Only `"driver/rmt_tx.h"` doesn't work properly.

Example of all pixel values of zero (this should not happen):
[image]

At other times, the pattern looks more like expeted:
[image]

suda-morris, 2023-05-24:
Hi @aquiot-inc when you see the "zero" frame, it's because the example is "clearing" the LEDs in [here](https://github.com/espressif/esp-idf/blob/master/examples/peripherals/rmt/led_strip/main/led_strip_example_main.c#L122)

I think it's not a bug neither in the example nor the driver. The v4.4 version is also having this clearing logic: https://github.com/espressif/esp-idf/blob/release/v4.4/examples/peripherals/rmt/led_strip/main/led_strip_main.c#L111

aquiot-inc, 2023-05-24:
> Hi @aquiot-inc when you see the "zero" frame, it's because the example is "clearing" the LEDs in [...]

Thanks for having a look.

Below gives a recognizable pattern. Especially since I only have 2 LEDs. The `start_rgb += 60;` makes the pattern too fast and the clearing doesn't help that.

That being said, the reason I tried the example is because my app broke (was emitting strange pulse codes like all 1's when trying to write a dim red) when porting to the new framework. So I will have to revisit my actual app but in the meantime I'm just going to keep using the old framework for now. In the meantime, perhaps someone will create/update a component for driving RGB LEDs with the new framework. This for example: https://github.com/JSchaenzle/ESP32-NeoPixel-WS2812-RMT

~~~diff
-#define RMT_LED_STRIP_GPIO_NUM      0
+#define RMT_LED_STRIP_GPIO_NUM      13

-#define EXAMPLE_LED_NUMBERS         24
+#define EXAMPLE_LED_NUMBERS         2
 #define EXAMPLE_CHASE_SPEED_MS      10

 static const char *TAG = "example";

 static uint8_t led_strip_pixels[EXAMPLE_LED_NUMBERS * 3];

 /**
  * @brief Simple helper function, converting HSV color space to RGB color space
  *
  * Wiki: https://en.wikipedia.org/wiki/HSL_and_HSV
@@ -104,25 +104,28 @@ void app_main(void)

     ESP_LOGI(TAG, "Start LED rainbow chase");
     rmt_transmit_config_t tx_config = {
         .loop_count = 0, // no transfer loop
     };
     while (1) {
         for (int i = 0; i < 3; i++) {
             for (int j = i; j < EXAMPLE_LED_NUMBERS; j += 3) {
                 // Build RGB pixels
                 hue = j * 360 / EXAMPLE_LED_NUMBERS + start_rgb;
-                led_strip_hsv2rgb(hue, 100, 100, &red, &green, &blue);
+                led_strip_hsv2rgb(hue, 100, 10, &red, &green, &blue);
                 led_strip_pixels[j * 3 + 0] = green;
                 led_strip_pixels[j * 3 + 1] = blue;
                 led_strip_pixels[j * 3 + 2] = red;
             }
             // Flush RGB values to LEDs
             ESP_ERROR_CHECK(rmt_transmit(led_chan, led_encoder, led_strip_pixels, sizeof(led_strip_pixels), &tx_config));
+            // ESP_ERROR_CHECK(rmt_tx_wait_all_done(led_chan, portMAX_DELAY));
             vTaskDelay(pdMS_TO_TICKS(EXAMPLE_CHASE_SPEED_MS));
-            memset(led_strip_pixels, 0, sizeof(led_strip_pixels));
-            ESP_ERROR_CHECK(rmt_transmit(led_chan, led_encoder, led_strip_pixels, sizeof(led_strip_pixels), &tx_config));
-            vTaskDelay(pdMS_TO_TICKS(EXAMPLE_CHASE_SPEED_MS));
+            // memset(led_strip_pixels, 0, sizeof(led_strip_pixels));
+            // ESP_ERROR_CHECK(rmt_transmit(led_chan, led_encoder, led_strip_pixels, sizeof(led_strip_pixels), &tx_config));
+            // ESP_ERROR_CHECK(rmt_tx_wait_all_done(led_chan, portMAX_DELAY));
+            // vTaskDelay(pdMS_TO_TICKS(EXAMPLE_CHASE_SPEED_MS));
         }
-        start_rgb += 60;
+        start_rgb += 5;
     }
 }
~~~

closed 2023-05-24 by aquiot-inc